gazebosim/gz-sim #1316: ignition installed on windows 8 but cant gazebo after "ign" in command line

Kakcalu13, 2022-01-30:
## Environment
* OS Version: **Windows 8**
* Source or binary build? **Both**
[image]

    
<details>
  
```
(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>igngazebo
'igngazebo' is not recognized as an internal or external command,
operable program or batch file.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign-gazebo
'ign-gazebo' is not recognized as an internal or external command,
operable program or batch file.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign --versions
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo --versions
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>s
's' is not recognized as an internal or external command,
operable program or batch file.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo >> log.txt

(ign-ws) C:\Users\Kevin\Documents\ign-ws>clear

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>ign gazebo --versions
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.

(ign-ws) C:\Users\Kevin\Documents\ign-ws>
```

</details>

## Description
* Expected behavior: Not supported :P I just want to at least try! It'd be great to have ignition on my personal pc!
* Actual behavior: It's actually installed on windows 8. I just can't open gazebo so do you know any trick to open it or run command without have a space in it? 

## Steps to reproduce

1. ign gazebo

## Output

[image]

Kakcalu13, 2022-01-30:
If there is no way to open ignition without `ign gazebo`, I will just close this issue.

traversaro, 2022-02-01:
I think to be able to run `ign gazebo` you just need to comment out the `if(NOT WIN32)`  in https://github.com/ignitionrobotics/ign-gazebo/blob/c08fc87e2b965ab388f6dc4a74f8668213ecae8f/src/CMakeLists.txt#L191 . I may have tried back in October 2021 when I worked a bit. However, note that probably you may encounter other problems (such as https://github.com/conda-forge/libignition-gazebo-feedstock/issues/6 or https://github.com/conda-forge/libignition-gazebo-feedstock/issues/32). I would be happy to help you if you want to work on solving them, but consider anyhow that will probably not work out of the box.

Note that when reporting issue like this, it is useful to report the version of Ignition Gazebo that you are using. For when you are using the binary, you can easily get the version with `conda list`, while when building from source `conda list`  (for the dependencies) and `vcs export --exact` for the ignition dependencies.

Kakcalu13, 2022-02-04:
Hey there!

Thank you so much for your reply! 

No worries, I will just upgrade my w8 to w11/w10 later. 

I was hoping that I could somewhat find shortest way to download without needed w10 :P 

Got it, I will keep that in mind to share the log in future

closed 2022-02-04 by Kakcalu13