gazebosim/gz-sim #1336: Gazebo viewer only

Kakcalu13, 2022-02-14:
## Desired behavior
Able to download the lightweight gazebo viewer where it just displays everything inside gazebo without editor like Fbx previewer concept. It's a lightweight program and hopefully, crossover on any platform of gazebo! 

## Additional context
I was thinking like:
1) Display the gazebo scene only.
2) No tools, add ons or editors
3) Has physics engine and able to move due to the bridge

Hopefully this is a simple request! I have a feeling that this might be a big request :P 

Keep up with awesome work! Love it so far! 

Side note:
Did you know that this worked well with ignition? 
https://ambientcg.com/

I used it last night and it worked well! I just gotta scale and all that but it worked perfectly!

ahcorde, 2022-02-15:
Hi @Kakcalu13,

the feature that you are requesting is already done. Inside `~/.ignition/gazebo` there is a file called `gui.config` which loads the default plugins which are going to run in your Ignition Gazebo GUI.

If you only want to run the visualizer, you can create your own `gui.config` file, then you should run gazebo using the option `--gui-config`: 

```bash
ign gazebo -v 4 shapes.sdf  --gui-config only_visualizer.config
```
### only_visualizer.config file
```xml
<?xml version="1.0"?>

<!-- Window -->
<window>
  <width>1000</width>
  <height>845</height>
  <style
    material_theme="Light"
    material_primary="DeepOrange"
    material_accent="LightBlue"
    toolbar_color_light="#f3f3f3"
    toolbar_text_color_light="#111111"
    toolbar_color_dark="#414141"
    toolbar_text_color_dark="#f3f3f3"
    plugin_toolbar_color_light="#bbdefb"
    plugin_toolbar_text_color_light="#111111"
    plugin_toolbar_color_dark="#607d8b"
    plugin_toolbar_text_color_dark="#eeeeee"
  />
  <menus>
    <drawer default="false">
    </drawer>
  </menus>
  <dialog_on_exit>true</dialog_on_exit>
</window>

<!-- GUI plugins -->

<!-- 3D scene -->
<plugin filename="MinimalScene" name="3D View">
  <ignition-gui>
    <title>3D View</title>
    <property type="bool" key="showTitleBar">false</property>
    <property type="string" key="state">docked</property>
  </ignition-gui>

  <engine>ogre2</engine>
  <scene>scene</scene>
  <ambient_light>0.4 0.4 0.4</ambient_light>
  <background_color>0.8 0.8 0.8</background_color>
  <camera_pose>-6 0 6 0 0.5 0</camera_pose>
</plugin>

<plugin filename="GzSceneManager" name="Scene Manager">
  <ignition-gui>
    <property key="resizable" type="bool">false</property>
    <property key="width" type="double">5</property>
    <property key="height" type="double">5</property>
    <property key="state" type="string">floating</property>
    <property key="showTitleBar" type="bool">false</property>
  </ignition-gui>
</plugin>
<plugin filename="InteractiveViewControl" name="Interactive view control">
  <ignition-gui>
    <property key="resizable" type="bool">false</property>
    <property key="width" type="double">5</property>
    <property key="height" type="double">5</property>
    <property key="state" type="string">floating</property>
    <property key="showTitleBar" type="bool">false</property>
  </ignition-gui>
</plugin>
```

Hope this solve your request. Please close the issue is your question is answered

Kakcalu13, 2022-02-17:
Thanks!! Testing it now and ill share the result asap!

Kakcalu13, 2022-02-17:
I just tested it. That did what I wanted, but I afraid it required a full package. I'm looking for something that can be done with one package due to memory size. 

The package allowed me to load is `libignition-gazebo3-dev` with `libignition-gui3-dev`

libignition-gui3-dev alone is like 1.2 gb if I remember right and it needed gui3 which is like 200 mb. 

Do you know if there's a way to reduce the memory size for docker? I'm trying to find something smaller and lightweight. 

A completely and unrelated question :angel: 

Do you know anyone who is very good with [sdf exporter](https://github.com/ignitionrobotics/ign-gazebo/issues/1348)?

azeey, 2023-02-16:
I believe you don't need the `-dev` packages for Garden. You'll need to install `libgz-sim7` and the `gz-sim7-cli`. There might be a couple more, but I would try those first.

Kakcalu13, 2023-02-16:
Thanks, I will try it out soon!