gazebosim/gz-sim #1350: Launch ignition through network issue

Kakcalu13, 2022-02-18:
## Environment
* OS Version: Ubuntu 20.04
* Source
  **Citadel** and **Fortress** (Citadel on docker and desktop while laptop is on Fortress. I use them at the same time every time)
 
 TLDR: I followed on [doc](https://ignitionrobotics.org/api/gazebo/4.6/distributedsimulation.html) and [example](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo6/examples/scripts/distributed) and it returns either
```
Error [Element.cc:914] Missing element description for [level]
```
from example and doc.

What is the right way to do this also can I update doc on example? :angel:

<details>
So I was trying to launch gazebo and respawn models during runtime. I succeed with it on the local only. 

So I wanted to send model to gazebo, build a new bridge then connect with the topics from the new model, or launch gazebo from other computer/container in docker. Like I wanted to keep gazebo as a server while clients send models to gazebo in server, you know? 

So I decided to start off with load models inside the gazebo through a different computer. 

So, I started with the laptop with this command to load empty world,
`ign gazebo --force-version 3.12.0 --network-role=primary` 
the error returns `Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work."

So I tried to find the network path or something like `ign gazebo -p 127.x.x.x -r test_shape.sdf` or maybe `ign service -s /world/world/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 300 --req 'sdf_filename: ''"test_shape.sdf" pose: {position: {z: 5}} ''name: "new_name" ''allow_renaming: true'  -p 127.x.x.x`

Something like that to pass the information through network. I couldn't find anything like this in doc. So I tried with --network-role and --secondary-role again but with a different tag because  I found [example](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo6/examples/scripts/distributed) but then I noticed that some commands on it are not same as doc. 

So, I decided to test it by having ./primary in my laptop(with/without --force-version 3.12.0  tag) and 3 of ./secondary.sh in my desktop. No response or any excited event happened, just that error returned `Error [Element.cc:914] Missing element description for [level]` or `[Err] [ServerPrivate.cc:156] Failed to start network, simulation terminating`. 
[image]

./primary is on laptop. Nothing.

I tried it on docker just in case if it would make a difference. Same result as expected. So, I tried all of them on one desktop (without or with ./standalone.sh) and it worked, obviously. 

So, 
  
  Now, here are my questions. The first question is how do I start with `--network-role primary` and `--secondary-role <N>`? Second question is does `--secondary-role` required to be fixed numbers or can I just load it during runtime without worry about limited numbers? Thirdly, is this a bug? lol  Fourth, what is the network path? For example, in ros2..it is `ROS_DOMAIN=ID` but what is the command for Ign? Lastly, once I learn how to do all of this..can I update on your doc as well? I feel like this will be used by people frequently.
</details>

## Description
* Expected behavior: Send model to gazebo on a different computer/container
* Actual behavior: Nothing, no responses.

## Steps to reproduce

1. ign gazebo --force-version 3.12.0 --network-role primary

## Output
`[Err] [LevelManager.cc:172] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.`

Shared few in details tab too.

chapulina, 2022-03-01:
> does --secondary-role required to be fixed numbers or can I just load it during runtime without worry about limited numbers?

If you mean `--network-secondaries`, yes, that does need to be set at startup. From the docs:

_–network-secondaries=<N> - The number of secondaries expected to join. Simulation will not begin until N secondaries have been discovered._

> I tried to find the network path or something like ign gazebo -p 127.x.x.x -r test_shape.sdf

I haven't found any documentation about a `-p` option

Kakcalu13, 2022-03-01:
Hey, thank you so much for your time!

> I haven't found any documentation about a -p option

I assumed with the `-p` as port. I was trying to connect from one pc to another pc/docker.  I just re-read what I wrote..looks like I forgot to add the port but it should be like 20003 for example. 

> –network-secondaries= - The number of secondaries expected to join. Simulation will not begin until N secondaries have been discovered.

I was trying to make a server without expected an exact number. I was trying to see if it is possible to generate as many models as it can and delete as many models as it can during runtime. 

P.S. my sincerely apologies if this is not a bug!

Kakcalu13, 2022-03-03:
Closing the issue since this needs to clarification and re-do my work further to provide more details

closed 2022-03-03 by Kakcalu13