gazebosim/gz-sim #1361: /clock not published when using  "ros_ign_bridge/parameter_bridge" to convert clock message to ROS

paszoujp, 2022-02-24:
## Description
* Expected behavior: 
use ros_ign_bridge/parameter_bridge node to convert ign /clock topic to ros clock topic   
* Actual behavior: 
after using the ros_ign_bridge/parameter_bridge node with `/clock@rosgraph_msgs/Clock@ignition.msgs.Clock`, the clock stopped publishing even in ignition topics, and output this error:
[image]  
I found the code generating the error. It seems that the ignition-gazebo checks on whether there are other clock publishers by only check the topic name, and thus mistaken the `os_ign_bridge/parameter_bridge` ,which is actually a listener started up earlier, as a clock publisher and stopped publish clock msgs.  
I have done the following modification to route this issue, but I doubt it be safe in other situations:
[image]

## Steps to reproduce

roslaunch the folloing .launch file:
```xml
<launch>
<node
    pkg="ros_ign_gazebo"
    type="ign_gazebo"
    name="ign_gazebo"
    output="screen"
    required="true"
    args="-r -v 3 empty.sdf ">
  </node>

<node
    pkg="ros_ign_bridge"
    type="parameter_bridge"
    name="$(anon ros_ign_bridge)"
    output="screen"
    args="/clock@rosgraph_msgs/Clock@ignition.msgs.Clock ">
</node>
</launch>
```

## Output
[image]

azeey, 2022-02-24:
The issue is that the `parameter_bridge`, when used with `@` separator, becomes a bidirectional bridge, which means it creates a subscriber as well as a publisher on `/clock`. This confuses ign-gazebo into thinking there's another instance running and so it will instead publish only on `/world/<name>/clock`. 

The solution is to use the `parameter_bridge` in a single direction mode. I just realized this is not in the README, but you can find it in the usage output when running the command. You can also see it here:https://github.com/ignitionrobotics/ros_ign/blob/dad6d8e540f672085f2fa28a42df3760b4721709/ros_ign_bridge/src/parameter_bridge.cpp#L41-L63

I'll close the ticket since this is working as intended. Perhaps an issue on `ros_ign` for better documentation is more appropriate.

closed 2022-02-24 by azeey

paszoujp, 2022-02-25:
got it