gazebosim/gz-sim #1362: Crash on M1 chip instantly even using docker

Kakcalu13, 2022-02-24:
## Environment
* OS Version: macOS Monterey with Apple M1 chip using Docker (Ubuntu 20.04)
* Source or binary build: **Citadel binary build**

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 

## **GUI Bug**
      - [ ] GUI rendering error. 
Full log after launch gazebo such as `ign gazebo --verbose` 
[log.txt](https://github.com/ignitionrobotics/ign-gazebo/files/8134442/log.txt)

Ran `ign gazebo -v 3` and it has the same as above. Ran `ign gazebo -v 4` and it showed 4 extra blue line. Here log while it was inside middle log like above first log:
[log 2.txt](https://github.com/ignitionrobotics/ign-gazebo/files/8134457/log.2.txt)

I tried to do it locally and this results the exact same as this [issue](https://github.com/osrf/homebrew-simulation/issues/1628)

So that leaves me two options: Pull gazebo from gazebo website and run docker using [feagi.yml](https://github.com/feagi/feagi/tree/staging/docker)

So, I ran feagi.yml following this steps:
1) git clone git@github.com:feagi/feagi.git
2) cd feagi/docker
3) docker-compose -f feagi.yml build --no-cache
4) Once it's complete, `docker-compose -f feagi.yml up`
5) Open [Browser (VNC)](http://127.0.0.1:6080)
6) Wait for it to complete colcon build so it can show you the gazebo. You should see the error instantly once Colon build ends.

The error looks exact same as log.txt (That's actually where I copied all from)

So, that leaves me the last option.
1) `docker pull gazebo` [(This is from gazebo's docker hub)](https://hub.docker.com/_/gazebo)
2) `docker run gazebo`
You can see the error instantly too. 

The full log (I can't get more than that)
```
station1@Station1s-Mini docker % docker run gazebo                                     
Error setting socket option (IP_MULTICAST_IF).
Error setting socket option (IP_MULTICAST_IF).
Did you set the environment variable IGN_IP with a correct IP address? 
  [172.17.0.2] seems an invalid local IP address.
  Using 127.0.0.1 as hostname.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
qemu: uncaught target signal 6 (Aborted) - core dumped
```

So this seems like IGN_IP issue, GPU and ogre altogether. 

    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on real hardware
      - [ ] running in Docker/Singularity
    - Rendering system info:
       Inside docker since it's using Ubuntu 20.04
      - On Linux, provide the outputs of the following commands:
      
          ```
            root@d45c22edad72:~# bash 
            root@d45c22edad72:~# LANG=C lspci -nn | grep VGA  # might require installing pciutils
            root@d45c22edad72:~# echo "$DISPLAY"
            :1.0
            root@d45c22edad72:~# LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
            direct rendering: Yes
                Preferred profile: core (0x1)
                Max core profile version: 4.5
                Max compat profile version: 3.1
                Max GLES1 profile version: 1.1
                Max GLES[23] profile version: 3.2
            OpenGL vendor string: Mesa/X.org
            OpenGL renderer string: llvmpipe (LLVM 11.0.0, 128 bits)
            OpenGL core profile version string: 4.5 (Core Profile) Mesa 20.2.6
            OpenGL core profile shading language version string: 4.50
            OpenGL core profile context flags: (none)
            OpenGL core profile profile mask: core profile
            OpenGL version string: 3.1 Mesa 20.2.6
            OpenGL shading language version string: 1.40
            OpenGL context flags: (none)
            OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.2.6
            OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
            root@d45c22edad72:~# ps aux | grep Xorg
            root       922  0.0  0.3 151008  7212 pts/0    Sl+  17:38   0:00 /usr/bin/qemu-x86_64 /usr/bin/grep --color=auto Xorg
            root@d45c22edad72:~# sudo env LANG=C X -version
            /usr/bin/env: ‘X’: No such file or directory
            root@d45c22edad72:~# 
          ```

- On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
```
        station1@Station1s-Mini docker % system_profiler SPDisplaysDataType
     Graphics/Displays:

    Apple M1:

      Chipset Model: Apple M1
      Type: GPU
      Bus: Built-In
      Total Number of Cores: 8
      Vendor: Apple (0x106b)
      Metal Family: Supported, Metal GPUFamily Apple 7
      Displays:
        ASUS VP28U:
          Resolution: 5120 x 2880 (5K/UHD+ - Ultra High Definition Plus)
          UI Looks like: 2560 x 1440 @ 60.00Hz
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Rotation: Supported
          Automatically Adjust Brightness: Yes
      station1@Station1s-Mini docker % 
```

## Description
* Expected behavior: Just able to load gazebo. `ign gazebo` 
* Actual behavior: [log.txt](https://github.com/ignitionrobotics/ign-gazebo/files/8134656/log.txt)

Addition information: I ran --platform = linux/arm64 too or anything related to this command. Everything else worked except Gazebo.

brolin-botsync, 2022-03-27:
Hi @chapulina ,
I am also facing the exact same issue. I am trying to get gazebo running on docker but no success. I have an M1 mac and want to get this working soon. If you managed to get it working can you please help you here.

Kakcalu13, 2022-03-28:
Hi @brolin-botsync,

Things are ready to provide, see here: https://github.com/osrf/homebrew-simulation/issues/1628

I have this M1 few weeks ago and I'm so addicted with this computer! Super fast! So, keep that in mind..I joined Mac world at the same time I got this M1. 

Maybe you know how to do this. I will list them all below.

See here:
https://github.com/srmainwaring/ign-rendering/wiki

I understand that this is outdated for M1 now. 
When this wiki says,
`cd /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation`

It's for Mac with intel. This M1's location is now `/opt/homebrew/Library/Taps/osrf/homebrew-simulation`

Another thing, see here in Formula.
https://github.com/osrf/homebrew-simulation/pull/1823

Full Formula file:
https://github.com/osrf/homebrew-simulation/blob/a9302ef21aa59e7213adb97f7a7d532830120b2f/Formula/ogre2.2.rb

I tried this too, but didn't work. It could because I'm inexperience with this area. Do you know how to do this?  

If you do know how to set them up, can you teach me how? Apparently, docker in container couldn't run gazebo on M1 due to chip difference even if you set container as linux. So, to reproduce this has to be on local first, right? If we can make it work on local, then we can find a way in container immediately. Let me know if it worked for you. Once it worked for you, tell us the secret too! ;) 

I know this is a little too complicated but I think it's because of homebrew situation.

srmainwaring, 2022-03-28:
@Kakcalu13  and @brolin-botsync - most of the changes required to support Ignition on macOS using Metal have now been merged into the main branches of various repos. The [outstanding PR for macOS is in ignition-gazebo](https://github.com/ignitionrobotics/ign-gazebo/pull/1225) (this PR runs fine on macOS but there are some Linux/OpenGL threading compatibility issues to resolve for GPU enabled sensors). I believe that the formula for Ogre2.2 in osrf/simulation is also now up to date for Metal and should work on a M1 machine as a couple of the devs reviewing the changes for Metal support were using these machines.

I should update or remove the wiki notes you refer to above - they have mostly been superseded now. 

For Metal support you should use the Garden branch and will need to build from source.

I'd be happy to help you both get a native installation of ignition gazebo running on macOS - it would help figure out what the platform specific installation notes should look like. Starting point is to install Garden and test that the examples in ignition-rendering are working with the ogre2 metal flags (e.g. `ign-rendering/examples/build/ogre2demo ogre2 metal`). If ogre2 metal rendering is running then you should be able to get the full Gazebo environment working.

I have not had much luck running ignition using virtualised versions of Ubuntu with accelerated graphics post Ignition Edifice. The main problem is a lack of OpenGL 4.3+ support in the hypervisor drivers. For example VMWare fusion will only support OpenGL 4.1 and that is not sufficient for Ogre2.2. I have not looked at running docker machines with hardware acceleration for a while, but suspect you may run into a similar issue there. Software rendering may work, but that will be an order of magnitude slower for interesting rendering environments.

Kakcalu13, 2022-03-28:
Thank you so much for your time, @srmainwaring! This is actually helpful information! 

I will update you with the status on virtualised version part. 

I've been trying to install Citadel/Fortress on M1 but I will start with Garden. I will keep you guys posted along with steps or issues.

srmainwaring, 2022-03-28:
The key step is getting ignition-rendering (`main` branch) built and running the examples. Once you're at that stage ping me and I can walk you through the rest of the set up.

brolin-botsync, 2022-03-28:
@Kakcalu13 I am at the same spot as you are. I bought a mac book just 2 days back. I am also new to the mac environment. Its like fish out of water. I am still finding my way around the OS. I am trying to run ROS and Gazebo using docker on the machine and I am finding it exhausting. Any help is appreciated from either you or @srmainwaring. I will also try the suggestion given by @srmainwaring. I will keep you posted on my improvements.

Kakcalu13, 2022-03-28:
@brolin-botsync Join my boat, I have more space here. :P

I'm currently building one. Let us know on your end. I will post more information on my end

Kakcalu13, 2022-03-28:
After spend like 10 minutes, I eventually realized I was in ign-rendering7 by default. 

So here is what I ran after removed ign-rendering7 branch:
1) git clone -b main git@github.com:ignitionrobotics/ign-rendering.git
2) cd ign-rendering (following this [link](https://ignitionrobotics.org/api/rendering/5.0/installation.html) now)
3) mkdir examples/build
4) cd examples
5) cmake .. -DCMAKE_INSTALL_PREFIX=build/ _(and /build/)_
6) make

This instantly gave me an error:
[CMakeError.log](https://github.com/ignitionrobotics/ign-gazebo/files/8364538/CMakeError.log)
[CMakeOutput.log](https://github.com/ignitionrobotics/ign-gazebo/files/8364539/CMakeOutput.log)

So I followed exact same as the link. 

Same result.

but I didn't have this issue when I was using ign-rendering7 (just saying `ogre2demo` is not found command)

So I assumed I might need to follow garden from the [docs](https://ignitionrobotics.org/docs/garden/install_osx_src). This didn't go well either. 

Did I started somewhere wrong?

srmainwaring, 2022-03-28:
> `git clone -b feature/ign-gazebo7-metal-demo git@github.com:srmainwaring/ign-gazebo.git` seems it's more promising!
> 
> There are few errors which can be resolved on my own end. I just gotta download few things. I will post the result shortly!

ok - you're jumping straight to getting the ignition-gazebo metal PR built! I'd still recommend going step by step through the libraries, testing at each state, as it will be easier to try figure out what's going wrong if gazebo doesn't run from the get go.

Kakcalu13, 2022-03-28:
> ok - you're jumping straight to getting the ignition-gazebo metal PR built! I'd still recommend going step by step through the libraries, testing at each stage, as it will be easier to try figure out what's going wrong if gazebo doesn't run from the get go.

Sorry! I just never had any luck with gazebo on M1. I'm dying to see what it look like on M1 since few weeks ago lol :'( 

> ```shell
> $ cd ~/ign_ws
> $ colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib
> ```

I think this is the problem here. I don't think I'm on the same page. First of all, I was following [ign-rendering](https://ignitionrobotics.org/api/rendering/5.0/installation.html)**5** instead of latest. This is the result above. So scratch that. 

So, I followed: [here](https://ignitionrobotics.org/api/rendering/7.0/installation.html)

instead of `git clone https://github.com/ignitionrobotics/ign-rendering -b ign-rendering<#>`
I ran `git clone https://github.com/ignitionrobotics/ign-rendering -b main` inside `~/workspace`
Then I ran this line as you mentioned above:
`colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib`

This part didn't go through. 

The error log:
```
station1@Station1s-Mac-mini ign-rendering % colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib
[0.813s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/usr/local/opt/dartsim@6.10.0' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.813s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/usr/local/opt/qt@5' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> ignition-rendering7
--- stderr: ignition-rendering7                         
CMake Warning at /opt/homebrew/Cellar/cmake/3.22.3/share/cmake/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findignition-common5-geospatial.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "ignition-common5-geospatial", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "ignition-common5-geospatial" (requested version 5.0.0) with any of the
  following names:

    ignition-common5-geospatialConfig.cmake
    ignition-common5-geospatial-config.cmake

  Add the installation prefix of "ignition-common5-geospatial" to
  CMAKE_PREFIX_PATH or set "ignition-common5-geospatial_DIR" to a directory
  containing one of the above files.  If "ignition-common5-geospatial"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /Users/station1/workspace/install/lib/cmake/ignition-common5/ignition-common5-config.cmake:188 (find_dependency)
  /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:189 (find_package)
  CMakeLists.txt:38 (ign_find_package)

CMake Warning at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:189 (find_package):
  Found package configuration file:

    /Users/station1/workspace/install/lib/cmake/ignition-common5/ignition-common5-config.cmake

  but it set ignition-common5_FOUND to FALSE so package "ignition-common5" is
  considered to be NOT FOUND.  Reason given by package:

  ignition-common5 could not be found because dependency
  ignition-common5-geospatial could not be found.

Call Stack (most recent call first):
  CMakeLists.txt:38 (ign_find_package)

CUDA_TOOLKIT_ROOT_DIR not found or specified
CMake Warning at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:55 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [optix]: Missing dependency [OptiX].
      ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:172 (ign_configure_build)

CMake Error at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:172 (ign_configure_build)

CMake Error at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [ignition-common5] (Components: graphics, events,
  geospatial)
Call Stack (most recent call first):
  CMakeLists.txt:172 (ign_configure_build)

CMake Error at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:63 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:172 (ign_configure_build)

CMake Error at /Users/station1/workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:68 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:172 (ign_configure_build)

---
Failed   <<< ignition-rendering7 [1.25s, exited with code 1]

Summary: 0 packages finished [1.56s]
  1 package failed: ignition-rendering7
  1 package had stderr output: ignition-rendering7

```

srmainwaring, 2022-03-28:
Looks like you need to modify a couple of steps from the [install instructions](https://ignitionrobotics.org/docs/garden/install_osx_src) (or the env variables got wiped)

1. `dart` and `qt5`

Modify the paths below for an M1 machine

```bash
# dartsim@6.10.0
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/usr/local/opt/dartsim@6.10.0
export DYLD_FALLBACK_LIBRARY_PATH=${DYLD_FALLBACK_LIBRARY_PATH}:/usr/local/opt/dartsim@6.10.0/lib:/usr/local/opt/octomap/local
export PKG_CONFIG_PATH=${PKG_CONFIG_PATH}:/usr/local/opt/dartsim@6.10.0/lib/pkgconfig
# qt5
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/usr/local/opt/qt@5
```

2. ignition-common5 has a new dependency on `gdal` which you'll need to install with brew

```bash
brew install gdal
```

3. Check that you've pulled the latest versions of all the dependencies (i.e. ignition-common5) and you're on the correct branch for ignition garden. For most repos this will be `main`, but for some you'll want specific versions (i.e. `ignition-msgs9`, `ign-transport12`). 

```bash
$ cd ~/ign_ws/src/ign-common
$ git fetch
$ git pull
```

4. Try building each of the libraries in the order given by `colcon graph` by adding `--packages-select <ign-library#>` to the colcon build command above.

Kakcalu13, 2022-03-28:
Unfortunately, that didn't work :( I modified the path like this:

```
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/homebrew/opt/dartsim@6.10.0
export DYLD_FALLBACK_LIBRARY_PATH=${DYLD_FALLBACK_LIBRARY_PATH}:/opt/homebrew/opt/dartsim@6.10.0/lib:/opt/homebrew/opt/octomap/
export PKG_CONFIG_PATH=${PKG_CONFIG_PATH}:/opt/homebrew/opt/dartsim@6.10.0/lib/pkgconfig
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/homebrew/opt/qt@5
```

So, I decided to start all over and started from scratch:
1) /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
(skipped brew install python3, python3 -m pip install vcstool, python3 -m pip install -U colcon-common-extensions because I already installed them)
2) mkdir -p ~/workspace/src
3) cd ~/workspace/src
4) wget https://raw.githubusercontent.com/ignition-tooling/gazebodistro/master/collection-garden.yaml
5) vcs import < collection-garden.yaml
6) brew update
7) brew tap osrf/simulation
8) brew install xquartz
9) brew install assimp boost bullet cmake cppzmq dartsim@6.10.0 doxygen eigen fcl ffmpeg flann freeimage freetype gflags google-benchmark gts ipopt irrlicht jsoncpp libccd libyaml libzzip libzip nlopt ode open-scene-graph ossp-uuid ogre1.9 ogre2.2 pkg-config protobuf qt qwt rapidjson ruby tbb tinyxml tinyxml2 urdfdom zeromq
10) 
```
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/homebrew/opt/dartsim@6.10.0

export DYLD_FALLBACK_LIBRARY_PATH=${DYLD_FALLBACK_LIBRARY_PATH}:/opt/homebrew/opt/dartsim@6.10.0/lib:/opt/homebrew/opt/octomap/

export PKG_CONFIG_PATH=${PKG_CONFIG_PATH}:/opt/homebrew/opt/dartsim@6.10.0/lib/pkgconfig

export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/homebrew/opt/qt@5
```
11) brew install gdal
12) $ cd ~/workspace/src/ign-common
$ git fetch
$ git pull
13) colcon graph 

The output: 
```
ignition-cmake2       + **************
ignition-tools         +       * . .**
ignition-plugin1        +     *  **.**
ignition-utils1          +**..***.*.**
ignition-math7            +****.******
ignition-common5           + ** ******
ignition-msgs9              +* * * ***
ignition-fuel_tools8         +      *.
ignition-rendering7           +  * **.
ignition-transport12           + * ***
sdformat13                      + ***.
ignition-gui7                    +  **
ignition-physics6                 + *.
ignition-sensors7                  +*.
ignition-gazebo7                    +*
ignition-launch6                     +
```
Lastly, `cd ~/workspace` then `colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib`

Returns error. :S 

```
Failed   <<< ignition-gui7 [4.55s, exited with code 2]
Aborted  <<< ignition-sensors7 [4.63s]                                   

Summary: 12 packages finished [27.6s]
  1 package failed: ignition-gui7
  1 package aborted: ignition-sensors7
  12 packages had stderr output: ignition-cmake2 ignition-common5 ignition-fuel_tools8 ignition-gui7 ignition-math7 ignition-msgs9 ignition-physics6 ignition-plugin1 ignition-rendering7 ignition-sensors7 ignition-transport12 ignition-utils1
  2 packages not processed
```

I tried this 2 times. Eventually, I noticed that X-code isn't installed by default on M1. I'm not sure how much it impacts on everything. 

SO after download the x-code, I re-do steps above and other message where I posted steps with cmake. Both of em didn't go through, still. The only thing it changed is how it halts the program really fast after I installed xcode.

my brain is fried so I'll try again in few hours or tomorrow. 

This is EXACTLY how I see M1 right now
[image]

This is different from linux, tbh. As for m1, there is a lot of stuff involves with pathing, library and setting. Linux do have it, but they are more accessible in some way? Maybe I'm wrong on this. I will try again tomorrow anyway 😎

iche033, 2022-03-28:
I think colcon should show some error msgs on why ign-gui failed.

Some differences I saw between your step 9 and mine are these libraries:

```
brew install qt@5 qwt-qt5 ffmpeg@4
```

The paths setup in step 10 looks the same as mine. I had one extra export which I'm not sure if it's actually needed

```
export PKG_CONFIG_PATH=/opt/homebrew/opt/ffmpeg@4/lib/pkgconfig
```

srmainwaring, 2022-03-29:
> I noticed that X-code isn't installed by default on M1

You will need Xcode and the matching Command line tools (which need to be installed from the Apple developer downloads pages).

It looks like you managed to get `ignition-rendering7` built. What was the outcome of building and running the examples `simple_demo` and `ogre2demo`?

Kakcalu13, 2022-03-29:
Good morning, gazebo engineers! Round 2 for me! 

> ```
> brew install qt@5 qwt-qt5 ffmpeg@4
> ```
> ```
> export PKG_CONFIG_PATH=/opt/homebrew/opt/ffmpeg@4/lib/pkgconfig
> ```

I think those made my colcon build go smooth without getting halt immediately. 

> I think colcon should show some error msgs on why ign-gui failed.

[colcon-buld-error.txt](https://github.com/ignitionrobotics/ign-gazebo/files/8371352/colcon-buld-error.txt)
Here is the full log.

As for the ign-gui:
it just says code2

```
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [include/ignition/gui/CMakeFiles/ignition-gui7.dir/__/__/
__/src/Helpers.cc.o] Error 1
make[1]: *** [include/ignition/gui/CMakeFiles/ignition-gui7.dir/all]
Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 10 jobserver tokens available; should be
8!
---
Failed <<< ignition-gui7 [11.0s, exited with code 2]
Aborted <<< ignition-sensors7 [1min 9s]
Aborted <<< ignition-physics6 [3min 9s]
Summary: 11 packages finished [10min 7s]
 1 package failed: ignition-gui7
```

it's on the last page of log

> It looks like you managed to get `ignition-rendering7` built. What was the outcome of building and running the examples `simple_demo` and `ogre2demo`?

Ngl, this got me excited for a second lol

```
station1@Station1s-Mac-mini ~ % cd workspace 
station1@Station1s-Mac-mini workspace % source ./install/setup.zsh
station1@Station1s-Mac-mini workspace % cd src/ign-rendering 
station1@Station1s-Mac-mini ign-rendering % cd examples 
station1@Station1s-Mac-mini examples % cd simple_demo
station1@Station1s-Mac-mini simple_demo % ls
CMakeLists.txt	GlutWindow.cc	GlutWindow.hh	Main.cc
station1@Station1s-Mac-mini simple_demo % mkdir build && cd build

station1@Station1s-Mac-mini build % cmake ..
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for ignition-rendering7 -- found version 7.0.0~pre1
-- Searching for dependencies of ignition-rendering7
-- Looking for ignition-math7 -- found version 7.0.0~pre1
-- Searching for dependencies of ignition-math7
[... 42 lines cut ...]
-- Build files have been written to: /Users/station1/workspace/src/ign-rendering/examples/simple_demo/build
station1@Station1s-Mac-mini build % make -j4
[ 66%] Building CXX object CMakeFiles/simple_demo.dir/GlutWindow.cc.o
[ 66%] Building CXX object CMakeFiles/simple_demo.dir/Main.cc.o
[100%] Linking CXX executable simple_demo
[100%] Built target simple_demo
station1@Station1s-Mac-mini build % ./simple_demo ogre2 metal
[Msg] Loading plugin [ignition-rendering-ogre2]
Error while loading the library [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib]: dlopen(/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib, 0x0005): Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsPbs.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreMain.2.2.6.dylib' (no such file), '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsUnlit.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreMain.2.2.6.dylib' (no such file), '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreHlmsUnlit.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreOverlay.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file)
[Err] [RenderEngineManager.cc:470] Failed to load plugin [ignition-rendering-ogre2] : couldn't load library on path [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib].
Engine 'ogre2' is not supported
[Msg] Loading plugin [optix]
[Err] [RenderEngineManager.cc:461] Failed to load plugin [optix] : couldn't find shared library.
Engine 'optix' is not supported
[Err] [GlutWindow.cc:178] No cameras found. Scene will not be rendered
station1@Station1s-Mac-mini build % 
```

I think we are really really close in ign-rendering tho

Kakcalu13, 2022-03-29:
This question came to my mind, which M1 did you have? M1 PRO, M1 max or just M1? I'm using just M1

srmainwaring, 2022-03-29:
@Kakcalu13 can you try exporting the following before running the `simple_demo`:

```bash
export OGRE2_RESOURCE_PATH=/opt/homebrew/opt/ogre2.2/lib/OGRE-2.2/OGRE
```

(double check I've got that path right). The failure is macOS not resolving an rpath. It's trying a couple of paths from `CMAKE_PREFIX_PATH` but none of them contain the correct location for the brew installed ogre2.2 libraries.

Kakcalu13, 2022-03-29:
I had to re-do the steps due to experiment and tried another ways.

The error persists. FYI, colcon build failed too. Just letting you know.

```
station1@Station1s-Mac-mini build % ./simple_demo ogre2 metal                                             
[Msg] Loading plugin [ignition-rendering-ogre2]
Error while loading the library [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib]: dlopen(/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib, 0x0005): Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsPbs.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreMain.2.2.6.dylib' (no such file), '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsUnlit.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreMain.2.2.6.dylib' (no such file), '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreHlmsUnlit.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreOverlay.2.2.6.dylib
  Reason: tried: '/Users/station1/workspace/install/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file)
[Err] [RenderEngineManager.cc:470] Failed to load plugin [ignition-rendering-ogre2] : couldn't load library on path [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib].
Engine 'ogre2' is not supported
[Msg] Loading plugin [optix]
[Err] [RenderEngineManager.cc:461] Failed to load plugin [optix] : couldn't find shared library.
Engine 'optix' is not supported
[Err] [GlutWindow.cc:178] No cameras found. Scene will not be rendered
station1@Station1s-Mac-mini build % 
```

I got this after I run this:
```
$ cd ~/workspace/src/ign-rendering/examples/simple_demo
$ mkdir build && cd build
$ cmake ..
$ make -j4
$ export OGRE2_RESOURCE_PATH=/opt/homebrew/opt/ogre2.2/lib/OGRE-2.2/OGRE
$ ./simple_demo ogre2 metal
```

Those error seems like they needed colcon build first. Any suggestion? 😅

Kakcalu13, 2022-04-01:
Hi @iche033!

Sorry for the late reply. 

So, apparently I already left it on using your new fix. See here:
[image]

I ran `brew install ogre2.2.rb`

It returns error

```
station1@Station1s-Mini Formula % brew install ogre2.2.rb 
Running `brew update --preinstall`...
==> Auto-updated Homebrew!
Updated 3 taps (osrf/simulation, homebrew/core and homebrew/cask).
==> New Formulae
cloudflare-quiche          dagger                     gi-docgen
==> Updated Formulae
Updated 434 formulae.
==> New Casks
lapce                                    simplemind
==> Updated Casks
Updated 261 casks.

Error: Failed to load cask: ogre2.2.rb
Cask 'ogre2.2' is unreadable: wrong constant name #<Class:0x000000014612a880>
Warning: Treating ogre2.2.rb as a formula.
Warning: osrf/simulation/ogre2.2 2.2.6+20211021~312bf40_1 is already installed and up-to-date.
To reinstall 2.2.6+20211021~312bf40_1, run:
  brew reinstall ogre2.2
station1@Station1s-Mini Formula % 
```

`-DCMAKE_MACOSX_RPATH=FALSE` is already inside my colcon build. I was following the steps in few messages previous and noticed this so I tried again using like 

`colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib -DCMAKE_MACOSX_RPATH=FALSE` and `cmake .. -DCMAKE_MACOSX_RPATH=FALSE`

First one returns the same thing but the second one...it's literally THIS close to open ign gazebo but then it crashed immediately.

The error log:
```
station1@Station1s-Mini build % ./simple_demo ogre2 metal
[Msg] Loading plugin [ignition-rendering-ogre2]
Error while loading the library [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib]: dlopen(/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib, 0x0005): Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsPbs.2.2.6.dylib
  Reason: tried: '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreMain.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreHlmsUnlit.2.2.6.dylib
  Reason: tried: '/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreMain.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreMain.2.2.6.dylib' (no such file)Library not loaded: @rpath/libOgreHlmsUnlit.2.2.6.dylib
  Referenced from: /opt/homebrew/Cellar/ogre2.2/2.2.6+20211021~312bf40_1/lib/libOgreOverlay.2.2.6.dylib
  Reason: tried: '/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/dartsim@6.10.0/lib/libOgreHlmsUnlit.2.2.6.dylib' (no such file), '/opt/homebrew/opt/octomap//libOgreHlmsUnlit.2.2.6.dylib' (no such file)
[Err] [RenderEngineManager.cc:470] Failed to load plugin [ignition-rendering-ogre2] : couldn't load library on path [/Users/station1/workspace/install/lib/ign-rendering-7/engine-plugins/libignition-rendering-ogre2.dylib].
Engine 'ogre2' is not supported
[Msg] Loading plugin [optix]
[Err] [RenderEngineManager.cc:461] Failed to load plugin [optix] : couldn't find shared library.
Engine 'optix' is not supported
[Err] [GlutWindow.cc:178] No cameras found. Scene will not be rendered
station1@Station1s-Mini build %
```
[image]

Kakcalu13, 2022-04-01:
WAIT!!!

just `./simple_demo` do the trick but i can't do anything to it at all though..
[image]

```
station1@Station1s-Mini build % ./simple_demo
[Msg] Loading plugin [ignition-rendering-ogre]
[Msg] Loading plugin [optix]
[Err] [RenderEngineManager.cc:461] Failed to load plugin [optix] : couldn't find shared library.
Engine 'optix' is not supported
===============================
  TAB - Switch render engines
  ESC - Exit
===============================
^C
station1@Station1s-Mini build %
```

Is this the one I was supposed to confirm? 

https://user-images.githubusercontent.com/65916520/161336227-7957f634-8b80-42af-b378-2cfc3041ed32.mov

iche033, 2022-04-02:
> -DCMAKE_MACOSX_RPATH=FALSE is already inside my colcon build.

yes you'll need that but what I mean is also adding it into ogre2.2.rb file here:
https://github.com/osrf/homebrew-simulation/blob/86e9a94dbbb3aac7e6d9ddd26343e5c8363f1072/Formula/ogre2.2.rb#L68

running `./simple_demo` launches the example with `ogre1.x` render engine. Good to see that it's working. On the other hand `./simple_demo ogre2 metal` uses `ogre2.2` which is the default rendering engine in ignition gazebo.

Kakcalu13, 2022-04-14:
Oops, I was looking for ignition to be on website and then I found this.

Let me test it again, gimme 10 minutes or less

Kakcalu13, 2022-04-14:
Seems like I already have it in my ogre2.2.rb.

See here:
[image]

I ran colcon build in the root of workspace (~/workspace)

See here:
[image]

`colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib`

which returns error.

See here:

[image]

was colcon build supposed to be in the root of workspace? Just wanted to confirm this. 

BTW I still would love to have ign on m1! I even dream about it last night! ;)

jasonbyun8, 2022-05-01:
@srmainwaring @iche033 
I got the ogre2 example working with my ignition-rendering7;
[image]
I am unable to compile the colcon build though as I ign-physics6 fails.

`CMake Error at /Users/lapyx/ign-workspace/build/ignition-physics6/test/FAKE_INSTALL-prefix/src/FAKE_INSTALL-stamp/FAKE_INSTALL-build-RelWithDebInfo.cmake:49 (message):
  Command failed: 2

   '/Applications/Xcode.app/Contents/Developer/usr/bin/make'

  See also

    /Users/lapyx/ign-workspace/build/ignition-physics6/test/FAKE_INSTALL-prefix/src/FAKE_INSTALL-stamp/FAKE_INSTALL-build-*.log

make[2]: *** [test/FAKE_INSTALL-prefix/src/FAKE_INSTALL-stamp/FAKE_INSTALL-build] Error 1
make[1]: *** [test/CMakeFiles/FAKE_INSTALL.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 12 jobserver tokens available; should be 10!
---
Failed   <<< ignition-physics6 [53.3s, exited with code 2]

Summary: 13 packages finished [7min 58s]
  1 package failed: ignition-physics6
  14 packages had stderr output: ignition-cmake3 ignition-common5 ignition-fuel_tools8 ignition-gui7 ignition-math7 ignition-msgs9 ignition-physics6 ignition-plugin2 ignition-rendering7 ignition-sensors7 ignition-tools2 ignition-transport12 ignition-utils2 sdformat13
  2 packages not processed
`
Have you experienced any issue like this before?

Below is the colcon graph I got from the latest garden yaml file;
`(base) ➜  ign-workspace colcon graph
ignition-cmake3       + **************
ignition-tools2        +      *   ..**
ignition-utils2         +**.*******.**
ignition-math7           +****.*******
ignition-common5          + ** *******
ignition-msgs9             +* *   ****
ignition-fuel_tools8        +       *.
ignition-plugin2             +  ***.**
ignition-transport12          +   ****
sdformat13                     +*  **.
ignition-physics6               +   *.
ignition-rendering7              +***.
ignition-gui7                     + **
ignition-sensors7                  +*.
ignition-gazebo7                    +*
ignition-launch6                     +
`

iche033, 2022-05-03:
I was able to reproduce the issue with ign-physics. Could be due to recent changes in Garden.

This patch is probably not a proper fix but it got ign-physics to build on my mac M1 machine:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0645982..fc958e8 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -63,6 +63,8 @@ ign_find_package(EIGEN3 REQUIRED)
 ign_find_package(sdformat13
   REQUIRED_BY sdf dartsim tpe bullet)
 
+unset(TINYXML2_LIBRARIES)
+
 #--------------------------------------
 # Find dartsim for the dartsim plugin wrapper
 ign_find_package(DART
```

jasonbyun8, 2022-05-03:
> I was able to reproduce the issue with ign-physics. Could be due to recent changes in Garden.
>
> This patch is probably not a proper fix but it got ign-physics to build on my mac M1 machine:
>
> ```diff [...]

That worked!

I had more issues to be addressed though

1. There was an no member name error in `src/ign-launch/src/vendor/backward.hpp` 
```
ign-garden-workspace/src/ign-launch/src/vendor/backward.hpp:3938:60: error: member reference type 'struct __darwin_mcontext64 *' is a pointer; did you mean to use '->'?
    error_addr = reinterpret_cast<void *>(uctx->uc_mcontext.pc);
                                          ~~~~~~~~~~~~~~~~~^
                                                           ->
ign-garden-workspace/src/ign-launch/src/vendor/backward.hpp:3938:61: error: no member named 'pc' in '__darwin_mcontext64'
    error_addr = reinterpret_cast<void *>(uctx->uc_mcontext.pc);
                                          ~~~~~~~~~~~~~~~~~ ^
2 errors generated.

```
I had to replace https://github.com/gazebosim/gz-launch/blob/8c78b6c937943ba1650e887c3a82dbfaea6c154f/src/vendor/backward.hpp#L3937

with 

```
#elif defined(__aarch64__) && !defined(__APPLE__)
     error_addr = reinterpret_cast<void *>(uctx->uc_mcontext.pc);
#elif defined(__APPLE__) && defined(__aarch64__)
    error_addr = reinterpret_cast<void *>(uctx->uc_mcontext->__ss.__pc);

```

2. After the complete build the ignition gazebo could not find the libzmq.5.dylib.
```
Library error for [/Users/jaeseungbyun/ign-garden-workspace/install/lib/libignition-gazebo7-ign.7.0.0~pre1.dylib]: dlopen(/Users/jaeseungbyun/ign-garden-workspace/install/lib/libignition-gazebo7-ign.7.0.0~pre1.dylib, 0x0009): Library not loaded: @rpath/libzmq.5.dylib
  Referenced from: /Users/jaeseungbyun/ign-garden-workspace/install/lib/libignition-transport12.12.0.0~pre1.dylib
  Reason: tried: '/usr/local/lib/libzmq.5.dylib' (no such file), '/usr/lib/libzmq.5.dylib' (no such file)
```

I had to create a symlink of `libzeq.5.dylib` @`/usr/local/lib/` for that reason.

iche033, 2022-05-05:
> There was an no member name error in src/ign-launch/src/vendor/backward.hpp

oh looks like that's also fixed upstream in the backward repo: https://github.com/bombela/backward-cpp/blob/master/backward.hpp#L4220-L4225.

tadteo, 2022-05-18:
Is this solved?
I'm having the same issues and I'm not able to run gazebo on docker in M1.

Error is always:
```
Error setting socket option (IP_MULTICAST_IF).
Error setting socket option (IP_MULTICAST_IF).
Did you set the environment variable IGN_IP with a correct IP address?
  [172.19.0.2] seems an invalid local IP address.
  Using 127.0.0.1 as hostname.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
qemu: uncaught target signal 6 (Aborted) - core dumped
```

Kakcalu13, 2022-05-18:
> Is this solved? I'm having the same issues and I'm not able to run gazebo on docker in M1.
>
> Error is always:
>
> ```
> Error setting socket option (IP_MULTICAST_IF).
> Error setting socket option [...]

Hi Tadteo!

Not yet, but you can start with Garden though. See here: https://github.com/osrf/homebrew-simulation/issues/1628#issuecomment-1116958513

I still haven't had a chance to test it. Hopefully soon for me! 

If it worked for you, feel free to post it. It's important to know that it worked for you so others can try it out too

Kakcalu13, 2022-05-25:
> Is this solved? I'm having the same issues and I'm not able to run gazebo on docker in M1.
>
> Error is always:
>
> ```
> Error setting socket option (IP_MULTICAST_IF).
> Error setting socket option [...]

After spend time with Gazebo on M1 using docker (Specifically using ubuntu 20.04 on docker)..I get the same output as above.

Full log:
```
Error setting socket option (IP_MULTICAST_IF).
Error setting socket option (IP_MULTICAST_IF).
Did you set the environment variable IGN_IP with a correct IP address? 
  [172.18.0.4] seems an invalid local IP address.
  Using 127.0.0.1 as hostname.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6886f, in rb_vm_exec
#30   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a62130, in 
#29   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a51405, in 
#28   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6adaa, in 
#27   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a5e025, in 
#26   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40019bc1be, in 
#25   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40018fc5d2, in rb_protect
#24   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a759f0, in rb_yield
#23   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6886f, in rb_vm_exec
#22   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a62130, in 
#21   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a51405, in 
#20   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6adaa, in 
#19   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a5e025, in 
#18   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x40055ca714, in 
#17   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a2f6d9, in rb_nogvl
#16   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x40055ca8fb, in 
#15   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x40055ef409, in 
#14   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x40055efff4, in 
#13   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo3-ign.so.3.12.0", at 0x40060527b3, in runServer
#12   Object "/lib/x86_64-linux-gnu/libignition-gazebo3.so.3", at 0x4006544622, in ignition::gazebo::v3::Server::Server(ignition::gazebo::v3::ServerConfig const&)
#11   Object "/lib/x86_64-linux-gnu/libignition-gazebo3.so.3", at 0x400654cd6a, in 
#10   Object "/lib/x86_64-linux-gnu/libignition-transport8.so.8", at 0x4006964eeb, in ignition::transport::v8::Node::Node(ignition::transport::v8::NodeOptions const&)
#9    Object "/lib/x86_64-linux-gnu/libignition-transport8.so.8", at 0x400694466e, in 
#8    Object "/lib/x86_64-linux-gnu/libignition-transport8.so.8", at 0x400697eaf4, in ignition::transport::v8::NodeShared::NodeShared()
#7    Object "/lib/x86_64-linux-gnu/libignition-transport8.so.8", at 0x400698dff9, in ignition::transport::v8::Discovery<ignition::transport::v8::MessagePublisher>::Discovery(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, bool)
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x40056b93aa, in 
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x40056c26a8, in __cxa_throw
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x40056c23f6, in std::terminate()
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x40056c238b, in 
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x40056b6910, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x4001bc5858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x4001be618b, in gsignal
Aborted (Signal sent by tkill() 1305 0)
qemu: uncaught target signal 6 (Aborted) - core dumped
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40018fe2ed, in ruby_run_node
#30   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40018f9490, in 
#29   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6886f, in rb_vm_exec
#28   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a62130, in 
#27   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a51405, in 
#26   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6adaa, in 
#25   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a5e025, in 
#24   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40019bc1be, in 
#23   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x40018fc5d2, in rb_protect
[... 3 lines cut ...]
#19   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a51405, in 
#18   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a6adaa, in 
#17   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a5e025, in 
#16   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x40055ca714, in 
#15   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x4001a2f6d9, in rb_nogvl
[... cut ...]
#2    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x400718baac, in QMessageLogger::fatal(char const*, ...) const
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x4001bc5858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x4001be618b, in gsignal
Aborted (Signal sent by tkill() 1307 0)
qemu: uncaught target signal 6 (Aborted) - core dumped
```

I assume building from gazebo on local inside M1 would find the root cause in Ubuntu 20.04 Docker and solve the issue right away, but this seems like it is not the case anymore. It's mostly from the Ruby issue.

srmainwaring, 2022-05-26:
@Kakcalu13 wondering how to best progress with the macOS help here. There seem to be two different objectives in this issue and it might help to focus on one of them.

1. Running Gazebo Sim natively on a M1 machine using Ogre2 and the Metal render system
2. Running Gazebo Sim using docker on a M1 machine and displaying the output on the host with hardware acceleration

For item 1, we know it works for Intel based macs on Big Sur 11.6.2, and most likely Monterey (although I have not verified the latter). I am fairly sure a couple of the OSRF team have built Garden for a M1 machine and managed to get it working. You may need to use a couple of open PRs for gz-gui and gz-sim to get all the features.

For item 2, I am not aware of how to get hardware acceleration working on a mac with Docker for applications requiring OpenGL 4.2+ (Gazebo Sim or otherwise). The docker based hardware support is for devices with Nvidia graphics cards (so not applicable to modern macs), alternatively you can use software rendering (but it is very slow, and it seems like a sub-optimal use of a M1 machine to do that). 
 
Which of 1 and 2 would you like to solve? If 1, we can probably help. If 2, then I think you need to verify that you have a method to get hardware accelerated GUI applications working with Docker on macOS, but would not recommend starting with Gazebo Sim, but choose a simpler application with fewer dependencies. I'd be very interested if you do get it working as I currently do not have a virtualised OpenGL 4.2+ environment for macOS and it would be extremely useful to have a solution for this.

Kakcalu13, 2022-05-26:
My apologies. I realize this does seem like it have two objectives.

This issue is for the item 2 only as the item 1 is on https://github.com/osrf/homebrew-simulation/issues/1628. That issue focus on Gazebo Sim natively only.

The reason why we were talking about Gazebo Sim natively on here because if we were able to build from the source then we can certainly build it on Docker using network mode host. Unfortunately, it doesn't seem like it can be the case due to multiple issues and heavy slow outside of Gazebo situation.

Would you like me to close this and create an issue to focus on docker and hardware accelerated GUI applications only? Maybe it can make things more clear? 

Side note, I will start with verify the method to get hardware accelerated GUI applications working with Docker on MacOS M1 now.

srmainwaring, 2022-05-26:
> Would you like me to close this and create an issue to focus on docker and hardware accelerated GUI applications only? Maybe it can make things more clear?

No - good to keep this open and thanks for clarifying.

> The reason why we were talking about Gazebo Sim natively on here because if we were able to build from the source then we can certainly build it on Docker using network mode host.

I'm missing something here. How does building Gazebo natively on macOS help with building Gazebo on an Ubuntu Docker machine and vice-versa? I can see how a network mode host would let you share IP addresses between applications running natively on macOS with those running in a docker container - but the toolchains for building are completely different?

> Side note, I will start with verify the method to get hardware accelerated GUI applications working with Docker on MacOS M1 now.

I really think this is the essential starting point. Because once you have established you can run OpenGL accelerated applications you can use one of the canned OSRF docker images and a nightly build of Gazebo Sim (Garden) for Ubuntu Jammy: the rest is docker configuration.

Kakcalu13, 2022-05-26:
> I'm missing something here. How does building Gazebo natively on macOS help with building Gazebo on an Ubuntu Docker machine and vice-versa? I can see how a network mode host would let you share IP [...]

Because there's somewhere missing with the package. Unlike other machine, this M1 integrated GPU and CPU in one same spot. So, the package somewhere MIGHT be denied. I came up several options to workaround:
1) Run M1 natively on docker instead of using ubuntu 20.04 image. (which is why we talked about running gazebo sim natively)
2) Run Ubuntu using a different arm (like amd64, arm64 or so) (This doesn't work well on m1)
3) Build from source from the step one (same as reason as first one) and run on Ubuntu Docker

The goal is to find whatever works on M1. Due to my lack of knowledge with how Gazebo are set up is also a severe factor to this. I thought that maybe if I know the toolchains for buildings on Ubuntu/M1, I would've found the solution despite if they are different. I hope this makes sense! 😓 

> I really think this is the essential starting point. Because once you have established you can run OpenGL accelerated applications you can use one of the canned OSRF docker images and a nightly build of Gazebo Sim (Garden) for Ubuntu Jammy: the rest is docker configuration.

I didn't know Garden on Docker is out now. Thanks!!! Still middle of OpenGL accelerated applications. I will post updates on here once I get it up

P.S.

if you have a LITTLE spare of time, do you know the answer for ign-gui on https://github.com/osrf/homebrew-simulation/issues/1628#issuecomment-1138561654? 👼 if no, all good!

srmainwaring, 2022-05-26:
> I didn't know Garden on Docker is out now.

I don't know about that. One of the ArduPilot devs used a nightly build of Garden on an Ubuntu Jammy machine, so I guess the packages are available if you wanted to roll your own docker file.

Kakcalu13, 2022-05-26:
Unfortunately, I'm really out of ideas. Maybe this is good time for me to take a break for a bit then resume this in few days. 

So, basically..I decided to check openGL on docker on ubuntu and ubuntu natively vs docker on m1 and m1 ubuntu.

Ubuntu natively:
`OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.2.6`

20.04 focal ubuntu on Ubuntu:
[image]

M1 natively:
[image]

20.04 focal ubuntu on Ubuntu:
[image]

So, this led me to believe that both are exactly same which as it should be! 

So I ran tests on M1/Ubuntu using `glmark2` in docker on those OS respectively. 

Both made the exact same result. See here:

https://user-images.githubusercontent.com/65916520/170559367-629c7c43-759d-4833-8f6d-f6eae31f335b.mov

As you can see that this is a simple test for acceleration GUI just fine.

So I tried the gui by git clone gui only and run examples from standalone/scene_provider. For some reason, it just didn't work with m1. 

```
root@3cdc9c63ce25:/root/Desktop/gz-gui-ign-gui6/examples/standalone/scene_provider/build# cmake ..
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findignition-msgs8.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-msgs8", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-msgs8"
  with any of the following names:

    ignition-msgs8Config.cmake
    ignition-msgs8-config.cmake

  Add the installation prefix of "ignition-msgs8" to CMAKE_PREFIX_PATH or set
  "ignition-msgs8_DIR" to a directory containing one of the above files.  If
  "ignition-msgs8" provides a separate development package or SDK, be sure it
  has been installed.

-- Configuring incomplete, errors occurred!
See also "/root/Desktop/gz-gui-ign-gui6/examples/standalone/scene_provider/build/CMakeFiles/CMakeOutput.log".
```

It wouldn't work if you install sudo apt lib ignition-msgs8 either. 

I could be wrong but I think maybe some packages are being denied during install..

Also,

According to Ruby issue I created this morning, Gazebo calls runGui in `libignition-gazebo3-ign.so.3.12.0` through FFI. They mentioned that the parameters for runGui might have been broken by fiddle + libffi. 

I need to look into somewhere to make sure everything in GUI is working properly. It's the GUI prevents the program to run. 

I know I've said this already, but I strongly believe that GUI is the issue at this point. It is just that it seems there's some missing packages within GUI. 

As for the Garden on docker. It sounds like it's all ready for Jammy, is that correct? DO you know if anyone with m1 managed to install it using Jammy? For some reason, I have no luck with Garden on docker either.

I honestly think my newbie knowledge on Mac contributes issue too. I'm a linux guy, I almost never use M1 until last February. Oh well, I will resume this later. 🖖

Kakcalu13, 2022-06-06:
It's officially working, everyone!!!

See here on M1 where I spinned my robot:

https://user-images.githubusercontent.com/65916520/172218164-31af000e-38e4-4da0-9574-aad3cbe7db9f.mov

ON M1! It's surprisingly faster and almost same speed as my local linux (it lags on docker on linux)

Here is the solution:
I found out that Gazebo can run arm64 so I switched from vnc (amd64 by default) to arm64 and I swapped amd64 to arm64 on ros2 for the FOXY ros2/gazebo bridge.

Here is what I used for docker:
https://hub.docker.com/r/dorowu/ubuntu-desktop-lxde-vnc/tags
(I use dockerfile so it is `FROM dorowu/ubuntu-desktop-lxde-vnc:focal-arm64`)

That's it!!! 

Looks like amd64 is **NOT** supported on Gazebo/M1.
Arm64 is available on M1 only. 

I believe we can build Gazebo using arm64 only

Thank you, everyone! This helped me a ton! Hope it helps you a lot too, guys! See you around in Gazebo community!

closed 2022-06-06 by Kakcalu13