gazebosim/gz-sim #1363: Support loading more than one visual plugin on GUI side

iche033, 2022-02-24:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
binary and source, ign-gazebo6

## Description

Visual plugin support on the GUI side is added in https://github.com/ignitionrobotics/ign-gazebo/pull/1275. However when multiple visual plugins are specified for a `<visual>`, only the first one will be loaded, see 
https://github.com/ignitionrobotics/ign-gazebo/pull/1335#issuecomment-1049728543

* Expected behavior: ign-gazebo's gui loads multiple visual plugin per visual
* Actual behavior: ign-gazebo's gui only loads the first visual plugin

This is mainly because we are only creating attaching one VisualPlugin component to a visual entity on the server side, which then get serialized and sent to the GUI side:
https://github.com/ignitionrobotics/ign-gazebo/blob/0dead0578ba3f26cf97f8162e55a0dbe1112484c/src/SdfEntityCreator.cc#L752-L761

A potential fix is to use component that stores a list of visual plugins instead

chapulina, 2022-03-16:
> A potential fix is to use component that stores a list of visual plugins instead

We'll need some way of passing system information from the server to the client for #19. I was thinking of creating a component that stores a [ignition::msgs::Plugin_V](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs8/proto/ignition/msgs/plugin_v.proto). Do you think that could address all the needs of the `VisualPlugin`, @iche033 ? I think that would be better to serialize than an `sdf::ElementPtr`.

iche033, 2022-03-16:
I think so. As long as contents of the plugin.proto msg can be converted back to an `sdf::ElementPtr` then it should be fine. For loading visual plugins, it comes down to this call on the GUI side:
https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/gui/GuiRunner.cc#L343-L344