gazebosim/gz-sim #1370: Rendering becomes very slow compared to ignition-edifice -- e.g. follow_actor.sdf and shapes_population.sdf

tmefmd, 2022-03-02:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
 Tried all source & binary with ignition gazebo 6.5, 6.6, 7.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. 
    - Rendering plugin: ogre2. [ign-gazebo_6.6.log](https://github.com/ignitionrobotics/ign-gazebo/files/8172306/ign-gazebo_6.6.log)
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - running on real hardware
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
      1a:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:2204] (rev a1)
      68:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:2204] (rev a1)

direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3090/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 470.103.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 470.103.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 470.103.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

root        1219  0.0  0.0 25358308 64096 tty1   Sl+  10:44   0:01 /usr/lib/xorg/Xorg vt1 -displayfd 3 -auth /run/user/125/gdm/Xauthority -background none -noreset -keeptty -verbose 3
root        1660  3.2  0.0 25377832 83412 tty2   Rl+  10:44   0:58 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
lapyx       7832  0.0  0.0   6440   660 pts/4    S+   11:13   0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn Xorg
 
X.Org X Server 1.20.13
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux lambda-dual 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.13.0-30-generic root=UUID=62f7664f-41fd-4df7-aaaa-2b16cca6cae3 ro quiet
Build Date: 14 December 2021  02:14:13PM
xorg-server 2:1.20.13-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.38.4
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

 [ogre2.log](https://github.com/ignitionrobotics/ign-gazebo/files/8172328/ogre2.log)

## Description
* Expected behavior:
1.  If I run shapes_population.sdf (with 3000 objects) with ignition gazebo 5.3, the rendering is very fast and I can see immediate actions without a delay.
2. If I run follow_actors.sdf and move one of the modes with ignition gazebo 5.3, I am able to see the movement of the actors.
* Actual behavior: 
1. If you run the shapes_population.sdf (with 3000 objects) from ignition gazebo higher than 6, the rendering takes extreme amount of time. It takes about 10 mins to be rendered, and if you relocate one of the objects it will take additional 10 mins to be updated.
2. If I run follow_actors.sdf and move one of the modes with ignition gazebo higher than 6, then the scene freezes.

## Steps to reproduce

1. You can just install any gazebo version higher than 6.0, and compare the result with 5.3.

## Output

tmefmd, 2022-03-02:
I tested the same thing with 3 different linux machines but the same focal 20.04 OS, and none of these were working

iche033, 2022-03-03:
I think this is related to https://github.com/ignitionrobotics/ign-gazebo/issues/1358

I tested the patch from https://github.com/ignitionrobotics/ign-gazebo/issues/1358#issuecomment-1057416590 and it no longer freezes

tmefmd, 2022-03-03:
This actually worked!
Although everything works, now I am facing with an error 

[Err] [EntityComponentManager.cc:1010] Failed to create component of type [0] for entity [11092]. Type has not been properly registered.

Is this also related to the patch you suggested? @iche033

iche033, 2022-03-03:
It could be related. The patch in that comment disables serialization to help pin out the cause of the performance issue. We'll need to figure out a proper fix.

tmefmd, 2022-03-03:
Ok, in that case I will keep this issue open until there is a proper fix for it, thanks for your help!

chapulina, 2022-03-07:
I'll close this issue in favor of #1358. Please reopen if it turns out that this is a different issue.