gazebosim/gz-sim #1376: Gazebo crashes in specific worlds

harshmahesheka, 2022-03-05:
Environment
* OS Version:Ubuntu 20.04 
* Source or binary build:Tried Fortress source and binary and Citadel source
 Rendering plugin: [ogre2]- [ogre2.log](https://github.com/ignitionrobotics/ign-gazebo/files/8190528/ogre2.log)
   
 Running on a dual GPU machine (integrated GPU(Radeon™ Vega graphics) + discrete GPU(VIDIA® GeForce® GTX 1650 (4 GB GDDR6 dedicated)
Rendering system info: https://user-images.githubusercontent.com/78502324/156880698-327a3da2-bbf5-443b-8dad-413a81b94e2d.png

Description-
I source build igniton fortress and it works fine in simple worlds but crashes with models and complex worlds like auv_controls.sdf.I also tried citadel source build and faced same error.Then I tried binary build with docker and it worked but when I source something from my main os into docker and then try to run same world crashes(same error).I thought it could be clashing with some of my libraries hence I reinstalled Ubuntu and tried it in clean OS but faced same error.I have been facing this error from a good many days and I tried asking it on http://answers.gazebosim.org  but got no reply. Any help will be much appreciated 

Steps to reproduce
```  ign gazebo auv_controls.sdf ```
Output-
[image]

Complete Log- 
[ign gazebo auv_controls.sdf -v4.log](https://github.com/ignitionrobotics/ign-gazebo/files/8190532/ign.gazebo.auv_controls.sdf.-v4.log)

arjo129, 2022-03-08:
Hey Harsh which version of GCC are you using. This looks like an issue with the verable `std::filesystem` . Looks similar to ://github.com/ignition-tooling/release-tools/issues/639

harshmahesheka, 2022-03-09:
Hey, @arjo129 I am using gcc (Ubuntu 8.4.0-3ubuntu2) 8.4.0.
Also I tried changes you did in https://github.com/ignitionrobotics/ign-gazebo/pull/1346 and https://github.com/ignitionrobotics/ign-common/pull/312 but insted of  https://github.com/ignitionrobotics/ign-common/blob/9d0b0683030b96b7042d46de491aa1171693a9ed/src/Filesystem.cc#L77 I used 
```
 if (ignition::common::absPath(_path)==_path)
   return false;
 else 
   return true; 
```
as std::filesystem is not used here in ignition fortress and everything works.Thanks for the help

arjo129, 2022-03-09:
You might want to use GCC 9 for everything then the `std::filesystem` issue should go away without code changes.

closed 2022-03-09 by arjo129