gazebosim/gz-sim #1378: Pending TODOs Ignition Garden

ahcorde, 2022-03-08:
List of TODOs to review before Garden release
 -  /// \todo(anyone) Move to GuiRunner::Implementation when porting to v5
    - https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo7/src/gui/GuiRunner.hh#L74
 - // \todo(anyone) Change this to uint64_t in Ignition H?
    - https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/systems/shader_param/ShaderParam.cc#L288
 - / \todo(anyone) instead of using ifdef to check for macOS, expose add an accessor function to get the GraphicsApi from rendering::RenderEngine
    - https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/systems/shader_param/ShaderParam.cc#L317-L320
 - // \todo handle args for constants in ign-rendering
    - https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/systems/shader_param/ShaderParam.cc#L365

chapulina, 2022-03-11:
We can also update our context menu:

https://github.com/ignitionrobotics/ign-gazebo/blob/fc9cadd680e1414b38be71b404958aa8e16ff70d/src/gui/plugins/modules/EntityContextMenu.qml#L52-L73

ahcorde, 2022-03-28:
- Use the field is_light_off in light.proto from Garden on.
 - Use the field visualize_visual in light.proto from Garden on.

jennuine, 2022-08-30:
>  /// \todo(anyone) Move to GuiRunner::Implementation when porting to v5
The `todo` comment will be removed in https://github.com/gazebosim/gz-sim/pull/1678

azeey, 2022-09-21:
Looks like the first item is done as @jennuine mentioned. I don't think we have time to handle the rest before the Garden release.

iche033, 2022-09-21:
the 3rd todo should be addressed by https://github.com/gazebosim/gz-sim/pull/1713

azeey, 2022-10-26:
I think only the last item is left. @iche033 Can we close this issue and create a separate issue for item 4?

iche033, 2022-11-07:
> I think only the last item is left. @iche033 Can we close this issue and create a separate issue for item 4?

yes that sounds good to me