gazebosim/gz-sim #1389: BuoyancyPlugin: graded_buoyancy does strange things with cylinders

clydemcqueen, 2022-03-13:
## Environment
* OS Version: Ubuntu 20.04
* Source build, last commit is fc9cadd680e1414b38be71b404958aa8e16ff70d
  
  

## Description
* Expected behavior: a model consisting of 1 neutrally buoyant cylinder should be neutrally buoyant when using graded_buoyancy
* Actual behavior: with uniform density the cylinder doesn't move (it is neutrally buoyant as expected), but with graded buoyancy the cylinder drops

## Steps to reproduce

This cylinder is neutrally buoyant (0.5^2 * pi * 1 * 1000 = 785.398163397), so doesn't move:

~~~
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="cylinder">
    <plugin
        filename="ignition-gazebo-physics-system"
        name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
        filename="ignition-gazebo-user-commands-system"
        name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
        filename="ignition-gazebo-scene-broadcaster-system"
        name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <plugin
        filename="ignition-gazebo-buoyancy-system"
        name="ignition::gazebo::systems::Buoyancy">
      <uniform_fluid_density>1000</uniform_fluid_density>
    </plugin>
    <model name='cyl_neutral_buoyancy'>
      <pose>0 0 -3 0 0 0</pose>
      <link name='cyl'>
        <inertial>
          <mass>785.398163397</mass>
          <pose>0 0 0 0 0 0</pose>
          <inertia>
            <ixx>86.28907821859966</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>86.28907821859966</iyy>
            <iyz>0</iyz>
            <izz>5.026548245743671</izz>
          </inertia>
        </inertial>
        <visual name='cyl_visual'>
          <geometry>
            <cylinder>
              <radius>0.5</radius>
              <length>1</length>
            </cylinder>
          </geometry>
        </visual>
        <collision name='cyl_collision'>
          <geometry>
            <cylinder>
              <radius>0.5</radius>
              <length>1</length>
            </cylinder>
          </geometry>
        </collision>
      </link>
    </model>
  </world>
</sdf>
~~~

If you replace `<uniform_fluid_density>1000</uniform_fluid_density>` with this section:

~~~
      <graded_buoyancy>
        <default_density>1000</default_density>
        <density_change>
          <above_depth>0</above_depth>
          <density>1</density>
        </density_change>
      </graded_buoyancy>
~~~

Then the cylinder drops like a stone.

clydemcqueen, 2022-03-13:
Ah, I see from Buoyancy.cc that only boxes and spheres are supported, not cylinders. Not a bug. Closing.

closed 2022-03-13 by clydemcqueen

chapulina, 2022-03-14:
To provide more context, to add cylinder support we'd need to add the `Cylinder::CenterOfVolumeBelow` function. It was added to `Box` and `Sphere` in

* https://github.com/ignitionrobotics/ign-math/pull/219