gazebosim/gz-sim #1409: Doxygen segfaults on ign-gazebo6 and ign-gazebo7

j-rivero, 2022-03-24:
## Environment
* OS Version: Ubuntu Jammy
* Source or binary build? Building packages
    
<details>

https://build.osrfoundation.org/job/ign-gazebo7-debbuilder/271/console
```
warning: Tag 'TCL_SUBST' at line 237 of file '/home/jenkins/workspace/ign-gazebo7-debbuilder/build/ign-gazebo/obj-x86_64-linux-gnu/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 1044 of file '/home/jenkins/workspace/ign-gazebo7-debbuilder/build/ign-gazebo/obj-x86_64-linux-gnu/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
make[3]: *** [CMakeFiles/doc.dir/build.make:75: CMakeFiles/doc] Segmentation fault (core dumped)
```

</details>

## Description

While building packages, the building job in build.osrfoundation.org was unable to run doxygen properly. It always failed with a segfault. The cause could be a huge consumption of memory or other problem with doxygen internals. There are warnings about deprecated behaviour that probably need to be addressed in ign-cmake.

chapulina, 2022-03-24:
Our doc upload jobs have been passing for `ign-gazebo6` using BIonic:

https://github.com/ignitionrobotics/docs/runs/5671770965?check_suite_focus=true

j-rivero, 2022-04-05:
I've been looking into the problem. Some findings:

gdb run:
```gdb
0x00005580dc0647ec in toDefinitionMutable (d=0x5580e1e7c410) at ./src/definition.cpp:2209
2209	./src/definition.cpp: No such file or directory.
(gdb) bt
#0  0x00005580dc0647ec in toDefinitionMutable (d=0x5580e1e7c410) at ./src/definition.cpp:2209
#1  0x00005580dbc0e67f in buildNamespaceList (root=0x5580df972780) at ./src/doxygen.cpp:1644
#2  0x00005580dbc0e0aa in buildNamespaceList (root=<optimized out>) at ./src/doxygen.cpp:1685
#3  0x00005580dbc0e0aa in buildNamespaceList (root=<optimized out>) at ./src/doxygen.cpp:1685
#4  0x00005580dbc0e0aa in buildNamespaceList (root=<optimized out>) at ./src/doxygen.cpp:1685
#5  0x00005580dbc0e0aa in buildNamespaceList (root=<optimized out>) at ./src/doxygen.cpp:1685
#6  0x00005580dbc0e0aa in buildNamespaceList (root=<optimized out>) at ./src/doxygen.cpp:1685
#7  0x00005580dbc3c1bb in parseInput () at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#8  0x00005580dbbdc1fe in main (argc=2, argv=0x7ffec49eb288) at ./src/main.cpp:37
```
gdb breakpoints and inspections:
```gdb
Breakpoint 9, toDefinitionMutable (d=0x56336968dbd8) at ./src/definition.cpp:2209
2209	in ./src/definition.cpp
(gdb) print d->getReference()
$33 = {m_rep = ""}
(gdb) print d->displayName(TRUE)
$34 = {m_rep = "ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::systems"}
(gdb) print d->name()
$35 = {m_rep = "ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::systems"}
(gdb) c
Continuing.

Breakpoint 9, toDefinitionMutable (d=0x56336968dbd8) at ./src/definition.cpp:2209
2209	in ./src/definition.cpp
(gdb) print d->name()
$36 = {m_rep = "ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::systems"}
(gdb) c
Continuing.

Breakpoint 9, toDefinitionMutable (d=0x56336b7f6410) at ./src/definition.cpp:2209
2209	in ./src/definition.cpp
(gdb) print d->name()
Cannot access memory at address 0x563608b644c6
...

Program received signal SIGSEGV, Segmentation fault.
```

doxygen -d preprocessor api.dox
```
... snip
Preprocessor output of /home/jrivero/code/ignition/ign-gazebo/src/gui/plugins/visualize_lidar/VisualizeLidar.hh (size: 4342 bytes):
---------
00001 /*
00002  * Copyright (C) 2020 Open Source Robotics Foundation
00003  *
00004  * Licensed under the Apache License, Version 2.0 (the "License");
00005  * you may not use this file except in compliance with the License.
00006  * You may obtain a copy of the License at
00007  *
00008  *     http://www.apache.org/licenses/LICENSE-2.0
00009  *
00010  * Unless required by applicable law or agreed to in writing, software
00011  * distributed under the License is distributed on an "AS IS" BASIS,
00012  * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
00013  * See the License for the specific language governing permissions and
00014  * limitations under the License.
00015  *
00016 */
00017 
00018 
00019 
[... 108 lines cut ...]
00128     signals: void MaxRangeChanged();
00129 
00130     /// \brief Get the maximum range of the lidar sensor (in metres).
00131     /// \return Range, the maximum distance sensed by the sensor.
00132     public: Q_INVOKABLE QString MaxRange() const;
00133 
00134     /// \brief Get the minimum range of the lidar sensor (in metres).
00135     /// \return Range, the minimum distance sensed by the sensor.
00136     public: Q_INVOKABLE QString MinRange() const;
00137 
00138     /// \internal
00139     /// \brief Pointer to private data
00140     private: std::unique_ptr<VisualizeLidarPrivate> dataPtr;
00141   };
00142 }
00143 }
00144 }
00145 
00146 
---------
No macros accessible in this file (/home/jrivero/code/ignition/ign-gazebo/src/gui/plugins/visualize_lidar/VisualizeLidar.hh).
Segmentation fault (core dumped)
```

Hard for me to know what is exactly going on and where. The good news is that using doxygen 1.9.3 statically linked from the doxygen page does not suffer from the problem and works just fine. Unless someone want to spend a bit more of time looking for a workaround in the comments affecting doxygen, I feel like we should workaround the problem by using a newer version of doxygen or not using it at all.

chapulina, 2022-04-06:
> we should workaround the problem by using a newer version of doxygen or not using it at all.

+1 for forcing a newer version if it solves the issue. We're using doxygen to generate our docs here:

https://github.com/ignitionrobotics/docs/blob/master/tools/build_docs.sh

And to run linters here:

https://github.com/ignition-tooling/action-ignition-ci/blob/ca2cabaf9565ba0e8850a7b6409323441d7e7bd4/entrypoint.sh#L151-L156

chapulina, 2022-07-13:
It looks like the problem is with `ADDITIONAL_INPUT_DIRS`, commenting out this line fixes compilation on Jammy:

https://github.com/gazebosim/gz-sim/blob/f39bc25da4a93a0b2ccf5fca8c3e1b6863185a9d/CMakeLists.txt#L242

I suspect this happens because Jammy uses a newer version of doxygen. I started updating the doxygen file from Bionic to Focal in

* https://github.com/gazebosim/gz-cmake/pull/276

Interestingly, `gz-gui` [also uses](https://github.com/gazebosim/gz-gui/blob/2ba68b8cdbff9dd01e4aead7d47835008d1d01ca/CMakeLists.txt#L111) `ADDITIONAL_INPUT_DIRS` and it doesn't fail. The issue could be that we're passing more than one directory.

chapulina, 2022-08-11:
Here's the fix:

* https://github.com/gazebosim/gz-sim/pull/1643

Only after opening the PR I noticed that the fix had already been mentioned in 

* https://github.com/gazebosim/gz-sim/issues/520#issuecomment-752407762

Doxygen's newer versions are really sensitive to those inline namespaces

closed 2022-08-17 by chapulina