gazebosim/gz-sim #1414: SceneBroadcaster's _hertz options crash with doubles

chapulina, 2022-03-25:
## Environment
* OS Version:  Ubuntu 20.02
* Source or binary build? either
  
  

## Description
* Expected behavior:  I can pass double values to `<state_hertz>` and `<dynamic_state_hertz>`, like `0.01`
* Actual behavior:  A crash

## Steps to reproduce

Load a world that has a `< 0` double value for the frequency, for example

```
      <plugin
        filename="ignition-gazebo-scene-broadcaster-system"
        name="ignition::gazebo::systems::SceneBroadcaster">
        <state_hertz>0.01</state_hertz>
      </plugin> 
```

## Output

Backtrace:

```
Thread 1 "INTEGRATION_sce" received signal SIGFPE, Arithmetic exception.                                                                 
0x00007fc1f83c4e04 in ignition::gazebo::v6::systems::SceneBroadcaster::Configure (this=0x563b60e176e0, _entity=<optimized out>, _sdf=..., 
    _ecm=...) at /usr/include/c++/9/chrono:331                                                                                              331               constexpr explicit duration(const _Rep2& __rep)                                                                           
(gdb) bt                                                                                                                                    
#0  0x00007fc1f83c4e04 in ignition::gazebo::v6::systems::SceneBroadcaster::Configure (this=0x563b60e176e0, _entity=<optimized out>, 
    _sdf=..., _ecm=...) at /usr/include/c++/9/chrono:331  
```

Thodoris1999, 2022-03-26:
Hello, it seems to be caused by [stateHerz](https://github.com/ignitionrobotics/ign-gazebo/blob/9b07b245e8384628fa20d09a563c48a6a063c0c0/src/systems/scene_broadcaster/SceneBroadcaster.cc#L267) being read as an integer from the SDF, which rounds values between 0 and 1 to 0 and then being reciprocated to get the period duration

A couple of ways to fix this:
Keep representing the update period as integer milliseconds but avoid crashes by rounding to and integer between 1 and 1000 (frequency over 1000 causes the value entered here `std::chrono::milliseconds(1000/stateHerz.first)` to be zero since `std::chrono::milliseconds` which can be unexpected. This maximum can be increased by using a smaller chono::duration period than milliseconds)

Represent the update period with a double precision tick like this `this->dataPtr->statePublishPeriod[false] = std::chrono::duration<double, std::ratio<1, 1000>>(1000/stateHerz.first);`. No frequency precision is lost (up to double accuracy). Only check for non-negative given frequency.

chapulina, 2022-03-29:
> Also, dynamic_pose_hertz can be rounded down to zero which is probably unexpected but it is converted to integer in transport, should [msgsPerSec](https://github.com/ignitionrobotics/ign-transport/blob/ign-transport11/src/AdvertiseOptions.cc#L59) be changed to double as well?

Ah nice catch. I can't think of a reason to keep that as an integer, what do you think @caguero ?

PR #1417 (2022-05-04): SceneBroadcaster: Use double for state publish frequency instead of int
…

# 🦟 Bug fix

Fixes #1414 

## Summary
Changed the `state_hertz` publishing frequency data type from `int` to `double`. Previously, worlds with those params set between 0 and 1 would crash due to being rounded to 0 and `state_hertz` being inverted. Note that dynamic_pose_hertz still gets rounded down to 0 as it is represented as `int` in `AdvertiseOptions`, but does not cause a crash.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as [... cut]
closed 2022-05-04 by chapulina