gazebosim/gz-sim #1434: Mecanum drive is not working

olmerg, 2022-04-09:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
*binary fortress
 
The mecanum drive in v_y and w_z are not working as expected, I tested with https://app.ignitionrobotics.org/balookaloo/fuel/worlds/Edifice%20demo and with [test example](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/examples/worlds/mecanum_drive.sdf )

scpeters, 2022-04-18:
if this is related to the version of dart, then I believe it is a duplicate of #880

[Ubuntu 20.04 has dart 6.9.2](https://packages.ubuntu.com/focal/libdart-all-dev) as an upstream package, which does not have the APIs needed to support Mecanum wheels. You can try building dart from source or using the [Dartsim PPA](https://launchpad.net/~dartsim/+archive/ubuntu/ppa)

if you are currently using dart 6.9, then let's move any further discussion to #880

scpeters, 2022-04-26:
I'm marking this as a duplicate of #880; please re-open if you disagree

closed 2022-04-26 by scpeters