gazebosim/gz-sim #1465: During build, undefined references to protobuf

MyGithubNotYours, 2022-04-30:
I'm trying to build ign-gazebo, but it looks like I'm having trouble with protobuf. At first, the ign-gazebo building was giving me this error:

- error: google/protobuf/port_def.inc: No such file or directory
- using protobuf 3.6 from apt-get install protobuf

I figured that maybe protobuf 3.6 was too old, so I switched to the newest version - protobuf v3.20. Now, I'm getting these errors about Print. What's the solution?

From CMakeLists.txt in ign-msgs:
   --> PROTOBUF LIB: /usr/local/lib/libprotobuf.a;-lpthread
   --> PROTOBUF INCLUDE: /usr/local/include
   --> PROTOBUF VERSION: 3.20.1
   --> PROTOBUF Found: TRUE

```
$ colcon build --merge-install --cmake-clean-cache 
Starting >>> ignition-cmake3
Starting >>> ignition-tools2
Finished <<< ignition-tools2 [0.67s]                                                            
Finished <<< ignition-cmake3 [0.72s]                   
Starting >>> ignition-utils2
Finished <<< ignition-utils2 [2.70s]                     
Starting >>> ignition-math7
Finished <<< ignition-math7 [3.71s]                      
Starting >>> ignition-common5
Starting >>> ignition-msgs9
--- stderr: ignition-msgs9                                                                          

/usr/bin/ld: CMakeFiles/ign_msgs_gen.dir/Generator.cc.o: in function `google::protobuf::compiler::cpp::Generator::Generate(google::protobuf::FileDescriptor const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::compiler::GeneratorContext*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) const':
/home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:91: undefined reference to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:92: undefined reference to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:93: undefined reference to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:94: undefined reference to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:95: undefined reference to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: CMakeFiles/ign_msgs_gen.dir/Generator.cc.o:/home/notyou/workspace-ign-gazebo/ign-msgs/src/Generator.cc:97: more undefined references to `google::protobuf::io::Printer::Print(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' follow
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/ign_msgs_gen.dir/build.make:101: bin/ign_msgs_gen] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1236: src/CMakeFiles/ign_msgs_gen.dir/all] Error 2
gmake: *** [Makefile:163: all] Error 2
---
Failed   <<< ignition-msgs9 [3.13s, exited with code 2]
Aborted  <<< ignition-common5 [3.64s]                             

Summary: 4 packages finished [10.9s]
  1 package failed: ignition-msgs9
  1 package aborted: ignition-common5
  1 package had stderr output: ignition-msgs9
  10 packages not processed

```

chapulina, 2022-05-05:
I see you're compiling Garden. That will be [officially supported on Ubuntu Focal and Jammy](https://gazebosim.org/docs/garden/install#supported-platforms). They have protobuf versions 3.6.1 and 3.12.4 respectively.

We're willing to accept pull requests improving support for 3.20, as long as that doesn't break the lower versions.

---

⚠️ Be sure to fill the issue template next time. It makes the maintainers' lives harder if we have to deduce information about your system from posted logs.

chapulina, 2022-05-05:
I should also mention that the issue you're seeing is coming from the messages library, not this repository.

https://github.com/ignitionrobotics/ign-msgs/