gazebosim/gz-sim #1473: Problems when adding lidar and camera sensors to the same robot on Ignition

lucaslbmp, 2022-05-04:
## Environment
* OS Version: 
* Source or binary build?: Binary, igntion-gazebo-6.2.0

## Description
Trying to spawn a simple 2-wheeled robot model (named "r2d2" here) in Ignition containing both lidar and camera sensor.

* Expected behavior: The model is spawned and their respective topics are available, just what happens if only one of the sensors is spawned.
* Actual behavior: The simulator crashes and the following error is displayed:

## Steps to reproduce

1.  In the main xacro of a model, add <sensor> tags for each sensor

```
<gazebo reference="cam_front">
        <sensor name='${name}' type='camera'>
            <pose>${body_dim_x/2} 0 ${body_dim_z/2} 0 0 0</pose>
            <visualize>true</visualize>
            <update_rate>5</update_rate>
            <topic>${name}/camera_image</topic>
            <camera>
                <!-- <horizontal_fov>1.047</horizontal_fov> -->
                <image>
                    <!-- <width>320</width>
                    <height>240</height> -->
                    <width>640</width>
                    <height>420</height>
                    <!-- <format>B8G8R8</format> -->
                </image>
                <!-- <clip>
                    <near>0.1</near>
                    <far>100</far>
                </clip> -->
            </camera>

            <plugin filename="libignition-gazebo-camera-video-recorder-system.so" name="ignition::gazebo::systems::CameraVideoRecorder">
            </plugin>
        </sensor>
    </gazebo>
```

`<gazebo reference="base_lidar">
        <sensor name="${name}" type='gpu_lidar'>
            <pose>0 0 ${body_height*0.6} 0 0 0</pose>
            
            <topic>${name}/lidar</topic>
            <update_rate>10</update_rate>
            <ray>
                <scan>
                    <horizontal>
                        
                        <samples>10</samples>
                        <resolution>1</resolution>
                        <min_angle>${-radians(20)}</min_angle>
                        <max_angle>${radians(20)}</max_angle>
                    </horizontal>
                    <vertical>
                        <samples>10</samples>
                        <resolution>1</resolution>
                        <min_angle>${-radians(10)}</min_angle>
                        <max_angle>${radians(10)}</max_angle>
                    </vertical>
                </scan>
                <range>
                    
                    <min>0.1</min>
                    <max>2.0</max>
                    <resolution>0.01</resolution>
                </range>
            </ray>
            <always_on>1</always_on>
            <visualize>true</visualize>

            <plugin filename="libignition-gazebo-sensors-system.so" name="ignition::gazebo::systems::Sensors">
                <render_engine>ogre</render_engine>
            </plugin>
        </sensor>
    </gazebo>`

2. Launch the simulator: `ign gazebo empty.sdf --render-engine ogre`
3. Spawn the robot: ign service -s /world/empty/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 300 --req 'sdf_filename: "/home/lucas/ros2_ntu_ws/src/ntu_gazebo/models/teste_bot_v2_camera/urdf/teste_bot_v2_camera.urdf" '

## Output

The simulator crashes and the following message is displayed:

[image]

lucaslbmp, 2022-05-06:
I found out the problem: the plugin "ignition::gazebo::systems::Sensors" should be called only once, even though it is used by both sensors.

closed 2022-05-06 by lucaslbmp

chapulina, 2022-05-06:
> I found out the problem: the plugin "ignition::gazebo::systems::Sensors" should be called only once, even though it is used by both sensors.

Glad to hear it! It would be interesting to add some safety check to that plugin that prevents 2 instantiations.

msminirobot, 2023-12-18:
This is actually also very important for multi-robot simulations.