gazebosim/gz-sim #1476: Some ranges of lidar sensor return always 0.0

lucaslbmp, 2022-05-08:
## Environment
* OS Version: 
* Source or binary build?: Binary, Ignition Fortress (Gazebo 6.8.0)
  
  

## Description
I am spawning a model (teste_bot) containing a lidar sensor and echoing the lidar topic to check the ranges returned.
* Expected behavior: In the absence of any obstacle in the range of detection of the sensor, it is expected that all ranges return inf. In case of any detection, the range value should be a value between the minimum range and the maximum range.
* Actual behavior: All ranges return inf except the ones corresponding to the last (upper) horizontal layer and to the last ray in each horizontal layer, which return 0.0 invariably. It should be impossible even if any ray collision happened, since the values of ranges returned by the sensor should not be below the minimum range parameter, which is not zero in this case.

## Steps to reproduce

1. Add a sensor component to the xacro file of any model, using the plugin "ignition::gazebo::systems::Sensors":

```
<gazebo reference="base_lidar">
        <sensor name="${name}" type='gpu_lidar'>
            <pose>0 0 ${body_height*0.6} 0 0 0</pose>
            <topic>${name}/lidar</topic>
            <update_rate>10</update_rate>
            <ray>
                <scan>
                    <horizontal>
                        <samples>5</samples>
                        <min_angle>${-radians(20)}</min_angle>
                        <max_angle>${radians(20)}</max_angle>
                    </horizontal>
                    <vertical>
                        <samples>5</samples>
                        <min_angle>${-radians(10)}</min_angle>
                        <max_angle>${radians(10)}</max_angle>
                    </vertical>
                </scan>
                <range>
                    <min>0.1</min>
                    <max>2.0</max>
                    <resolution>0.01</resolution>
                </range>
            </ray>
            <always_on>1</always_on>
            <visualize>true</visualize>

            <plugin filename="libignition-gazebo-sensors-system.so" name="ignition::gazebo::systems::Sensors">
                <render_engine>ogre</render_engine>
            </plugin>
        </sensor>
    </gazebo>
```

2. Build the workspace: `colcon build`
3. Launch the simulator: `ign gazebo empty.sdf --render-engine ogre`
4. Spawn the robot: `ign service -s /world/empty/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 300 --req 'sdf_filename: "/home/lucas/ros2_ntu_ws/src/ntu_gazebo/models/teste_bot/urdf/teste_bot.urdf" '`

## Output

[image]

A screenshot of the point cloud of the lidar sensor, which was configured to have 5 samples in the <horizontal> layer and 5 samples in the <vertical> layer. The point cloud was generated using the "Visualize Lidar" plugin. It is possible to see that the distribuition becomes asymetric and the points corresponding to the "zeroed" ranges can be seen in the origin of the scan.

Could someone help me clarify why is this happening?

Gratefully,

Lucas

mjcarroll, 2022-06-15:
@lucaslbmp friendly ping.

mjcarroll, 2022-10-14:
@lucaslbmp I'm going to close this as I haven't heard back from you.  If you have any other problems, feel free to either re-open this issue or open a new issue.

closed 2022-10-14 by mjcarroll

jetag, 2023-11-03:
For some reason this doesn't work when I am trying to run it via a ros node. The camera and other sensors work fine but the lidar ends up giving a 0 reading in range, no matter what. On a stand alone gazebo instance it works perfectly well. Any thoughts?