gazebosim/gz-sim #1492: Gazebo Ignition fails to render scene in Ubuntu VM (using ogre2 = black screen) (using ogre = flickering of scene 1Hz)

marc-wittwer, 2022-05-17:
## Environment
* OS Version: Ubuntu 20.04 in VM (Host-Machine is running Windows)
* Binary build: Ignition Gazebo, version 6.9.0

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: 
    When using ign gazebo -v 3 lights.sdf --render-engine ogre : The scene renders but I get a 1Hz flickering in the scene window.
    When using ign gazebo -v 4 lights.sdf --render-engine ogre2 : The scene does not render and I get a black scene window  
      - [ ] Sensor rendering error. 
      - [x ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [x ] running in virtual machine (VMWare on a Windows host system)
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils

00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]

          echo "$DISPLAY"
:0

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.1
    Max compat profile version: 4.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: SVGA3D; build: RELEASE;  LLVM;
OpenGL core profile version string: 4.1 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.10
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.1 (Compatibility Profile) Mesa 21.2.6
OpenGL shading language version string: 4.10
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

          ps aux | grep Xorg
mwittwer   37608  0.0  0.0   8908   720 pts/0    S+   01:20   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager

X.Org X Server 1.20.13
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux ubuntu 5.13.0-41-generic #46~20.04.1-Ubuntu SMP Wed Apr 20 13:16:21 UTC 2022 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.13.0-41-generic root=UUID=1334f42f-c827-4526-9773-269d54391b5c ro find_preseed=/preseed.cfg auto noprompt priority=critical locale=en_US quiet
Build Date: 14 December 2021  02:14:13PM
xorg-server 2:1.20.13-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
[... 752 lines cut ...]
01:14:58: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
01:14:58: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
01:14:58: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
01:14:58: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
```

</details>

## Description
* Expected behavior: Ignition Gazebo GUI launching without flickering or black screen
* Actual behavior: When ogre was used: The scene flickered once every second (went gray for a second and content showed up again). If ogre2 was used the scene window stayed black.

## Steps to reproduce

1. Launch Ubuntu 20.4 in VM using VMWare
2. Install gazebo from packages: https://gazebosim.org/docs/fortress/install_ubuntu
3. Launch `ign gazebo -v 4 lights.sdf`

## Output

[image]

[image]

[image]

marc-wittwer, 2022-05-17:
When I built the GARDEN version from source the flickering did disappear on the ogre version. However, the black screen remained when I used the ogre2 configuration.

srmainwaring, 2022-05-17:
Hi @marc-wittwer, I suspect this is similar to https://github.com/gazebosim/gz-sim/issues/1116. 

It has to do with the level of OpenGL support offered by the hypervisor running the VM (rather than what Ubuntu tells you it has when running glxinfo or similar). You can switch to software rendering (v. slow), and for ogre (1) there are some MESA settings you can adjust to remove the flickering, and it's a bit of trial and error depending upon your machine / OS etc.

marc-wittwer, 2022-05-18:
@srmainwaring Thanks for your reply.

#1116 led me to this link: https://answers.gazebosim.org//question/27244/ignition-flashing-image-cant-quit/
This might be worth a try.

marc-wittwer, 2022-06-01:
Disabling accelereted 3D graphics in the VMWare settings, solved the flickering.

I still have to run the `ign gazebo` with the `--render-engine ogre` argument.

`ign gazebo lights.sdf -v 4 --render-engine ogre`

darksylinc, 2022-06-04:
According to your output VMWare supports GL 4.1

Although Ogre2 only requires OGL >= 3.3 we do depend on a couple of extensions that were introduced later on and backported to those drivers.

does ...

```bash
# Required
glxinfo | grep GL_ARB_shading_language_420pack
glxinfo | grep GL_ARB_texture_storage

# Optional but really good to have
glxinfo | grep GL_ARB_buffer_storage
glxinfo | grep GL_ARB_texture_buffer_range
glxinfo | grep GL_ARB_base_instance

# misc, just in case. In fact it may be better if this one returns nothing
glxinfo | grep GL_ARB_multi_draw_indirect
```

get hits in all of them for the required ones? And what does the rest get?

If the required are not supported, the VMWare can't run ogre2 with acceleration (use `LIBGL_ALWAYS_SOFTWARE` environment variable to use SW emulation, though it's slow).

If the optional have some missing functions, it is possible you hit a bug; since we don't often hit those paths.

srmainwaring, 2024-05-15:
Appreciate this issue had been inactive for a while but as it's still open, and gz-harmonic has been released for Ubuntu 24.04, here is an update on rendering in VM.

Host System: macOS Sonoma 14.4.1, arm64
VM: Parallels Desktop 19.3.1
VM System: Ubuntu 24.04, arm64

### Summary

- Hardware acceleration for `ogre` is working.
- Hardware acceleration for `ogre2` is not working.

The required extensions listed above are present:

```bash
# Required
glxinfo | grep GL_ARB_shading_language_420pack
glxinfo | grep GL_ARB_texture_storage
```

The optional extensions are not:

```bash
glxinfo | grep GL_ARB_buffer_storage
glxinfo | grep GL_ARB_texture_buffer_range
glxinfo | grep GL_ARB_base_instance
```

There are some errors in the `Hlms/Common/GLSL/CrossPlatformSettings_piece_all.glsl` script concerning the handling of `texelFetch`. Any use of `texelFetch` should be aliased to `bufferFetch` as there are fallback definitions for this when `GL_ARB_texture_buffer_range`, however not all cases are aliased. This causes a script compilation error.

```bash
```bash
$ simple_demo ogre2

terminate called after throwing an instance of 'Ogre::RenderingAPIException'
  what():  OGRE EXCEPTION(3:RenderingAPIException): Fragment Program 100000003PixelShader_ps failed to compile. See compile log above for details. in GLSLShader::compile at ./RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 361)
Aborted (core dumped)
```

Once this is fixed, running the `gz-rendering/examples/simple_demo` with the `ogre2` stops crashing, but there is no scene rendered (black screen).

Similarly running the `gz-sim` GUI will render a black screen with the command:

```bash
QT_QPA_PLATFORM=xcb gz sim -v4 -g --render-engine ogre2 --render-engine-gui-api-backend opengl
```

oursland, 2024-09-03:
I can confirm that `ogre` works, but `ogre2` does not, failing with the fragment shader compilation failure exception.

georgetok, 2024-11-05:
@srmainwaring can u elaborate how you launch with your config? 

I do `LIBGL_ALWAYS_SOFTWARE=1 QT_QPA_PLATFORM=xcb gz sim --render-engine ogre` and get "Gazebo GUI" is not responding.

alepistella, 2026-11-05:
An update to the Parallels / Ubuntu 24.04 results above: **OGRE 2 still fails in my setup after the Parallels 27.1 update**.

Environment: MacBook Pro M5 Max, Ubuntu 24.04 ARM64, Gazebo Harmonic (gz-sim 8.15.0), ROS 2 Jazzy and system Mesa 25.2.8. The latest check used **Parallels Desktop and Tools 27.1.0 (58756)**.

With the packaged OGRE 2 configuration and no private Mesa or runtime overrides, the shader compilation fails with:

```text
no matching function for call to texelFetch(usampler2D, int)
```

This is followed by a fragment shader compilation exception. It looks related to the shader fallback problem described in the 2024 update, although I haven’t established that the underlying cause is identical.

For OGRE 1, rendering the scene was only part of the problem: I also encountered flickering, incorrect GPU LiDAR ranges and black camera frames when noise was enabled. My working profile combines a patched OGRE 1 rendering build, disabled shadows and **private Mesa 24.0.5 loaded only by Gazebo**. System Mesa remains unchanged.

The original tests on **Parallels 27.0.3 (58680)** reached around **60 GUI fps**, with working GPU LiDAR and RGB camera output. I rechecked rendering successfully on 27.1.0; that was a functional check, not a repeat of the full performance and sensor measurements. No software-rendering fallback is used.

[Installation, patches and rollback](https://github.com/alepistella/dum-e/blob/main/gazebo-parallels/INSTALL.md) · [27.1 retest and logs](https://github.com/alepistella/dum-e/tree/main/gazebo-parallels/retests/2026-11-05)

This is a usable OGRE 1 workaround on one tested machine, not a fix for OGRE 2 or evidence that the original VMware case has the same cause.