gazebosim/gz-sim #1494: Robot should stop when battery is discharged.

adityapande-1995, 2022-05-17:
## Environment
* OS Version:  Ubuntu 20.04
* Source build of Gazebo Garden

## Description
In the example file ``linear_battery_demo.sdf``, this line (https://github.com/gazebosim/gz-sim/blob/9370793caf94e96a49ba8103c6c950644a573f46/examples/worlds/linear_battery_demo.sdf#L31) says that the blue vehicle should stop when the battery runs out. I tried running the example and letting the battery run out, but the blue vehicle did not stop.

* Expected behavior: The blue vehicle should stop after the battery is drained, respective of the velocity command.
* Actual behavior: The vehicle continues to move.

## Steps to reproduce
1. In the file ``linear_battery_demo.sdf`` change the line 476 to have a high power load, say 660 instead of the default 6.6, so that the battery drains quicker.
2. Run the file using ``ign gazebo -v 4 linear_battery_demo.sdf -z 1000000``
3. In a separate terminal, run : ``ign topic -t "/model/vehicle_blue/cmd_vel" -m ignition.msgs.Twist -p "linear: {x: 6.0} angular: {z: 0.4}"``
4. Check the battery status in a separate terminal : ``ign topic -e -t /model/vehicle_blue/battery/linear_battery/state``. 

The blue vehicle continues to move even after the battery is fully drained.

[image]

arjo129, 2022-05-18:
I did a quick search of the usages of the `components::Battery`. https://github.com/gazebosim/gz-sim/search?q=components%3A%3ABatterySoC

Seems like systems should read the BatterySoC component like the `Sensor` system does: https://github.com/gazebosim/gz-sim/search?q=components%3A%3ABatterySoC

adityapande-1995, 2022-05-18:
This section in ``Physics.cc`` is checking for the battery status and is supposed to stop the joints from rotating : https://github.com/gazebosim/gz-sim/blob/f4198be19ae432bbcc584b73887178320e69ed90/src/systems/physics/Physics.cc#L1802-L1807

arjo129, 2022-05-18:
Makes sense for the case of the example above. However, there are many systems which apply forces/set velocities directly on links. These will not avail of the joint force property.

adityapande-1995, 2022-05-27:
This PR fixes it for this particular example : https://github.com/gazebosim/gz-sim/pull/1500

closed 2022-05-27 by adityapande-1995