gazebosim/gz-sim #1509: /world/<name_world>/pose/info made random numbers

Kakcalu13, 2022-05-27:
## Environment
* OS Version: **20.04 Ubuntu**
* Source or binary build? **Binary**  `Ignition Gazebo, version 3.12.0`

## Description
* Expected behavior: Expected it to display like 
```
  name: "freenove_smart_car"
  id: 25
  position {
    x: 0
    y: 0
    z: -0.1  
}

```
* Actual behavior:   n
```
name: "freenove_smart_car"
  id: 25
  position {
    x: -1.4130059618855037e-10
    y: 3.4495946332577985e-12
    z: -0.0448807174173829
  }
```
Generated x and y randomly. See here video:
[image]

## Steps to reproduce

1. `ign topic -e -t /world/free_world/pose/info | grep "freenove_smart_car" -A6` in a separate terminal
2. Load world in a separate terminal
3. Load robot using the command: ign service -s /world/free_world/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 300 --req 'sdf_filename: ''"models/sdf/freenove_smart_car.sdf" pose: {position: {z: 5}} ' in a separate terminal
4. Seeing X and Y generated random numbers from the first terminal

I made a zip for you to reproduce it easier:
[delete_this.zip](https://github.com/gazebosim/gz-sim/files/8787335/delete_this.zip)

If you are seeing this, ignore it since there is no png pictures in this folder. It is expected errors and doesn't impact on actual project.
[image]

But I've found alternate way to "fix" this by add x and y greater than 0.0 which is here:
```
ign service -s /world/free_world/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 300 --req 'sdf_filename: ''"models/sdf/freenove_smart_car.sdf" pose: {position: {x: 0.1, y: 0.1, z: 5}} ' 
```

This fixed the issue, but why can't it do that if it's 0 or no input on coordinate?

Kakcalu13, 2022-05-27:
Fortress does the same thing too
[image]

Kakcalu13, 2022-05-27:
Updated: [OdometryPublisher](https://github.com/gazebosim/ros_gz/issues/172) worked nicely! However, it's on Fortress+ only

luca-della-vedova, 2022-06-02:
I'm not sure I see where the issue is, generally speaking some deviation is expected because of floating point errors, or approximations in the physics solver.
From what I can see the values between iterations change by a value that is much less than 10^-9 for x and y (below one nanometer!) while the model seems to be slowly moving up in Z (maybe you spawned it a bit too far below ground and now it's slowly moving up to avoid having the mesh intersecting with the ground plane?).

closed 2022-06-02 by Kakcalu13