gazebosim/gz-sim #1550: Introspect forces on a body

adityapande-1995, 2022-06-22:
## Desired behavior
It would be helpful to have a topic and later a GUI plugin that can visualize forces on a selected body.
Currently there is no way to do that except fr adding debug prints in the plugins.

## Implementation suggestion
We could have an additional plugin or make Physics plugin publish net forces on a topic. Ideally it would be nice to have a breakdown of the forces, (say due to gravity, hydrodynamics, normal reaction, actuators, etc) with the net force on the body.

chapulina, 2022-06-23:
There's some work started here to visualize forces:

* https://github.com/gazebosim/gz-sim/pull/1184

The implementation needs a bit more thought though. Ideally the server wouldn't be publishing markers directly. Instead, it could publish the wrenches applied by each plugin, and a GUI plugin is the one which aggregates that and publishes markers according to a configuration set by the user (color, marker scale, etc).

Another thing that is missing from that PR is the possibility of visualizing forces internal to the physics engine, like gravity. I don't have a clear idea on how to expose that though.

arjo129, 2022-08-24:
Yeah this is something thats been on my backlog for some time. I agree with whatever @chapulina has said. I have not had time to work on this to refine it in any way shape or form.

adityapande-1995, 2022-08-31:
Even if we do not have the breakdown of forces, we can still have a topic published by a "ForceInspector" plugin, that can publish the net forces and torques on links. It can use the linear and angular velocity data from the ecm, which would also include gravity, since these velocities are the result of all forces.

arjo129, 2022-09-28:
So I had tried this method earlier, its not particularly performant. Perhaps I will get time to work on it soon.