gazebosim/gz-sim #1553: :man_farmer: Gazebo @ main fails to build on Windows

Blast545, 2022-06-24:
## Build Regression

## Environment
* OS Version: Windows
* Source build 

## Details / Output
Reference build: See [Jenkins  gazebo build](https://build.osrfoundation.org/job/ign_gazebo-ci-win/199/)

<details>

```
CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:73 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [ignition-common5] (Components: profiler, events,
  av)
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:77 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:82 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

-- Configuring incomplete, errors occurred!
See also "C:/Jenkins/workspace/ign_gazebo-ci-win/ws/build/ignition-gazebo7/CMakeFiles/CMakeOutput.log".
---
--- stderr: ignition-gazebo7
CMake Warning at C:/vcpkg/scripts/buildsystems/vcpkg.cmake:782 (_find_package):
  By not providing "Findignition-common5-av.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-common5-av", but CMake did not find one.

  Could not find a package configuration file provided by
  "ignition-common5-av" (requested version 5.0.0) with any of the following
  names:

    ignition-common5-avConfig.cmake
    ignition-common5-av-config.cmake

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:73 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [ignition-common5] (Components: profiler, events,
  av)
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:77 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:82 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:214 (gz_configure_build)

```

</details>

## Description
* Expected behavior: Job builds and completes successfully
* Actual behavior: Fails to compile

## Steps to reproduce
1. Launch a job on  [Jenkins  gazebo build](https://build.osrfoundation.org/job/ig/n_gazebo-ci-win/)
2. See it failing

## Additional information
It fails on `colcon compilation without test for dependencies of ignition-gazebo7`. It's failing to find `gz-common5`, potentially related to missing dependencies on this package at CMake level.

`Missing dependency [ignition-common5] (Components: profiler, events, av)`. Pending further investigation with Gazebo team.

chapulina, 2022-07-25:
Can this issue be closed, @Blast545 ?

https://build.osrfoundation.org/job/ign_gazebo-ci-win/

mjcarroll, 2022-10-27:
I think this is stale.  gz-sim7 builds (mostly) without issue.