gazebosim/gz-sim #1554: Able to add relative with length of shapes

Kakcalu13, 2022-06-24:
I think this will be amazing feature if we can have this. See here of actual [example](https://gazebosim.org/docs/fortress/building_robot#left-wheel) in general:

```
<link name='left_wheel'>
    <pose relative_to="chassis">-0.5 0.6 0 -1.5707 0 0</pose>
```

But if I want to add length based on chassis' size.

Something like this would be nice:

```
<link name='left_wheel'>
    <pose relative_to="chassis::box">-0.5 0.6 0 -1.5707 0 0</pose>
```

So that way, it can move based on size of box. Not only box, it can be cylinder, box, or any shape!

chapulina, 2022-06-27:
I believe such logic is better left to other tools that can be used to generate SDF files. For example, see the [ERB](https://gazebosim.org/api/gazebo/6.9/erbtemplate.html) tutorial.

* PS1: please use the issue template next time
* PS2: this ticket would have been better fit for the https://github.com/gazebosim/sdformat repo.

closed 2022-06-27 by chapulina

Kakcalu13, 2022-06-27:
Ah, thank you!

I will be more careful to post ticket from now on! Very much appreciate for the tutorial!