gazebosim/gz-sim #1583: Changing the config + log folder location?

darksylinc, 2022-07-11:
Currently Gazebo's ecosystem saves everything to:

`/home/username/.ignition`

It's quite common to use `/home/username/.config/project_name` (or `.local`) instead to avoid cluttering the home folder.

Given that Gazebo, formerly Ignition, is undergoing a name change (from Ignition to Gazebo) we may take the opportunity to change location as well.

## Desired behavior

Logs & Config files should be saved in `~/.config/gz`

chapulina, 2022-07-11:
After some quick searching, it looks like using `~/.config` is [recently being preferred](https://stackoverflow.com/a/1024216/6451468) to `~/.application`. I still see many files with the old pattern though, like `~/.git`, `~/.aws`, `~/.docker`, `~/.vim`, to cite a few. But I do see others in the new location though, like `~/.config/inkscape`, `~/.config/gh`, etc.

So even though it seems to be the latest recommendation, it doesn't seem to be widely adopted yet. I imagine the old directories have a lot of momentum, with users expecting them to be there, and lots of documentation still referring to them. I agree that if we'd make a change, the renaming to `gz` may be a good time for it.

Another thing to take into account though is that `~/.config` is meant for user-specific [configuration files](https://stackoverflow.com/a/1024339/6451468), but we have way more than that in `~/.ignition`. We also have logs, cached data, etc. So we would need to split the contents of `~/.ignition` into multiple directories (`~/.config/gz`, `~/.share/gz`, `~/.cache/gz`, `/var/log/gz`...). I'm so used to `~/.ignition` and wiping it out as needed that a split feels awkward to me, but that's not enough reason to stick to an old pattern.

darksylinc, 2022-07-11:
Personally I prefer everything in the same folder (i.e. `~/.config/gz`) because it makes uninstallion (or cleanup) much easier.

Separating global config into /etc and logs into /var/log makes sense for long-running processes such as Apache or X11, where global configuration (i.e. touching files in /etc) happens sporadically (e.g. during setup or server migration) and logs in /var/log are either an important part of auditing (so you want all relevant logs from multiple processes in a single location) or something you rarely look at (i.e. you can delete everything in /var/log to free up space and forget about it).

1. Why didn't the system boot properly? Let's look in /var/log
2. Was there a security breach? Let's look at Apache /Nginx+ mysql + syslog in /var/log
3. Why did this installer I just ran fail to install? Let's look in /var/log or in /tmp
4. Why did this specific app ABC just fail on me? Let's look in  `~/.config/ABC`

Apps that are incredibly specific or only run while the user is interacting with it IMHO are best to place everything in a single folder.

As for the use of `~/.cache` I'm torn with.

For example it's very useful for me that Chrome, Firefox and Mesa (the gpu driver) writes into it. Because I know that deleting these folders are safe when I need to free up space or fix that malfunctioning app (without breaking config). But these are apps that are part of my daily life.

So my stance is to either:

1. Only have everything `~/.config/gz`
2. Only have `~/.config/gz` + `~/.cache/gz`

The rest IMHO are excessive for Gazebo and would only increase the mental load.

nuclearsandwich, 2022-07-11:
> It seems the ideal thing to do is to use the `$XDG_CONFIG_HOME` variable and then fall back to `~/.config/`, as specified by the standard.

I would definitely advocate adopting the XDG base directory specifications for where to put stuff. It makes it much easier to version control configuration files separate from data files between machines. It will also make it much easier to do cursed/blessed things with [`systemd-homed`](https://man.archlinux.org/man/systemd-homed.8)

chapulina, 2022-07-20:
Just a quick update, although we think this is generally a good idea, it may be tight to get in before feature freeze (9 days from now).

chapulina, 2022-07-27:
Just a note that even though we generally think this is a nice idea, we won't be able to get this into Garden before feature freeze. The change can be considered for Gazebo H+, but then we'll have to deal with the migration from `~/.gz` to `~/.config/gz`.