gazebosim/gz-sim #1619: Add TF publishing in MecanumDrive system

cccleon, 2022-07-29:
Thanks for your excellent work!
I use the mecanum plugin in my sdf file: 
`<plugin
        filename="ignition-gazebo-mecanum-drive-system"
        name="ignition::gazebo::systems::MecanumDrive">
        <front_left_joint>front_left_wheel_joint</front_left_joint>
        <front_right_joint>front_right_wheel_joint</front_right_joint>
        <back_left_joint>rear_left_wheel_joint</back_left_joint>
        <back_right_joint>rear_right_wheel_joint</back_right_joint>
        <min_acceleration>-5</min_acceleration>
        <max_acceleration>5</max_acceleration>
      </plugin>`
By running 'ign topic -l', I can see the topic "/model/xxx/odometry", but I get nothing when I echo this topic.
Additionally, "odom" frame is not in the TF tree, and there is no element like "publish_odom_tf" or "odom_frame" in MecanumDrive.cc, which can be found in the source code of navigation2 tutorial(diff-drive) .The tutorial uses 'libgazebo_ros_diff_drive.so', but I can't find the mecanum_drive.so

chapulina, 2022-07-29:
Hi @cccleon , can you please update your feature request to follow the template?

```
<!-- If you're not sure on the specifics of the feature or would like a broader
discussion, please consider posting a proposal to
http://community.gazebosim.org/ instead.-->

## Desired behavior
<!-- Describe the current problem and the feature you want implemented.-->

## Alternatives considered
<!-- Describe alternate solutions or features you've considered.-->

## Implementation suggestion
<!-- Provide a suggestion on how to implement this feature, which could help us
expedite this implementation.-->

## Additional context
<!-- Provide any other additional context (e.g., screenshots) about your feature
request.-->

```

andyblarblar, 2022-07-31:
It looks like the MecanumDrive system lacks an updateOdometry function unlike the ackermann and diffDrive systems. To get odom and TF you'll need to first calculate accurate odom for a Mec drive, and then copy that odom message exactly over to a Pose_V message for Tf. The publishers are in the mec drive file, but unused.

cccleon, 2022-08-01:
> It looks like the MecanumDrive system lacks an updateOdometry function unlike the ackermann and diffDrive systems. To get odom and TF you'll need to first calculate accurate odom for a Mec drive, [...]

Thanks for your reply. You are right. The publisher is defined in line 308 but unused.

cccleon, 2022-08-01:
## Desired behavior
Add updateOdometry function in MecanumDrive.cc

## Implementation suggestion
Caculate the odometry message from joint states and publish.

## Additional context
Maybe users can specify the <robot_base_frame> and <odom_frame> in sdf, thus the plugin can publish the TF between odom frame and robot_base frame as they want.

> Hi @cccleon , can you please update your feature request to follow the template?
> 
> ```
> 
> 
> ## Desired behavior
> 
> 
> ## Alternatives considered
> 
> 
> ## Implementation suggestion
> 
> 
> ## Additional context
> 
> ```
Thanks for your reply! I think the problem is the lack of updateOdometry function as @andyblarblar  mentioned.

## Desired behavior
Add updateOdometry function in MecanumDrive.cc

## Implementation suggestion
Caculate the odometry message from joint states and publish.

## Additional context
Maybe users can specify the <robot_base_frame> and <odom_frame> in sdf, thus the plugin can publish the TF between odom frame and robot_base frame as they want.

ahcorde, 2022-08-25:
Hi @danilogsch,

thank you for your contributions! I added some comments in the gz-math PR https://github.com/gazebosim/gz-math/pull/486

Feel free to ask any question you have!

danilogsch, 2022-08-25:
HI @ahcorde ,
I resolved most of it, on the gz-sim PR I am not able to rebase and sign off the commit, I am closing that to open a new one.

ahcorde, 2025-03-10:
Related PR https://github.com/gazebosim/gz-sim/pull/2297