gazebosim/gz-sim #1648: RPY gives different data from the ign topic

Kakcalu13, 2022-08-15:
## Environment
* OS Version: 20.04
* Source or binary build?  
  Citadel (3.13.0) and Fortress (6.10.0)
    
<details>
  
The RPY on the GUI displays different number than the topic's data. See here:
[image]

I implemented the `<custom_rpy>` on the sdf. The data shows differently. 

Here is the sdf file that I was able to re-created on a simple and different sdf from the actual sdf. 

Rename this `test.txt` to `test.sdf`. Github won't let me upload the sdf.
[test.txt](https://github.com/gazebosim/gz-sim/files/9339474/test.txt)

I will explain the steps below.

</details>

## Description
* Expected behavior: Expected topic to publish the exact same number as GUI shows. 
* Actual behavior: Does not show the exact same numbers. 

## Steps to reproduce

1. `ign gazebo -r test.sdf` (Fortress or Citadel, doesn't matter)
2. Open a new terminal and run this command: `ign topic -e -t /imu | grep orientation -A6`
3. Put 1 in Yaw, See the z(is Yaw due to setting) in the terminal

https://user-images.githubusercontent.com/65916520/184707397-a555b0aa-76bc-4385-a8da-1ff2bf3327f0.mp4

Let me know if you are seeing the same thing. If it's just on my end, I will close this issue

chapulina, 2022-08-16:
The orientation is RPY in the GUI, but it's a quaternion on the topic echo. See that it has not only xyz, but also w.

closed 2022-08-16 by chapulina

Kakcalu13, 2022-08-16:
Oh, I thought that `<custom_rpy>` would "re-wire" the quaternion to RPY. 

It seems that this is just technical question at this point, I will move to answers.gazebo. 

Thank you for your time!