gazebosim/gz-sim #1662: `Conveyor` & `TrackedVehicle` example worlds are not working

jasmeet0915, 2022-08-25:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
Tested on multiple builds as mentioned below:
Binary: Fortress v6.10
Source: Edifice v5.3 & Garden v7.0~pre1

## Description
* Expected behavior: The [`conveyor.sdf`](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/conveyor.sdf) and [`tracked_vehicle_simple.sdf`](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/tracked_vehicle_simple.sdf) examples should work using the TrackController plugin.
* Actual behavior: The examples are not working. The tracks do not move using Keypress or manually publishing values to required topics. Setting `<debug>` as true does not print any debug messages nor any other error messages are observed. Tested the same on multiple systems and builds but the result is the same.

## Output
[image]

[image]

ahcorde, 2022-08-25:
Hi @jasmeet0915,

You need to install this dart from 
 - this PPA https://launchpad.net/~openrobotics/+archive/ubuntu/dart `dart6.10` 
 - or from source https://github.com/gazebo-forks/dart `release-6.10`

because you need a [specific header](https://github.com/gazebosim/gz-physics/blob/ign-physics5/dartsim/CMakeLists.txt#L53) which is not available in ubuntu 20.04.

Anyhow I was testing this feature on Garden and I found some issues with the conveyor, can you test it too?

jasmeet0915, 2022-08-26:
Hi @ahcorde,

Thank you for your answer. I installed the `release-6.10` of dart from the source.
The `tracked_vehicle_simple.sdf` example now seems to work on the Garden source build I had. Should I add those instructions to the comments in the sdf files?
Also, the default `conveyor.sdf` for Garden seems to be missing the `WorldControl` GUI plugin. After adding the plugin to the sdf I am able to start the sim and it works as well. 
Was this one of the issues you were talking about? Are there any other issues that I might be overlooking? I would be happy to test it out further.

Also, those examples seem to work on the Garden source build but still don't work on the Fortress Binary install as well as the Edifice source install I had. Any ideas about what might be the issue there?

frederikjchristensen, 2022-09-12:
@ahcorde 
I used Synaptic Package manager to check after adding the PPA. Because it still was not working for me. It seemed there was multiple versions installed.

Both 6.9.2.2 and 6.10.0.0.

[image]

azazdeaz, 2022-11-02:
I reproduced this issue on 22.04+Fortress/Garden and Ubuntu20.04+Fortress/Dome. The examples worked on Ubuntu 18.04+Dome.

How can i get these examples working under Ubuntu 22.04?

I tried building `https://github.com/gazebo-forks/dart` and use it as the `--physics-engine` but Gazebo complains that it isn't wrapped as an IgnitionPlugin: 
```bash
Library [/home/azazdeaz/repos/temp/dart/build/lib/libdart.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[Err] [Physics.cc:771] Unable to load the [/home/azazdeaz/repos/temp/dart/build/lib/libdart.so] library.
```
Is there any documentationtutorial i can follow to get this working? (I found things like [this](https://github.com/gazebosim/gz-physics/blob/ign-physics5/tutorials/04-switching-physics-engines.md), but they didn't get me closer)

liushuya7, 2022-11-10:
I am having the same issue on Ubuntu 22.04 + Garden. I noticed that gazebo uses DART by default and `libdart-dev` is already installed in the system.

```bash
$ gz sim -h

  --physics-engine [arg]       Gazebo Physics engine plugin to load.            
                               Gazebo will use DART by default.                 
                               (gz-physics-dartsim-plugin)                
                               Make sure custom plugins are in                  
                               GZ_SIM_PHYSICS_ENGINE_PATH. 
```
However, nothing is happening with the example. No any topic named `/model/${model_name}/link/${link_name}/track_cmd_vel` as described in [TrackController Class Reference](https://gazebosim.org/api/sim/7/classgz_1_1sim_1_1systems_1_1TrackController.html)

jrutgeer, 2023-03-27:
To prevent anyone else from loosing a lot of time on this:

> You need to install [...] `dart6.10`
> because you need a specific header which is not available in ubuntu 20.04.

I wrongly interpreted this as "You need `dart6.10` and Ubuntu 20.04 _has an older version which lacks_ a specific header."
As Ubuntu 22.04 ships with `dart6.12` I thought this was ok, which obviously it isn't.
What is actually meant, is that you need a _specific patched version_. This patch has been merged in `dart6.13`, see [the release info](https://github.com/dartsim/dart/releases) and [corresponding PR](https://github.com/dartsim/dart/pull/1626).

So once I realized this, I got it working by:

- Uninstalling libdart 6.12:
   - `sudo apt remove libdart*`
   - As the gazebo/ignition packages have a dependency on this, these also need to be uninstalled,
   - Not sure if it is possible to reinstall these after installing dart 6.13. 
- Per [these instructions](https://dartsim.github.io/install_dart_on_ubuntu.html) add the PPA's and install libdart-dev.
  - I installed "all": `sudo apt-get install libdart6-all-dev`
- Per [these instructions](https://gazebosim.org/docs/garden/install_ubuntu_src) install Gazebo Garden from source
  - `dart6.13` should be found automatically. You can check this: `build/gz-physics6/CMakeCache.txt` will contain `DART_HAS_CONTACT_SURFACE_HEADER:INTERNAL=1`
  - There was a compile error about a missing file included from `#include <dart/utils/utils.hpp>` in `src/gz-physics/test/plugins/DARTDoublePendulum.cc`
    - I just commented out that line (`//#include <dart/utils/utils.hpp>`) and/or I compiled the second time with tests disabled `colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install`
- Finally bash complained that `gz` could not be found in `install/gz-tools2/bin/` but that was resolved by restarting the xterm and sourcing `install/setup.bash`.

</br>

@ahcorde 
> Anyhow I was testing this feature on Garden and I found some issues with the conveyor, can you test it too?

What issues? If I start the simulation, the cube drops onto the conveyor and if I press 'w' it starts moving. 

The only thing I assume is that belt velocities are instantaneously set, whereas the acceleration of the object on the belt is probably limited by the maximum friction force between belt and object.

Bi0T1N, 2023-03-28:
I also tried to get it working without knowing that you replied but my steps are similar to yours. I used Ignition Fortress instead of Gazebo Garden on Ubuntu 22.04. However, these are my findings you didn't mentioned:
1. The following command from the source install instruction will install `libdart` from the Ubuntu repository. Thus I edited the corresponding `packages.apt` inside the `.github/ci/` folder (rename to `libdart6` for the PPA version). This may not be necessary as CMake prefers the newer version anyway as it seems to work just fine with your approach.
```bash
sudo apt -y install \
  $(sort -u $(find . -iname 'packages-'`lsb_release -cs`'.apt' -o -iname 'packages.apt' | grep -v '/\.git/') | sed '/ignition\|sdf/d' | tr '\n' ' ')
```
2. I directly used `colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install` so can't say if my approach will result in a compilation error

> 
> > Anyhow I was testing this feature on Garden and I found some issues with the conveyor, can you test it too?
> 
> What issues? If I start the simulation, the cube drops onto the conveyor and if I press 'w' it starts moving.
> 

For me it works fine. `w` for forward, `s` for stop and `x` for backwards. Just make sure it doesn't leave the conveyor belt because then it's gone forever (restart simulation required).
However, the [tracked_vehicle_simple.sdf](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/tracked_vehicle_simple.sdf) example still won't work for me. It doesn't load the simulation but that might be something on my environment (have to retest in a container to rule this out). Can you confirm that it works for you?

jrutgeer, 2023-03-29:
@Bi0T1N 

> The following command from the source install instruction will install libdart from the Ubuntu repository.

Indeed, I had installed Garden from source first, and then lateron removed libdart, installed dart 6.13 and recompiled Garden. So I didn't run the `apt -y install` command a second time.

</br>

> However, the [tracked_vehicle_simple.sdf](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/tracked_vehicle_simple.sdf) example still won't work for me.

I tried the `conveyor.sdf`, `track_drive.sdf` and `tracked_vehicle_simple.sdf` demos and all work fine. I hadn't run the tracked vehicle examples before and they are strangely cool, actually.  #boyswillbeboys ;-)

Bi0T1N, 2023-03-30:
> I tried the `conveyor.sdf`, `track_drive.sdf` and `tracked_vehicle_simple.sdf` demos and all work fine. I hadn't run the tracked vehicle examples before and they are strangely cool, actually. [...]

After I tested it again, it also worked perfectly for me.

However, I think this shows some major issues:
1. All the examples that are shipped with Gazebo (starting points for beginners!) should have strong tests running within the CI pipeline to detect problems like this immediately. However, for the ones mentioned in this issue there is [one](https://github.com/gazebosim/gz-sim/blob/gz-sim7/test/integration/tracked_vehicle_system.cc) that tests `tracked_vehicle_simple.sdf` and `conveyor.sdf` (note that the used SDF files are different than those in `examples/worlds` folder that a user would see in his installation).
2. The simulation tests are passing on [e.g. the Focal CI](https://github.com/gazebosim/gz-sim/actions/runs/4513428880/jobs/7948246336) while it fails on a normal installation which shouldn't occur if the CI would use the same steps/environment that are written on the [Binary Ubuntu Install](https://gazebosim.org/docs/fortress/install_ubuntu) page.
```
          Start 211: INTEGRATION_tracked_vehicle_system
  211/235 Test #211: INTEGRATION_tracked_vehicle_system .....................   Passed    1.34 sec
          Start 212: check_INTEGRATION_tracked_vehicle_system
  212/235 Test #212: check_INTEGRATION_tracked_vehicle_system ...............   Passed    0.02 sec
```
3. Either the official `libdart-*` packages in Ubuntu Focal and Jammy get updated to 6.13 or the install instructions should advice to use the PPA repository and thus the `packages.apt`  files need to be adapted to use `libdart6-*` packages.

benmalenfant, 2023-12-01:
Is there a recommanded install procedure for the new dart lib? I'm trying to install the new dart from the PPA with the ros release of gazebo fortress ros-humble-ros-gz however, when I try to install ros-humble-ros-gz with apt (after installing dart-dev from the PPA), I seem to create a conflict and basically brick apt... A colleague managed to make it work building gazebo from source in a docker but it'd be nice to have a easier way to do it in the future with apt

azeey, 2023-12-13:
gz-harmonic is shipped with DART 6.13, so the Conveyer and TrackedVehicle work without compiling DART from source.

closed 2023-12-13 by azeey

peci1, 2024-02-26:
This issue is not solved on 20.04/Fortress. The system DART does not support tracked vehicles at all (it has even many more missing features as it is DART 6.9 and not the 6.10 relied upon during SubT when the tracked vehicles support was added).

For reference, this is a short (and possibly non-complete) guide to what I did to get it working without breaking Gazebo 11 and keeping most of Fortress installed from binary (except physics):

```
git clone https://github.com/gazebo-forks/dart -b release-6.10
mkdir dart/build
cd dart/build
sudo apt install coinor-libipopt-dev libnlopt-cxx-dev
cmake .. -DCMAKE_INSTALL_PREFIX=/opt/dart6.10 -DCMAKE_BUILD_TYPE=Release
make -j12 && sudo make -j12 install
cd ../..
git clone https://github.com/gazebosim/gz-physics -b ign-physics5
mkdir gz-physics/build
cd gz-physics/build
CMAKE_PREFIX_PATH=/opt/dart6.10/share/dart/cmake:$CMAKE_PREFIX_PATH cmake .. -DCMAKE_INSTALL_PREFIX=/opt/ign-physics5 -DCMAKE_BUILD_TYPE=RelWithDebInfo
make -j12 && sudo make -j12 install
# Now this is ugly hack but there is no way to convince gz::common::SystemPaths to prefer a custom engine named the same as a preinstalled one.
sudo ln -sf /opt/ign-physics5/lib/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so /usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so
echo 'export LD_LIBRARY_PATH=/opt/ign-physics5/lib:/opt/dart6.10/lib:$LD_LIBRARY_PATH' >> ~/.bashrc
```

To disable the ugly hack and use stock ign-physics, do:

    sudo ln -sf libignition-physics5-dartsim-plugin.so /usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so

@azeey If there's a better way to run a custom build of ign-physics without building ign-gazebo from source, please tell me.

pucciland95, 2024-07-16:
Hi, as of today the example does not work using Gazebo Fortress. 

Do I really need to build from source Gazebo to make the example work?

benmalenfant, 2024-07-16:
I'd recommand moving to a later version of gazebo, the solution we ended up with was to use harmonic and build ros_gz from source with the correct version flag

pucciland95, 2024-07-16:
Yes, with Gazebo Harmonic works. However, I have now broken everything related to ros2_control with Gazebo since I cannot find the related plugins. Do any of you guys have experience with this? Do I need to compile from source also ros2 control?