gazebosim/gz-sim #1668: Pose3d and Vector3d widgets do not display on macOS

srmainwaring, 2022-08-26:
## Environment

* OS Version: macOS Monterey, Version 12.5, Xcode 13.4.1.
* Source or binary build? build: source, branch: `gz-sim7`, commit: d90fab283c7b4c26ffa8a5af9fff055ff21718b1.

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section.

    - Rendering plugin: [`ogre2`].
      - [ x ] running on real hardware.
      - [ x ] GPU is concurrently used for other tasks.
        - [ x ] desktop acceleration.
    - Rendering system info:
```
2022-08-13 15:45:06.469 system_profiler[80028:24179151] Device PreExisted [000000010000096a] AMD Radeon Pro W5700X
Graphics/Displays:

    AMD Radeon Pro W5700X:

      Chipset Model: AMD Radeon Pro W5700X
      Type: GPU
      Bus: PCIe
      Slot: Slot-1
      PCIe Lane Width: x16
      VRAM (Total): 16 GB
      Vendor: AMD (0x1002)
      Device ID: 0x7310
      Revision ID: 0x0000
      ROM Revision: 113-D1860W-213
      VBIOS Version: 113-D186A2XT-019
      Option ROM Version: 113-D186A2XT-019
      EFI Driver Version: 01.01.200
      Metal Family: Supported, Metal GPUFamily macOS 2
      Displays:
        Cinema HD:
          Display Type: LCD
          Resolution: 2560 x 1600
          UI Looks like: 2560 x 1600
          Framebuffer Depth: 30-Bit Color (ARGB2101010)
          Display Serial Number: CY628129UG1
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Rotation: Supported
          Automatically Adjust Brightness: Yes
          Connection Type: DVI or HDMI
          Adapter Firmware Version: 2.21
```

## Description

The Pose3d and Vector3d widgets in the Component Inspector are not displaying correctly on macOS.

#### Expected behaviour

The pose and vector widgets should display correctly.

#### Actual behaviour

The pose and vector widgets are empty.

## Steps to reproduce

Run the `shapes.sdf` example on macOS:

```zsh
#server
gz sim -s -r shapes.sdf
```

```zsh
#client
gz sim -g
```

## Output

*Vector3d fails to display*

[image]

*Pose3d fails to display*

[image]

### Possible cause

The problem seems to be that `Layout.fillWidth: true` is not sizing the widget correctly (at least on macOS).

https://github.com/gazebosim/gz-sim/blob/25bdef685919196be79af5083710152dcff4871d/src/gui/plugins/component_inspector/Vector3d.qml#L96-L100

A temporary workaround is to force the width of the widget to the parent less margin:

```js
        // Content
        GzVector3 {
          id: gzVectorInstance
          width: parent.width - 2*margin
          Layout.fillWidth: true
          gzUnit: model && model.unit != undefined ? model.unit : 'm'
```

This patch is here: https://github.com/srmainwaring/gz-sim/commit/b4de80f95d0b91782d060a35978fd8631689acc7 and provides a working solution for macOS, but may not be suitable for general use.

*Vector3d after fix*

[image]

*Pose3d after fix*

[image]

chapulina, 2022-08-27:
This may be related to the Qt version, Homebrew is often ahead of Ubuntu. Do you know what version you have, @srmainwaring ?

srmainwaring, 2022-08-27:
> Do you know what version you have

I'm using the latest version from brew:

```
% brew info qt5
==> qt@5: stable 5.15.5 (bottled) [keg-only]
Cross-platform application and UI framework
https://www.qt.io/
/usr/local/Cellar/qt@5/5.15.5_1 (10,760 files, 350.5MB)
  Poured from bottle on 2022-08-11 at 12:04:10
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/qt@5.rb
License: GFDL-1.3-only and GPL-2.0-only and GPL-3.0-only and LGPL-2.1-only and LGPL-3.0-only
```

The documentation for Qt 5.15 suggests that `Layout.fillWidth` is available (https://doc.qt.io/qt-5/qml-qtquick-layouts-layout.html#fillWidth-attached-prop).

I've changed my patch to use `Layout.preferredWidth: parent.width` instead which solves the issue and seems more in line with the recommended use of `Layout`. 

```xml
        // Content
        GzPose {
          id: gzPoseInstance
          Layout.fillWidth: true
          Layout.preferredWidth: parent.width
```

If this seems reasonable I'll submit a PR.

chapulina, 2022-08-29:
> If this seems reasonable I'll submit a PR.

Thanks, sounds reasonable, a PR would be appreciated

azeey, 2022-09-26:
Closed by #1680

PR #1680 (2022-09-13): Fix display of Pose3d and Vector3d on macOS
This PR fixes the display of Pose3d and Vector3d on macOS.

# 🦟 Bug fix

Fixes #1668

## Summary

The fix sets an additional layout attribute `Layout.preferredWidth: parent.width` to the GzPose and GzVector3 declarations in Pose3d.qml and Vector3d.qml of the component inspector. Without these the entries render empty on macOS / Metal.

Further details and tests are provided in https://github.com/gazebosim/gz-sim/issues/1668.

## Checklist
- [ x ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migrat [... cut]
closed 2022-09-26 by azeey