gazebosim/gz-sim #1672: /model/robot/cmd_vel is not published by ignition-gazebo-velocity-control-system

padhupradheep, 2022-08-29:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
  binary

## Description
* Expected behavior: needs to output the cmd_vel topic
* Actual behavior: cannot see the topic at all and there are no logs

## Steps to reproduce

1. Try to use `ignition-gazebo-velocity-control-system` in Ubuntu 22.04

## Output
NA

mjcarroll, 2022-08-29:
Can you supply some more information like the world file that you are launching?

padhupradheep, 2022-08-29:
I'm using empty_world and also the robot model is spawned using an URDF and not an SDF. Odometry publisher works without a problem, but cannot spawn the velocity control plugin. 

I'm not sure what else info I can provide

azeey, 2022-09-26:
The `velocity-control-system` subscribes to the `cmd_vel` topic, so there would be no output on that topic. It is intended for users to publish to that topic. Subscriptions are also not listed when doing `gz topic -l`. Are you saying you are publishing to that topic and you're not seeing the results you expected in the simulation?

padhupradheep, 2022-09-27:
>Are you saying you are publishing to that topic and you're not seeing the results you expected in the simulation?

Thanks for taking it up. Exactly, I tried to publish to that topic and I was not able to see any results in the simulation. 

If you need additional information, let me know.

azeey, 2022-11-28:
@padhupradheep 

Can you post the URDF file that you are using? Do you have the `velocity-control-system` placed in the URDF somehow? Are there any errors when running the simulation with `-v4` command line flag?

padhupradheep, 2022-11-28:
Hey, here is the [URDF](https://github.com/neobotix/neo_simulation2/blob/fortress-migration/robots/mpo_700/mpo_700.urdf). 

I’ll test it in a couple of days and get back.

azeey, 2022-11-28:
Seems to work fine.

https://user-images.githubusercontent.com/206116/204391885-5d7f262e-adac-42b1-bf36-87234bc3f94f.mp4

Do you know what command you used to publish the velocity? I used the following for the video:
```
ign topic -t "/model/mpo_700/cmd_vel" -m ignition.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.05}"
```

One issue might be that you used  `gz.msgs.Twist` instead of `ignition.msgs.Twist`. For Fortress, you need to use `ignition.msgs.Twist`.

padhupradheep, 2022-11-29:
Unfortunately I don’t remember the details on which command I used. But, I’ll try it once again for sure and get back within couple of days. Until then, can I keep the issue open?

padhupradheep, 2022-12-01:
Hey, I cannot see the video, it says that the file is corrupt..

azeey, 2022-12-01:
hmm, that's strange. I've converted it to a gif here. Hope this works for you:
[image]

padhupradheep, 2022-12-07:
unrelated: did you set the GZ_SIM_RESOURCE_PATH.. 

because I'm getting an error stating that URI not found, even if I set the `GZ_SIM_RESOURCE_PATH`.

azeey, 2022-12-07:
Yes, but you need to make sure you're using the correct environment variable as they have changed due to the `ign->gz` migration. If you're using Fortress, it is `IGN_GAZEBO_RESOURCE_PATH`. In Garden and later it's `GZ_SIM_RESOURCE_PATH`. We're in the process of allowing `GZ_SIM_*` environment variables to work in Fortress and earlier to avoid the confusion, but we're not there yet.

padhupradheep, 2022-12-07:
So, I was actually trying both garden and fortress.. irrespective I'm getting the same issue. I know it is entirely a different issue to the one posted, I'm just taking the liberty to post few logs and the command that I used:

So I used the resource path you mentioned.. when I do the echo of the variable I get:

`/home/user/garden_ws/src/neo_simulation2/robots/mpo_700`

In the logs, while I try to launch I get:

```
[ruby $(which ign) gazebo-1] [Err] [SystemPaths.cc:378] Unable to find file with URI [model://neo_simulation2/robots/mpo_700/meshes/MPO-700-BODY.dae]
[ruby $(which ign) gazebo-1] [Err] [SystemPaths.cc:473] Could not resolve file [model://neo_simulation2/robots/mpo_700/meshes/MPO-700-BODY.dae]
[ruby $(which ign) gazebo-1] [Err] [MeshManager.cc:173] Unable to find file[model://neo_simulation2/robots/mpo_700/meshes/MPO-700-BODY.dae]
```

am I doing the export properly.. 

Once again really sorry for hijacking the issue with another unrelated one..

azeey, 2022-12-07:
Ah, the `*_RESOURCE_PATH` needs to be set to the directory that contains `neo_simulation2` as it will basically look for `$GZ_SIM_RESOURCE_PATH/neo_simulation2/robots/mpo_700/meshes/MPO-700-BODY.dae` for example.
See https://gazebosim.org/api/sim/7/resources.html.

padhupradheep, 2022-12-07:
Thanks! That worked.. I was in that documentation for some time, but really didn't understand. 

I think it could be nice to have an example in the documentation, got it all mixed it up with gazebo classic.

padhupradheep, 2022-12-07:
Also the environment hooks shown in the ROSCon talk points to the folder containing the meshes..

azeey, 2022-12-07:
Glad it worked. And thanks for the feedback. I agree an example would be great. I think Gazebo itself could also print out all the paths it looked for the mesh/resource. If you time and are interested in contributing, I'd be happy to review PRs 🙂..

padhupradheep, 2022-12-07:
>  If you time and are interested in contributing, I'd be happy to review PRs

Sure, we @neobotix love to contribute.. we want to setup the nav2 stack with the fortress / garden. During the process, we'll surely love to provide feedback and contribute as well..

padhupradheep, 2022-12-07:
I still don't get the `/model/mpo_700/cmd_vel` topic

```
ign topic -l
/clock
/gazebo/resource_paths
/gui/camera/pose
/model/mpo_700/odometry
/model/mpo_700/odometry_with_covariance
/model/mpo_700/pose
/stats
/world/default/clock
/world/default/dynamic_pose/info
/world/default/pose/info
/world/default/scene/deletion
/world/default/scene/info
/world/default/state
/world/default/stats
```

azeey, 2022-12-07:
The topic won't be listed in `ign topic -l`. That command only lists topics that are being published whereas the `cmd_vel` topic is being subscribed to by the velocity controller system.

padhupradheep, 2022-12-07:
I can see the plugin loaded in the GUI, but when I publish the velocity like the way you do, there is no result..

padhupradheep, 2022-12-07:
Right, it was totally on me.. I didn't set the play on the bottom left.. There is another play/stop button on the top right, which are for the updates.. Somehow I missed the ones for the physics..

Thank you very much @azeey ! Really appreciate the time you spent. I'll definitely allocate some resource for contributing to documentation and to the stack overall if needed...

closed 2022-12-07 by padhupradheep

azeey, 2022-12-07:
Glad it's resolved!