gazebosim/gz-sim #1679: Ign Sim Library Error libgz-sim7-gz.so.7.0.0~pre1 cannot load

sachinkum0009, 2022-08-30:
## Environment
* OS Version:  Ubuntu 20.04
* Source or binary build?
  
  
Source, following [this guide](https://gazebosim.org/docs/garden/install_ubuntu_src) to install ignition gazebo garden 

Build is successful, but when I try to launch the ignition gazebo
It gives this error
```bash
Library error for [/home/asus/ignition_garden_ws/install/lib/libgz-sim7-gz.so.7.0.0~pre1]: can't load /home/asus/ignition_garden_ws/install/lib/libgz-sim7-gz.so.7.0.0~pre1
```
Not sure what is the issue. Please help me. Also please let me know what other details or log should I share to solve this issue.

sachinkum0009, 2022-09-02:
## More Logs

```bash
colcon build --merge-install
[0.163s] WARNING:colcon.colcon_core.package_identification:No valid Build content
[0.166s] WARNING:colcon.colcon_core.package_identification:_identify(src/gz-common) has multiple matches and therefore is being ignored: bazel, cmake
[0.176s] WARNING:colcon.colcon_core.package_identification:No valid Build content
[0.178s] WARNING:colcon.colcon_core.package_identification:_identify(src/gz-math) has multiple matches and therefore is being ignored: bazel, cmake
[0.182s] WARNING:colcon.colcon_core.package_identification:No valid Build content
[0.207s] WARNING:colcon.colcon_core.package_identification:_identify(src/gz-msgs) has multiple matches and therefore is being ignored: bazel, cmake
[0.239s] WARNING:colcon.colcon_core.package_identification:No valid Build content
[0.253s] WARNING:colcon.colcon_core.package_identification:_identify(src/gz-utils) has multiple matches and therefore is being ignored: bazel, cmake
Starting >>> gz-cmake3
Finished <<< gz-cmake3 [0.42s]                      
Starting >>> gz-plugin2
Starting >>> gz-transport12
Starting >>> sdformat13
Starting >>> gz-fuel_tools8
Starting >>> gz-tools2
Finished <<< gz-tools2 [0.48s]                                  
Finished <<< gz-plugin2 [2.22s]                                      
Starting >>> gz-rendering7
Finished <<< gz-transport12 [12.1s]                                        
Finished <<< gz-fuel_tools8 [14.4s]                                   
[Processing: gz-rendering7, sdformat13]                               
[Processing: gz-rendering7, sdformat13]                                       
[Processing: gz-rendering7, sdformat13]     
[Processing: gz-rendering7, sdformat13]                                       
[Processing: gz-rendering7, sdformat13]     
--- stderr: gz-rendering7                   
CMake Warning at /home/asus/workspace/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:69 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [optix]: Missing dependency [OptiX].
      ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:176 (gz_configure_build)

---
Finished <<< gz-rendering7 [2min 53s]
Starting >>> gz-gui7
[Processing: gz-gui7, sdformat13]                                             
Finished <<< sdformat13 [3min 54s]                                            
Starting >>> gz-physics6
Starting >>> gz-sensors7
Finished <<< gz-gui7 [1min 12s]                                              
[Processing: gz-physics6, gz-sensors7]                                      
[Processing: gz-physics6, gz-sensors7]                                      
[Processing: gz-physics6, gz-sensors7]      
[Processing: gz-physics6, gz-sensors7]      
--- stderr: gz-physics6                     
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [tpe/plugin/CMakeFiles/gz-physics6-tpe-plugin.dir/build.make:154: tpe/plugin/CMakeFiles/gz-physics6-tpe-plugin.dir/src/plugin.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3463: tpe/plugin/CMakeFiles/gz-physics6-tpe-plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:163: all] Error 2
---
Failed   <<< gz-physics6 [2min 33s, exited with code 2]
Aborted  <<< gz-sensors7 [2min 35s]                                          

Summary: 8 packages finished [6min 30s]
  1 package failed: gz-physics6
  1 package aborted: gz-sensors7
  2 packages had stderr output: gz-physics6 gz-rendering7
  2 packages not processed
```

Can you please tell why I am getting this warning about multiple matches?

azeey, 2022-09-26:
It looks like you're running out of memory when building gz-physics. Please see the suggestions in https://github.com/gazebosim/gz-physics/issues/349. I'll go ahead and close this, but feel free to open if the issue persists.

closed 2022-09-26 by azeey