gazebosim/gz-sim #1695: Help with writing Custom Plugin to move models as animation

NickNair, 2022-09-09:
I am trying to create animated models and as ignition doesn't support animation for non skeletal models yet I have decided to write a custom plugin to do the same. Are there any examples to help me get started?

arjo129, 2022-09-12:
Take a look at the [examples](https://github.com/gazebosim/gz-sim/tree/ign-gazebo6/examples/plugin) folder. I'm closing this issue as it is not an issue. In future ask on https://answers.gazebosim.org/questions/ or https://community.gazebosim.org/.

closed 2022-09-12 by arjo129

NickNair, 2023-03-29:
@arjo129  I think this issues ties in with #1932 and #418

I have created a custom plugin called MoveModel to animate non-skeletal models. 
[image]

## Usage 

To use the plugin add the following tag with the respective waypoints to the model tag in the sdf

```
<plugin filename="MoveModel" name="move_model::MoveModel">
    <waypoint>
        <time>0</time>
        <pose>0 0 1 0 0 0.7</pose>
    </waypoint>
    <waypoint>
        <time>10</time>
        <pose>-5 -5 1 0 0.7 0</pose>
    </waypoint>
    .
    .
    .
    <waypoint>
    <time>25</time>
    <pose>5 -5 6 1.54 0 0</pose>
    </waypoint>
</plugin>
```

ngifary, 2023-04-22:
@arjo129 Could you give me pointers on how to create this plugin? I need to do something similiar

NickNair, 2023-04-23:
> @arjo129 Could you give me pointers on how to create this plugin? I need to do something similiar

Hey @ngifary Did you even look at my previous comment. WHat is wrong with it?

ngifary, 2023-04-23:
Am I somehow asking this wrong? You have created a custom plugin MoveModel and showed how to use it. But what I'm asking is what is in the MoveModel plugin. If I were to run just the sdf without the plugin, it would ask me to load the plugin, wouldn't it?

NickNair, 2023-04-23:
> If I were to run just the sdf without the plugin, it would ask me to load the plugin, wouldn't it?
Well that's just how plugins work in gazbeo I guess. I don't know what to say about that.

The code for the MoveModel Plugin and the underlying libary is based of the Animation.h headers created for gazebo classic.

The code is here : https://github.com/NickNair/move_model_plugin

ngifary, 2023-04-23:
Yes the code for the MoveModel plugin is what I'm looking for. Since you weren't uploading it before, I thought you didn't want to and that's why I asked for pointers to create it myself. Sorry for the misunderstanding and thank you!

AdarshKaran, 2025-09-26:
Could anyone please let me know why this plugin hasn't been integrated into gz sim? Correct me if I'm wrong, but is it because the Actor plugin does something similar.