gazebosim/gz-sim #1700: Including ignition-gazebo6 package adds HOME directory to search path in Clion

Jakub-Krakowiak, 2022-09-13:
## Environment
* OS Version: Ubuntu 20.04
* Binary build version: ignition-fortress
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time

## Description
When adding ignition-gazebo6 package to a project, the HOME directory gets inluded in the search path, so Clion will potentially be indexing for a very long time unnecessarily.

## Steps to reproduce
1. install ignition-fortress from binary
2. open empty c++ library project in Clion and paste code below to CMakeLists.txt
```
cmake_minimum_required(VERSION 3.23)
project(test_project)

set(CMAKE_CXX_STANDARD 20)

find_package(ignition-gazebo6 REQUIRED)

add_library(GazeboPlugin SHARED main.cpp)
target_link_libraries(GazeboPlugin
        ignition-gazebo6::ignition-gazebo6
        )
```
3. reload cmake in Clion
4. added home directory should be visible in the External Libraries tab on the left

## Output
[image]

mjcarroll, 2022-09-26:
My suspicion is that it is this line: https://github.com/gazebosim/sdformat/blame/sdf12/src/CMakeLists.txt#L126

mjcarroll, 2022-09-26:
@Jakub-Krakowiak can you take a look at https://github.com/gazebosim/sdformat/pull/1170 and see if it fixes this for you.

Jakub-Krakowiak, 2022-11-10:
the issue is resolved, thank you <3

closed 2022-11-10 by Jakub-Krakowiak