gazebosim/gz-sim #1706: Lockstep video record results in deadlock

nkoenig, 2022-09-14:
## Environment
* OS Version: Ubunut
* Source or binary build? Source

## Description
* Expected behavior: Recording a video using the GUI and `<lockstep>true</lockstep>` should work.
* Actual behavior: When `<lockstep>` is true then the GUI deadlocks.

## Steps to reproduce

1. Modify the `video_record_dbl_pendulum.sdf` example world to set `<lockstep>true</lockstep>`
2. Run the world.
3. Select the video record button.

azeey, 2022-09-15:
Instead of clicking on the button on the top left, if you load the Video Recording plugin from the plugin menu and start recording, the deadlock doesn't happen.

https://user-images.githubusercontent.com/206116/190283649-42744b94-8cb8-49c7-889e-0ec50d465c46.mp4

tejalbarnwal, 2023-07-20:
Hey, is the issue been resolved?

tejalbarnwal, 2023-07-20:
Hello, has the problem concerning the top-left button causing a deadlock been resolved?

azeey, 2023-07-20:
As far as I know, it hasn't been addressed.