gazebosim/gz-sim #1708: Include collection name (eg. Fortress, Garden) along with the version

azeey, 2022-09-14:
## Desired behavior
The output of `gz sim --version` and the "About" dialog box currently only contain the version number, but users would find it more useful if they also contained the collection name since it's not easy to make the association between version numbers and the collection name without digging through the gazebosim website.

## Alternatives considered
n/a

## Implementation suggestion
This could be a CMake define so that it only needs to be updated in one place, in case if we need the name elsewhere.

## Additional context
Current dialog box:
[image]

ahcorde, 2022-10-12:
@azeey I checked that there is a CMake variable called `GZ_DISTRIBUTION`

[image]

[image]

PR https://github.com/gazebosim/gz-sim/pull/1756

azeey, 2022-10-12:
Oh that's convenient!

ahcorde, 2022-10-12:
PR is merged!

closed 2022-10-12 by ahcorde