gazebosim/gz-sim #1715: model.sdf not parsed

tejalbarnwal, 2022-09-16:
## Description
* Expected behavior: Any `.sdf` and `.urdf` file should be able to parse
* Actual behavior: Following [this](http://sdformat.org/tutorials?tut=quickstart) tutorial for Garden, wasn't able to parse sdf file when the sdf file is named as `mode.sdf`. When the same file is renamed with any other name, it works.

## Steps to reproduce

1. Download any `model.sdf` file from [here](http://models.gazebosim.org/)
2. Follow the tutorial [here](http://sdformat.org/tutorials?tut=quickstart)
3. Run `./check_sdf model.sdf`

## Output

> for `./check_sdf model.sdf`
```
model.sdf is not a valid SDF file!
```

## Environment
* OS: Ubuntu 20.04
* Build: Binary Garden Installation

azeey, 2022-09-26:
Thanks @tejalbarnwal. Since this is an SDFormat issue, would you mind creating the issue at https://github.com/gazebosim/sdformat?

I just said the same thing at https://github.com/gazebosim/gz-tools/issues/105, so there might already be an issue there, but otherwise, please feel free to move this issue.

tejalbarnwal, 2022-09-27:
Sure, I will do it

azeey, 2022-09-28:
Duplicate of https://github.com/gazebosim/sdformat/issues/1172

closed 2022-09-28 by azeey