gazebosim/gz-sim #1724: `gz service` is not available as a command, how to list the available services? (spawn_urdf tutorial)

gaileyleseman, 2022-09-19:
See [this issue](https://github.com/gazebosim/garden-tutorial-party/issues/2033#issuecomment-1247029776) as well as [this question](https://answers.gazebosim.org/question/28448/not-able-to-run-spawn_urdf-tutorial-with-garden/). 
The documentation refers to `gz service -l`, which does not exist as far as I can tell. 
I also tried `gz sim service -l`, which also does not exist. 

Output for `gz sim service -l`:
```
/usr/lib/ruby/gz/cmdsim7.rb:323:in `parse': ambiguous option: -l (OptionParser::AmbiguousOption)
	from /usr/lib/ruby/gz/cmdsim7.rb:337:in `execute'
	from /usr/bin/gz:308:in `<main>'
/usr/lib/ruby/gz/cmdsim7.rb:323:in `parse': invalid option: l (OptionParser::InvalidOption)
	from /usr/lib/ruby/gz/cmdsim7.rb:337:in `execute'
	from /usr/bin/gz:308:in `<main>'
```

## Environment
* OS Version: Ubuntu 22.04
* Build: Binary 7.0.0~pre1
* Rendering plugin: gz-rendering-ogre2
* GPU information
```
Tue Sep 13 00:34:32 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro P1000        Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   42C    P8    N/A /  N/A |      6MiB /  4096MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2056      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+
```

azeey, 2022-09-19:
This is related: https://github.com/gazebosim/gz-tools/issues/111

azeey, 2022-10-17:
This has been fixed upstream. The issue was that `gz-transport12-cli` wasn't being installed by default.

closed 2022-10-17 by azeey