gazebosim/gz-sim #1725: Rotate Gizmo not working on Jammy

azeey, 2022-09-20:
## Environment
* OS Version: Ubuntu 22.04 
* Source or binary build? binary, Garden
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [x] running in virtual machine
      - [x] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
            direct rendering: Yes
            OpenGL vendor string: NVIDIA Corporation
            OpenGL renderer string: NVIDIA GeForce GTX 780 Ti/PCIe/SSE2
            OpenGL core profile version string: 4.6.0 NVIDIA 470.141.03
            OpenGL core profile shading language version string: 4.60 NVIDIA
            OpenGL core profile context flags: (none)
            OpenGL core profile profile mask: core profile
            OpenGL version string: 4.6.0 NVIDIA 470.141.03
            OpenGL shading language version string: 4.60 NVIDIA
            OpenGL context flags: (none)
            OpenGL profile mask: (none)
            OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 470.141.03
            OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
          ```
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
16:58:44: Creating resource group General
16:58:44: Creating resource group Internal
16:58:44: Creating resource group Autodetect
16:58:44: SceneManagerFactory for type 'DefaultSceneManager' registered.
16:58:44: Registering ResourceManager for type Material
16:58:44: Registering ResourceManager for type Mesh
16:58:44: Registering ResourceManager for type Mesh2
16:58:44: Registering ResourceManager for type OldSkeleton
16:58:44: MovableObjectFactory for type 'ParticleSystem' registered.
16:58:44: ArchiveFactory for archive type FileSystem registered.
16:58:44: ArchiveFactory for archive type Zip registered.
16:58:44: ArchiveFactory for archive type EmbeddedZip registered.
16:58:44: DDS codec registering
16:58:44: FreeImage version: 3.18.0
16:58:44: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
16:58:44: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
16:58:44: ETC codec registering
16:58:44: OITD codec registering
16:58:44: Registering ResourceManager for type HighLevelGpuProgram
16:58:44: MovableObjectFactory for type 'Decal' registered.
16:58:44: MovableObjectFactory for type 'InternalCubemapProbe' registered.
16:58:44: MovableObjectFactory for type 'Entity' registered.
16:58:44: MovableObjectFactory for type 'Item' registered.
16:58:44: MovableObjectFactory for type 'Light' registered.
[... 641 lines cut ...]
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_full]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
```

iche033, 2022-09-26:
hmm I just setup Garden on Jammy in fresh docker container but I'm not able to reproduce the issue..

[image]

azeey, 2022-09-26:
@iche033 , interesting. Were you building from source or did you use binaries?

iche033, 2022-09-26:
I built Garden from source

ahcorde, 2022-10-05:
I can see these errors:

```bash
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_full]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
```

ahcorde, 2022-10-05:
I combined the installed `gz-garden` debs with `gz-rendering` from sources. The issue is in this library because this combination is working.

iche033, 2022-10-05:
I reproduced this issue using the debs and summarized my finding in https://github.com/gazebosim/gz-rendering/pull/732#issuecomment-1268804972

azeey, 2022-10-11:
Fixed by https://github.com/gazebosim/gz-common/pull/451

closed 2022-10-11 by azeey

aserbremen, 2023-06-07:
Will this be released for gazebo fortresss. I can't seem to rotate objects in gazebo fortress

iche033, 2023-06-07:
a gz-common 4.7.0 release was made with that patch (fortress) on 05/17/23. Maybe try updating `libignition-common4-graphics-dev`

aserbremen, 2023-06-08:
I tried updating `libignition-common4-graphics-dev` but it said it is already at the newest version. `gizmo_rotate` is still not available after starting e.g. `ign gazebo shapes.sdf`