gazebosim/gz-sim #1746: Fortress and Garden on 20.04 NVIDIA VirtualGL Docker gets kills immediately after start, ogre2 is affected but not ogre

ehfd, 2022-10-02:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Binary, Garden or Fortress
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2 but not ogre]. 
      - [ ] Sensor rendering error. 
      - [X] GUI rendering error. 
    - EGL headless mode:
      - [X] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [X] running on real hardware
      - [ ] running in virtual machine
      - [X] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [X] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [X] GPU is concurrently used for other tasks
        - [X] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
```
user@xgl:~$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
04:00.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 41)
3d:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [TITAN Xp] [10de:1b02] (rev a1)
60:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [TITAN Xp] [10de:1b02] (rev a1)
61:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [TITAN Xp] [10de:1b02] (rev a1)
b2:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [TITAN Xp] [10de:1b02] (rev a1)
da:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [TITAN Xp] [10de:1b02] (rev a1)
user@xgl:~$ 
user@xgl:~$           echo "$DISPLAY"
:0.0
user@xgl:~$ 
user@xgl:~$           LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA TITAN Xp/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 510.54
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 510.54
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
user@xgl:~$ 
user@xgl:~$           ps aux | grep Xorg
user       11005  0.0  0.0  18876  4896 pts/1    S+   19:29   0:00 grep --color=auto Xorg
user@xgl:~$ 
[... 198 lines cut ...]
#18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f96f1020645, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#17   Object "/usr/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f96f0fa9447, in gz::rendering::v7::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#16   Object "/usr/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f96f0fa923e, in gz::rendering::v7::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#15   Object "/usr/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7f96f0fb7b61, in gz::rendering::v7::BaseRenderEngine::Init()
#14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f96df04590e, in gz::rendering::v7::Ogre2RenderEngine::InitImpl()
#13   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f96df043317, in gz::rendering::v7::Ogre2RenderEngine::InitAttempt()
#12   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb3ac88, in Ogre::ResourceGroupManager::initialiseAllResourceGroups(bool)
#11   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb3a6f6, in Ogre::ResourceGroupManager::parseResourceGroupScripts(Ogre::ResourceGroupManager::ResourceGroup*)
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb83c51, in Ogre::ScriptCompilerManager::parseScript(Ogre::SharedPtr<Ogre::DataStream>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb83ade, in Ogre::ScriptCompiler::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb837f2, in Ogre::ScriptCompiler::compile(Ogre::SharedPtr<std::__cxx11::list<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::STLAllocator<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb933f5, in Ogre::MaterialTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#6    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb8e391, in Ogre::ScriptTranslator::processNode(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#5    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb996c8, in Ogre::TechniqueTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#4    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deb8e391, in Ogre::ScriptTranslator::processNode(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#3    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96debae6cc, in Ogre::PassTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#2    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96dec19269, in Ogre::Technique::createPass()
#1    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f96deae3cd4, in Ogre::Pass::Pass(Ogre::Technique*, unsigned short)
[... cut ...] <(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])
[Dbg] [SignalHandler.cc:141] Received signal[2].
[Dbg] [ServerPrivate.cc:115] Server received signal[2]
[Dbg] [gz.cc:405] Shutting down gz-sim-server
[Dbg] [SimulationRunner.cc:534] Exiting postupdate worker thread (0)
```

ehfd, 2022-10-02:
Note:
I have reproduced with https://github.com/ehfd/docker-nvidia-egl-desktop.

azeey, 2022-10-06:
Looks like a duplicate of https://github.com/gazebosim/gz-rendering/issues/526.

If the goal is to run it in headless mode, have you tried setting the `--headless-rendering` flag? That should work without Xvfb or VirtualGL.

ehfd, 2022-10-07:
@azeey Yeah, it's possible, but my users want to run with a remote desktop from inside a Docker container. I raised it because it works with Citadel + ogre2, and it's probably something that changed from the ogre2 version in Citadel to Fortress that made it wrong.

iche033, 2023-01-05:
should be fixed by https://github.com/gazebosim/gz-rendering/pull/794. Please re-open if issue persists

closed 2023-01-05 by iche033