gazebosim/gz-sim #1769: Distorted capsule geometry type on ignition gazebo fortress 6.12

Jakub-Krakowiak, 2022-10-25:
## Environment
* OS Version: Ubuntu 20.04
* Binary build, version 6.12
    - Rendering plugin: ogre2
      - [x] Sensor rendering error.
      - [x] GUI rendering error.
    - EGL headless mode:
      - [?] Running in EGL headless mode
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
    - Rendering system info:
    - GPU: GTX970M driver version: 515.65.01

Capsule geometry looks severely distorted.
[image]

## Description
* Expected behavior: Capsule geometry looks normal
* Actual behavior: Capsule geometry is distorted

## Steps to reproduce
Add an entity with capsule geometry to the scene, e.g. run ign gazebo shapes.sdf

mjcarroll, 2022-10-26:
@j-rivero reproduced this locally for me.

mjcarroll, 2022-11-03:
Closed by gazebosim/gz-common#475

closed 2022-11-03 by mjcarroll