gazebosim/gz-sim #1770: can not rotate in Ubuntu22.04

Xinyu-Wu-0000, 2022-10-27:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
     binary, Garden
  
    - Rendering plugin: ogre2.
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [x] running on real hardware
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          ```
          ```
          00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630] [8086:3e9b]
          ```
          ```
          echo "$DISPLAY"
          ```
          ```
          :1
          ````
          ```
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
          ```
          X.Org X Server 1.21.1.3
          X Protocol Version 11, Revision 0
          Current Operating System: Linux xinyu-Inspiron-7590 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64
         Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-52-generic root=UUID=51fda441-b282-4808-82ed-920227a221e8 ro quiet splash resume=UUID=d6101737-b6af-40c2-a410-c5e1c41a933d vt.handoff=7
        xorg-server 2:21.1.3-2ubuntu2.2 (For technical support please see http://www.ubuntu.com/support) 
        Current version of pixman: 0.40.0
	      Before reporting problems, check http://wiki.x.org
	      to make sure that you have the latest version.
        ```
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 
[ogre2.log](https://github.com/gazebosim/gz-sim/files/9876844/ogre2.log)

<details>
[ogre2.log](https://github.com/gazebosim/gz-sim/files/9876770/ogre2.log)
</details>

## Description
* Expected behavior: 
[image]
* Real behavior:  nothing happened in GUI Windows

## Steps to reproduce

1. install gazebo : `sudo apt install gz-garden'
2. run: `gz sim shapes.sdf`
3. rotate: https://gazebosim.org/docs/garden/manipulating_models

## Output

[image]

```
gz sim shapes.sdf
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[... 243 lines cut ...]
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_full]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_full]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
[GUI] [Err] [MeshDescriptor.cc:54] Mesh manager can't find mesh named [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:566] Cannot load null mesh [gizmo_rotate_handle]
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [gizmo_rotate_handle]
[GUI] [Err] [Ogre2Visual.cc:134] Cannot attach null geometry.
```

ahcorde, 2022-10-27:
This issue is already fixed https://github.com/gazebosim/gz-sim/issues/1725

Xinyu-Wu-0000, 2022-10-27:
> 

I have seen #1725 before posting the issue. I still come to the problem. Maybe the binary in ppa is out of date or something.

mjcarroll, 2022-10-27:
> Maybe the binary in ppa is out of date or something.

The fix has not been released yet.

azeey, 2022-10-27:
I believe this needs a release of gz-rendering7, which depends on https://github.com/gazebosim/gz-rendering/pull/750

azeey, 2022-11-02:
gz-rendering 7.1.0 had been released. It has fixed the issue in my local testing.

closed 2022-11-02 by azeey