gazebosim/gz-sim #1780: SimulationRunner_TEST seg-fault on macOS

scpeters, 2022-11-04:
## Environment
* OS Version: macOS Big Sur
* Source or binary build? source build of ign-gazebo6 61df2c3abe88bd8c428dde3e823227ff9a4671ed

## Description
* Expected behavior: `SimulationRunner_TEST` test should pass
* Actual behavior: `SimulationRunner_TEST` seg-faults during the `ServerRepeat/SimulationRunnerTest.GuiInfo/0` test case if all tests are run sequentially. Running the `GuiInfo` test case individually will pass successfully.

## Steps to reproduce

1. Compile `ign-gazebo6`
2. Run `bin/UNIT_SimulationRunner_TEST`
3. Observe failure

## Output

Console output:

~~~
$ bin/UNIT_SimulationRunner_TEST 
Running main() from /Users/jenkins/workspace/ignition_gazebo-ci-pr_any-homebrew-amd64/ign-gazebo/test/gtest/src/gtest_main.cc
Note: Google Test filter = *u*
[==========] Running 13 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 13 tests from ServerRepeat/SimulationRunnerTest
[ RUN      ] ServerRepeat/SimulationRunnerTest.CreateEntities/0
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:1027] Loaded (3) plugins from file [/Users/jenkins/workspace/ignition_gazebo-ci-pr_any-homebrew-amd64/build/test/fake_home/.ignition/gazebo/6/server.config]
[Dbg] [Physics.cc:803] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics5/5.2.0/lib/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Msg] Create service on [/world/default/create]
[Msg] Remove service on [/world/default/remove]
[Msg] Pose service on [/world/default/set_pose]
[Msg] Pose service on [/world/default/set_pose_vector]
[Msg] Light configuration service on [/world/default/light_config]
[Msg] Physics service on [/world/default/set_physics]
[Msg] SphericalCoordinates service on [/world/default/set_spherical_coordinates]
[Msg] Enable collision service on [/world/default/enable_collision]
[Msg] Disable collision service on [/world/default/disable_collision]
[Msg] Material service on [/world/default/visual_config]
[Msg] Material service on [/world/default/wheel_slip]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Msg] Serving world controls on [/world/default/control], [/world/default/control/state] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[       OK ] ServerRepeat/SimulationRunnerTest.CreateEntities/0 (516 ms)
[ RUN      ] ServerRepeat/SimulationRunnerTest.CreateLights/0
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:803] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics5/5.2.0/lib/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/lights/control], [/world/lights/control/state] and [/world/lights/playback/control]
[Msg] Serving GUI information on [/world/lights/gui/info]
[Msg] World [lights] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/lights/generate_world_sdf]
[       OK ] ServerRepeat/SimulationRunnerTest.CreateLights/0 (107 ms)
[ RUN      ] ServerRepeat/SimulationRunnerTest.CreateJointEntities/0
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:803] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics5/5.2.0/lib/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Dbg] [SystemManager.cc:67] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/default/control], [/world/default/control/state] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [fast] physics profile.
[... 231 lines cut ...]
    frame #1: 0x000000010034f1d4 libignition-gazebo6.6.dylib`ignition::gazebo::v6::components::Factory::New(this=<unavailable>, _type=0x00007ffeefbfec08, _data=0x00007ffeefbfecb0) at Factory.hh:295:30 [opt]
    frame #2: 0x0000000100349a1a libignition-gazebo6.6.dylib`ignition::gazebo::v6::EntityComponentManager::CreateComponentImplementation(this=0x00007ffeefbff340, _entity=1, _componentTypeId=15972659603938252679, _data=0x00007ffeefbfecb0) at EntityComponentManager.cc:1059:51 [opt]
    frame #3: 0x000000011080e0dc libTestWorldSystem.dylib`ignition::gazebo::v6::components::Component<double, ignition::gazebo::v6::components::WorldPluginComponentTag, ignition::gazebo::v6::serializers::DefaultSerializer<double> >* ignition::gazebo::v6::EntityComponentManager::CreateComponent<ignition::gazebo::v6::components::Component<double, ignition::gazebo::v6::components::WorldPluginComponentTag, ignition::gazebo::v6::serializers::DefaultSerializer<double> > >(this=0x00007ffeefbff340, _entity=1, _data=0x00007ffeefbfecb0) at EntityComponentManager.hh:90:27 [opt]
    frame #4: 0x000000011080de3c libTestWorldSystem.dylib`ignition::gazebo::TestWorldSystem::Configure(this=<unavailable>, _entity=0x00007ffeefbfedf8, _sdf=std::__1::shared_ptr<const sdf::v12::Element>::element_type @ 0x0000000109f3fc10 strong=2 weak=5, _ecm=0x00007ffeefbff340, (null)=<unavailable>) at TestWorldSystem.hh:60:16 [opt]
    frame #5: 0x000000010041d2b9 libignition-gazebo6.6.dylib`ignition::gazebo::v6::SystemManager::AddSystemImpl(this=0x000000011075cd50, _system=SystemInternal @ 0x00007ffeefbfedb8, _sdf=std::__1::shared_ptr<const sdf::v12::Element>::element_type @ 0x0000000109f3fc10 strong=2 weak=5) at SystemManager.cc:167:24 [opt]
    frame #6: 0x000000010041c7b1 libignition-gazebo6.6.dylib`ignition::gazebo::v6::SystemManager::AddSystem(this=0x000000011075cd50, _system=0x00007ffeefbfeec8, _entity=1, _sdf=std::__1::shared_ptr<const sdf::v12::Element>::element_type @ 0x0000000109f3fc10 strong=2 weak=5) at SystemManager.cc:124:9 [opt]
    frame #7: 0x000000010041c39e libignition-gazebo6.6.dylib`ignition::gazebo::v6::SystemManager::LoadPlugin(this=0x000000011075cd50, _entity=1, _plugin=0x0000000111310390) at SystemManager.cc:66:11 [opt]
    frame #8: 0x0000000100403907 libignition-gazebo6.6.dylib`ignition::gazebo::v6::SimulationRunner::LoadServerPlugins(std::__1::list<ignition::gazebo::v6::ServerConfig::PluginInfo, std::__1::allocator<ignition::gazebo::v6::ServerConfig::PluginInfo> > const&) [inlined] ignition::gazebo::v6::SimulationRunner::LoadPlugin(this=0x00007ffeefbff300, _entity=1, _plugin=<unavailable>) at SimulationRunner.cc:864:20 [opt]
    frame #9: 0x00000001004038f8 libignition-gazebo6.6.dylib`ignition::gazebo::v6::SimulationRunner::LoadServerPlugins(this=0x00007ffeefbff300, _plugins=size=2) at SimulationRunner.cc:944:13 [opt]
    frame #10: 0x0000000100400eef libignition-gazebo6.6.dylib`ignition::gazebo::v6::SimulationRunner::SimulationRunner(this=0x00007ffeefbff300, _world=0x000000011075b4f0, _systemLoader=std::__1::shared_ptr<ignition::gazebo::v6::SystemLoader>::element_type @ 0x0000000109b10148 strong=2 weak=1, _config=0x00007ffeefbff230) at SimulationRunner.cc:202:11 [opt]
    frame #11: 0x0000000100022078 UNIT_SimulationRunner_TEST`SimulationRunnerTest_GuiInfo_Test::TestBody(this=<unavailable>) at SimulationRunner_TEST.cc:1571:20 [opt]
    frame #12: 0x0000000100089d68 UNIT_SimulationRunner_TEST`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) [inlined] void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(object=<unavailable>, method=<unavailable>, location="the test body")(), char const*) at gtest.cc:2437:10 [opt]
[... cut ...]  listeners)")(), char const*) at gtest.cc:2473:14 [opt]
    frame #20: 0x0000000100098efc UNIT_SimulationRunner_TEST`testing::UnitTest::Run(this=0x00000001000d5388) at gtest.cc:4929:10 [opt]
    frame #21: 0x00000001000ac56a UNIT_SimulationRunner_TEST`main [inlined] RUN_ALL_TESTS() at gtest.h:2473:46 [opt]
    frame #22: 0x00000001000ac55d UNIT_SimulationRunner_TEST`main(argc=1, argv=<unavailable>) at gtest_main.cc:45:10 [opt]
    frame #23: 0x00007fff203ddf3d libdyld.dylib`start + 1
(lldb) 
~~~

azeey, 2022-11-04:
We can backport https://github.com/gazebosim/gz-sim/pull/1728 to fix this. It's a band-aid, but I haven't had time to work on a more permanent solution.

scpeters, 2022-11-04:
back porting in #1782

PR #1836 (2023-02-18): Use a queue to track component registration from mulitiple sources
# 🦟 Bug fix

Fixes #1780, #806

## Summary

When a plugin gets unloaded, the component types it registered will have invalid component descriptors. Attempting to create new components of those types will result in a segfault. This PR is an attempt to fix that by keeping track all component registrations and removing invalid ones when plugins get unloaded.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Pyth [... cut]