gazebosim/gz-sim #1797: Unable to set a new pose for a Model from a System Plugin

Myzhar, 2022-11-15:
## Desired behavior

I need to set the position of a Model in the System Plugin callbacks, but the available functions for Gazebo Garden do not allow it.

You can read more here on Gazebo Answers (no reply received): 
https://answers.gazebosim.org/question/28559/change-static-model-pose-in-postupdate/

## Alternatives considered

N/A

## Implementation suggestion

You should change the declaration of the `_ecm` parameter from `EntityComponentManager &_ecm` to `const EntityComponentManager &_ecm` because the callbacks provide a constant reference to _ecm.

https://github.com/gazebosim/gz-sim/blob/gz-sim7/include/gz/sim/Model.hh#L175-L176

## Additional context

I reached my target by forcing type assignment with the `-fpermissive` compile flag, but this could be dangerous for other parts of the code.

```
    gz::math::Pose3d newPose(X,Y,Z,Roll,Pitch,Yaw);

    gz::sim::EntityComponentManager* ecm = nullptr;
    ecm = &_ecm; // This is AWFUL!!!

    mModel.SetWorldPoseCmd(*ecm,newPose);
```

azeey, 2022-11-16:
Answered in https://answers.gazebosim.org/question/28559/change-static-model-pose-in-postupdate/?answer=28564#post-id-28564. I'll go ahead and close this. Feel free to continue the discussion on Gazebo Answers.

closed 2022-11-16 by azeey

hbonnavaud, 2024-04-15:
@azeey, your link is not reachable. Can you provide a new one?

azeey, 2024-04-15:
@hbonnavaud We're working on fixing the answers.gazebosim.org site, but that specific question/answer has been copied over to
https://robotics.stackexchange.com/a/29337/31574