gazebosim/gz-sim #1802: particle_emitters error on Garden

Kakcalu13, 2022-11-20:
## Environment
* OS Version: 20.04
* Source or binary build?
  Garden 7.1.0

<details>

```
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[... 48 lines cut ...]
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
```

</details>

## Description
* Expected behavior: See normal smokes. Following [this](https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/particle_emitter.sdf)
* Actual behavior: Creates a fog OK but not red glitches. See the video

## Steps to reproduce
1) gz sim -r particle_emitter.sdf

## Output
https://user-images.githubusercontent.com/65916520/202888242-47358987-48fb-4493-a948-64273007ef82.mp4

azeey, 2022-11-21:
I'm not if this is a bug. If you look at the sdf file, it states "Nested model example, where the particles are bright red       billboards"

https://github.com/gazebosim/gz-sim/blob/4fa1f26a738cbdb3ff0c4a37fdde9506f3cd9b4e/examples/worlds/particle_emitter.sdf#L93-L116

However, I do get this error printed continually on the console:

```
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [7] does not exist.
[Err] [Physics.cc:2840] Internal error: a physics entity ptr with an ID of [9] does not exist.
```

Kakcalu13, 2022-11-22:
I'm so sorry, this is not bug. Closing as it is working intended. 

I don't mind to update this sdf and PR. Maybe I will tonight