gazebosim/gz-sim #1828: Garden

ahcorde, 2022-12-02:
## Environment
* OS Version: Ubuntu 22.04
* Source 

When loading a world with tags from fuel Gazebo is crashing.

I can see these errors

```
[Err] [SDFFeatures.cc:320] Mesh construction from an SDF has not been implemented yet for dartsim.
[Err] [SDFFeatures.cc:787] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:320] Mesh construction from an SDF has not been implemented yet for dartsim.
[Err] [SDFFeatures.cc:787] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:320] Mesh construction from an SDF has not been implemented yet for dartsim.
[Err] [SDFFeatures.cc:787] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:309] Heightmap construction from an SDF has not been implemented yet for dartsim.
[Err] [SDFFeatures.cc:787] The geometry element of collision [collision] couldn't be created
malloc(): invalid size (unsorted)
```

mjcarroll, 2022-12-02:
The errors have already been reported and I addressed them here: https://github.com/gazebosim/gz-physics/pull/452

This fix was included in 6.2.0

The malloc is new/different, though.

mjcarroll, 2022-12-02:
I confirmed that the error is still appearing so the pr mentioned above is not sufficient to fix the problem.

azeey, 2022-12-05:
Do the Fuel models use the `fuel.ignitionrobotics.org` server? If so, it might be related to  https://github.com/gazebosim/gz-fuel-tools/pull/294#issuecomment-1281384267

azeey, 2022-12-05:
@ahcorde says it might be due to issues on his local machine.

closed 2022-12-05 by azeey

23pointsNorth, 2023-10-12:
Garden 7.6.0 still exhibits the same message on the terminal. Would additionally having a more descriptive error message be better? `The geometry element of collision [collision] couldn't be created` doesn't give much understanding to what entity in the tree this refers to.

mavneo, 2023-11-21:
[Err] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
same error still present, any updates on this, not all the model from fuel show this error, only few