gazebosim/gz-sim #1834: Cannot build simple demons in Garden because of libprotobuf

dimfot3, 2022-12-07:
Hello, i have Ubuntu 22 and installed Garden from binaries. The simulation works fine but when i try to make a simple c program that read gazebo topic i get  
``'/usr/bin/ld: CMakeFiles/lidar_node.dir/test.cc.o: undefined reference to symbol '_ZN6google8protobuf11MessageLite15ParseFromStringERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
/usr/bin/ld: /lib/x86_64-linux-gnu/libprotobuf.so.23: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/lidar_node.dir/build.make:104: lidar_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lidar_node.dir/all] Error 2
make: *** [Makefile:91: all] Error 2 ``

</br>
The Cmakelists.txt and the program i try to compile is found in tutorial: https://gazebosim.org/docs/garden/sensors

The problem seems protbuf library. My protobuf  is  3.12.4 and since Garden binaries installed successfully do they need another version?
I have tried to insert -lpthread -lz -lm in CmakeLists target_link_libraries but without any progress.
I tried also to reinstall Gazebo Garden from source but i get the same annoying error. Any clue about how to fix this issue? 

Thank you for your time!

azeey, 2022-12-07:
The code for the tutorial is available at https://github.com/gazebosim/docs/tree/master/garden/tutorials/sensors. I just tested it locally and it builds fine for me. Can you give it a try?

dimfot3, 2022-12-07:
Thank you for the reply. I testes it also on different machine with same specs and seems working outwith -lprotobuf. Probably i distroyed something in machine mentioned above and created problems with protobuf. I reinstall ubuntu and seems working fine. It whould be great to share if you have any clue why needed -lprotobuf to compile in my machine mentioned above. I start thinking maybe it was cmake problem because  i had installed it from source some time ago.

azeey, 2022-12-07:
One thing you could try is to clean and rerun make in verbose mode: `make VERBOSE=1`. It will show you the exact arguments it's using when invoking the compiler.

Since this seems like a machine configuration issue, I'll go ahead and close this ticket.

closed 2022-12-07 by azeey