gazebosim/gz-sim #1841: Instant crash on start (ARM64 Ubuntu 22.04)

bertaveira, 2022-12-18:
## Environment
* OS Version: Ubuntu 22.04 ARM64 (VM on M1 Macbook)
* Binary: gz-garden

## Description
Literally installed Gazebo Garden on a brand new Ubuntu virtual machine. Was trying to test the new gazebo since I only used the classic version before.

Did the command to start the demo: `gz gazebo shapes.sdf`

And got the error on the terminal:
```
terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at ./RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 685)
Stack trace (most recent call last):
#31   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa550b06b, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa556b523, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa4382f13, in g_main_context_iteration
#28   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa43db79b, in 
#27   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa4385a6f, in g_main_context_dispatch
#26   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa556c057, in 
#25   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa550ff87, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#24   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa550cb8f, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffffa69b2abf, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#22   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa553cfd3, in QObject::event(QEvent*)
#21   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff6c817237, in gz::gui::plugins::RenderWindowItem::Ready()
#20   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff6c816d6b, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#19   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff6c81f97b, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#18   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff6c818d33, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#17   Object "/lib/aarch64-linux-gnu/libgz-rendering7.so.7", at 0xffff6c673d23, in gz::rendering::v7::BaseRenderEngine::CreateScene(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#16   Object "/lib/aarch64-linux-gnu/libgz-rendering7.so.7", at 0xffff6c67b74f, in gz::rendering::v7::BaseRenderEngine::CreateScene(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#15   Object "/lib/aarch64-linux-gnu/libgz-rendering7.so.7", at 0xffff6c6871e7, in gz::rendering::v7::BaseScene::CreateMaterials()
#14   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0xffff47fe1533, in gz::rendering::v7::Ogre2Material::SetTextureMapImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::PbsTextureTypes)
#13   Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0xffff47c1b46b, in Ogre::TextureGpuManager::_waitFor(Ogre::TextureGpu*, bool)
#12   Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0xffff47c1b0af, in Ogre::TextureGpuManager::_update(bool)
#11   Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0xffff47afebaf, in Ogre::ObjCmdBuffer::execute()
#10   Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0xffff47afcf37, in Ogre::ObjCmdBuffer::NotifyDataIsReady::execute()
#9    Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0xffff47c0cef3, in Ogre::TextureFilter::GenerateHwMipmaps::_executeSerial(Ogre::TextureGpu*)
#8    Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0xffff47592633, in Ogre::GL3PlusTextureGpu::copyTo(Ogre::TextureGpu*, Ogre::TextureBox const&, unsigned char, Ogre::TextureBox const&, unsigned char, bool, Ogre::CopyEncTransitionMode::CopyEncTransitionMode, Ogre::CopyEncTransitionMode::CopyEncTransitionMode)
#7    Object "/usr/lib/aarch64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0xffff47554997, in 
#6    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffa86f2d23, in __cxa_throw
#5    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffa86f2a3f, in std::terminate()
#4    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffa86f29db, in 
#3    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffa86f51fb, in __gnu_cxx::__verbose_terminate_handler()
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffac45712f, in abort
#1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffac46a67b, in raise
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffac4af200, in 
Aborted (Signal sent by tkill() 12810 1000)
```

Also tried opening the gazebo from the app selector. It loads, and I click on one of the environments such as the empty one and it loads and crashes a few seconds later.

traversaro, 2022-12-18:
Interesting, this seems similar to https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1172201149 . 
Which VM program are you using? If the problem is exactly the one linked, you can try as a workaround to set the LIBGL_ALWAYS_SOFTWARE env variable to 1 before running Gazebo. This should force software 3D rendering that is definitely slower, but at least something should work (in theory).

bertaveira, 2022-12-18:
I am using Parallels 18 on an M1 Pro macbook. That workaround worked although a bit inconvenient long term. I will experiment a bit using it inside docker.

I can also confirm the same happens for Fortress

srmainwaring, 2022-12-19:
@bertaveira I've tried various VMs (UTM, Parallels, VMware Fusion, Virtual Box) on macOS (Intel and M1) and none of them have good enough support for OpenGL 4.3+ to run accelerated graphics for Gazebo Sim after the upgrade of ogre2 to 2.2. Same outcome for docker.

The most promising candidate so far is VMware Fusion 13 which claims to have support for OpenGL 4.3 (https://docs.vmware.com/en/VMware-Fusion/13.0/rn/vmware-fusion-130-release-notes/index.html). I have installed Gazebo Garden from source on Ubuntu Jammy (upgraded to 22.10 to get the latest mesa drivers). Unfortunately the hardware acceleration is still not there. The mesa drivers appear to support a suitable version of OpenGL: 

```bash
glxinfo |grep -i opengl
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: SVGA3D; build: RELEASE;  LLVM;
OpenGL core profile version string: 4.3 (Core Profile) Mesa 22.2.1
OpenGL core profile shading language version string: 4.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.3 (Compatibility Profile) Mesa 22.2.1
OpenGL shading language version string: 4.30
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 22.2.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:
```

However the rendering window on `gz sim` is grey, and the rendering examples (for example `simple_demo ogre2`) are not working. I have not done a full OpenGL debug yet, so there may be a combination of [mesa environment flags](https://docs.mesa3d.org/envvars.html) that will get it working.

If you're on mac, M1 or Intel, the Metal rendering for `gz sim` running natively works well.

azeey, 2022-12-19:
@bertaveira can you confirm if what @srmainwaring suggested works (setting `LIBGL_DRI3_DISABLE=1`)?

Since this seems like a limitation of the virtual machine, I'll close this issue. Feel free to reopen if the issue can be addressed in Gazebo.

closed 2022-12-19 by azeey

srmainwaring, 2022-12-19:
> @bertaveira can you confirm if what @srmainwaring suggested works (setting `LIBGL_DRI3_DISABLE=1`)?

I don't think this will work on @bertaveira's Parallels environment - AFAIK there isn't driver support for OpenGL 4.3+ in that hypervisor. There isn't a supported nightly build for Ubuntu 22.10 (Kinetic?) - I tried installing from the nightlys but they are not targeting this version yet - so it's a case of rolling your own environment - but for the first time in about 2y there is now a way of virtualising Gazebo on macOS!

ARLunan, 2023-01-18:
This works on MAC Mini M1 Ventura 13.1 with Parallels Desktop 18, LIBGL_ALWAYS_SOFTWARE=1 in /etc/environment

ericrohloff, 2023-01-25:
> This works on MAC Mini M1 Ventura 13.1 with Parallels Desktop 18, LIBGL_ALWAYS_SOFTWARE=1 in /etc/environment

How did you get this to work? I am trying to run the sim but still getting errors.

traversaro, 2023-01-25:
Which errors?

ARLunan, 2023-01-25:
Correcting my previous post, on Ubuntu 22.04 and Lubuntu 22.04, ~/gz sim works with the line "LIBGL_ALWAYS_SOFTWARE=1" in ~/.bashrc file, remembering to "$ source .bashrc" .  In my case this environment variable does not appear in /etc/environment, though it is displayed in $ printenv|grep ALWAYS.

srmainwaring, 2023-02-03:
Another option for users looking to run `gz sim` in a virtual machine on macOS.

WSL2 now has support for GUI acceleration, so a standard binary install of gazebo-garden on an Ubuntu 22.04 WSL2 machine will run in VMWare Fusion 13 without any need to set mesa environment variables. It won't win any performance awards but does use hardware acceleration and will most likely be faster than using software rendering (`LIBGL_ALWAYS_SOFTWARE=1`). 

[image]

traversaro, 2023-02-03:
> Another option for users looking to run `gz sim` in a virtual machine on macOS.
>
> WSL2 now has support for GUI acceleration, so a standard binary install of gazebo-garden on an Ubuntu 22.04 WSL2 [...]

Interesting, are you sure that you are actually using hardware acceleration (i.e. via the `d3d12` mesa driver) or it is actually using software rendering for some reason? Probably this can be checked by running `glxinfo -B | grep renderer` in the terminal in which you run `gz sim`. As far as I know, `gz-sim` still can't run on top of d3d12 due to the lack of the `GL_ARB_copy_image` OpenGL extension (see discussion in https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1172201149), I just tried with latest Ubuntu 22.04 and gz-sim and for me it is still crashing.

srmainwaring, 2023-02-03:
@traversaro you're right - got ahead of myself there. WSL is defaulting to llvmpipe for rendering on the Win11 VM - I wonder if I can force it to use the hardware acceleration via the Win11 VM? Will keep investigating.

ARLunan, 2023-02-03:
To clarify, my requirement is to run gz-sim on a Ubuntu 22.04 configured with ROS2 Desktop machine that I used for robotics development which is a Parallels VM on. MAC M1 Silicon arm64. The M1 has its own acceleration which I do not know if supported in any way. With the LIBGL environment value, Gazebo gz-garden does run though I’m having issues with some of the ros-gz tutorial demos that open and run but are not fully operational in rviz.

srmainwaring, 2023-02-03:
@ARLunan - may not be relevant, but you can run ROS2 Humble and gz sim natively on macOS (Intel and M1). It's a bit of a slog to set up the ROS2 side (gz sim is a brew install), but once configured runs very well. Rviz is fully supported, rqt requires some patches to the Qt python packages. Running gz sim virtualised with hardware acceleration requires quite advanced OpenGL support because of the requirements of the underlying ogre2 render engine. You could run gz sim with the ogre render engine, but may find a number of the models and sensors are not supported. Happy to help with gz sim / ROS Humble on macOS if you decide to go that route.

ericrohloff, 2023-02-04:
@srmainwaring I would love to see how you set up the gzsim on macOS I have been struggling with getting it up and running

srmainwaring, 2023-09-18:
> which one method gives a somewhat working gazebo build on apple silicon /arm64?

Build from source definitely works for Gazebo Garden and Harmonic (see the from source install testing for the Harmonic release which was all done on a MacBook Pro M1).

The brew binary distro should also work for Garden and Harmonic, however I have not independently verified this recently.

Software rendering works on various VMs running Ubuntu, no hardware acceleration is currently supported on VMs AFAIK.

devashishRaj, 2023-09-18:
> The brew binary distro should also work for Garden and Harmonic, however I have not independently verified this recently.

well i tried but some [issues ](https://github.com/gazebosim/gz-sim/issues/2112?notification_referrer_id=NT_kwDOAbw4IrM3NTgwNDQ5NzM1OjI5MTEyMzU0#issuecomment-1722708308)

azeey, 2023-09-18:
@devashishRaj Please take a look at the build commands used in https://github.com/gazebosim/gazebo_test_cases/issues/804. M1 (ARM) machines are not fully supported yet, so we don't have any official documentaiton.

devashishRaj, 2023-09-20:
> @devashishRaj Please take a look at the build commands used in [gazebosim/gazebo_test_cases#804](https://github.com/gazebosim/gazebo_test_cases/issues/804). M1 (ARM) machines are not fully supported yet, so we don't have any official documentaiton.

 https://gist.github.com/devashishRaj/83627f1d729ddb6d31bb80ea6cb687d6

srmainwaring, 2023-09-20:
@devashishRaj did you unlink `qt` when building `gz-gui`?

```bash
brew unlink qt
```

This is to prevent a conflict - `gz-gui` needs `qt@5` and the brew side-by-side install of different Qt versions does not work properly.

devashishRaj, 2023-09-20:
> @devashishRaj did you unlink `qt` when building `gz-gui`?
> 
> ```shell
> brew unlink qt
> ```
> 
> This is to prevent a conflict - `gz-gui` needs `qt@5` and the brew side-by-side install of different Qt versions does not work properly.

i did not , i followed this guide : https://gazebosim.org/docs/harmonic/install_osx_src
and there's step to mention qt5 in path

srmainwaring, 2023-09-20:
The unlinking qt (=qt@6) is an extra step. You still need to follow the instructions to install qt@5.

My oversight as I should request to have the instructions updated to include the unlink step if a conflict is observed.

devashishRaj, 2023-09-20:
> The unlinking qt (=qt@6) is an extra step. You still need to follow the instructions to install qt@5.
> 
> My oversight as I should request to have the instructions updated to include the unlink step if a conflict is observed.

i'll try source install again on weekend , also i tried brew install again after watching recent commits and gz-harmonic seems to working with some minor ui bugs on m2 pro  and it's just way faster.

staycoolish, 2023-11-05:
> @ARLunan - may not be relevant, but you can run ROS2 Humble and gz sim natively on macOS (Intel and M1). It's a bit of a slog to set up the ROS2 side (gz sim is a brew install), but once configured runs very well. Rviz is fully supported, rqt requires some patches to the Qt python packages. Running gz sim virtualised with hardware acceleration requires quite advanced OpenGL support in the hypervisor because of the requirements of the underlying ogre2 render engine. You could run gz sim with the ogre render engine, but may find a number of the models and sensors are not supported. Happy to help with gz sim / ROS Humble on macOS if you decide to go that route (prob should move discussion elsewhere though).

Hello @ARLunan , could you please tell how to install Humble and gz sim natively? You can point some sources as well if possible. Thanks a lot,

ARLunan, 2023-11-05:
Hi, As an exclusive ros2 user, I do not use Humble and gz-sim natively, using instead Parallels 19 VM configured with Ubuntu 22.04 & ROS 2 Humble Desktop Full. I usually disable Graphics Acceleration in Desktop Control and rviz and ign gazebo run ok. Parallels seems to have made some changes in this  ver 19 that works better. I should add that I cannot install the new gz harmony or iron in the UB22 VM  as arm64 binary install packages are not yet release and the compile from source don’t work.

staycoolish, 2023-11-05:
> Hi, As an exclusive ros2 user, I do not use Humble and gz-sim natively, using instead Parallels 19 VM configured with Ubuntu 22.04 & ROS 2 Humble Desktop Full. I usually disable Graphics [...]

Thank you for the information @ARLunan , let me ask you a quick detail then: I am also trying to use Parallels 19 with Ubuntu 22.04 and ROS Humble. I see that some people are very happy with the performance, but they have M1 Pro or Max chips I suppose, but I have a Macbook Pro M1 model from 2020. In the Parallels forum, some users say that they enable 3D Acceleration with a trick (by setting OGRE_RTT_MODE=Copy, you can check here: https://forum.parallels.com/threads/black-screen-with-3d-acceleration-enabled-for-gazebo-and-rviz-robotic.355825/ ), but when I do that Gazebo never responds, I get a black screen and it shuts down; when I disable 3D acceleration, performance is only acceptable for rather simple cases as far as I tried. I thought, maybe I couldn't install the right Gazebo version, especially since you mention ignition Gazebo as well.

I installed the ros humble desktop full, then installed gazebo garden using binary, from this link: https://gazebosim.org/docs/garden/install_ubuntu . Are you using this version as well? I didn't specifically install an ignition Gazebo version, but found this link: https://gazebosim.org/api/gazebo/6.1/install.html . Do you suggest me to install it?

I am fairly new to ROS2, and I hope more developers release arm64 versions for their packages. I am afraid it will take years though...

azeey, 2023-11-06:
> > may not be relevant, but you can run ROS2 Humble and gz sim natively on macOS (Intel and M1). It's a bit of a slog to set up the ROS2 side (gz sim is a brew install), but once configured runs very well. Rviz is fully supported, rqt requires some patches to the Qt python packages. Running gz sim virtualised with hardware acceleration requires quite advanced OpenGL support in the hypervisor because of the requirements of the underlying ogre2 render engine. You could run gz sim with the ogre render engine, but may find a number of the models and sensors are not supported. Happy to help with gz sim / ROS Humble on macOS if you decide to go that route (prob should move discussion elsewhere though).
> 
> @srmainwaring could you please elaborate on this topic, it would be great to run ROS and Gazebo on Mac natively (I have a MacBook Pro M1 from 2020 and performance of Gazebo -I tried Garden and Fortress- is barely acceptable). Any source or explanation will be great.
> 
> Thanks,

@Kakcalu13 Can you comment on what steps you took to install gazebo on your M2? Is it just `brew install gz-harmonic`?

Kakcalu13, 2023-11-06:
Hi @azeey,

No, unfortunately, it's not just one step.

What I did was perform these steps:

```bash
ulimit -n unlimited
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
brew tap osrf/simulation
brew install gz-harmonic
```

In my case (using @scpeters's theory, since it is most likely the case), I had to reboot my Mac after encountering an error like this:

```
==> Fetching osrf/simulation/gz-harmonic
==> Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f950
Already downloaded: /Users/bwuk/Library/Caches/Homebrew/downloads/5e8fd956f9e881b791ed69ea43da80fd61a81bfb2cfefd2f777fbddcc2ec53a0--PyYAML-5.3.1.tar.gz
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-harmonic/releases
Already downloaded: /Users/bwuk/Library/Caches/Homebrew/downloads/f8eabb3491767929352f21e1ad1bbf1adf5d757e1ba72daf052a8e63913362d1--gz-harmonic-1.0.0.tar.bz2
==> Installing gz-harmonic from osrf/simulation
Error: The maximum number of open files on this system has been reached. Use `ulimit -n` to increase this limit.
bwuk@KOs-Mac-mini ~ % ulimit -n               
unlimited
bwuk@KOs-Mac-mini ~ % 
```

My Mac is an M2 mini Mac Pro with 16 CPU cores and Ventura 13.4 for those who wonder what kind of Mac I'm using.

```
Chipset Model: Apple M2 Pro
Type: GPU
Bus: Built-In
Total Number of Cores: 16
Vendor: Apple (0x106b)
Metal Support: Metal 3
Displays:
  VN279:
    Resolution: 1920 x 1080 (1080p FHD - Full High Definition)
    UI Looks like: 1920 x 1080 @ 50.00Hz
    Main Display: Yes
    Mirror: Off
    Online: Yes
    Rotation: Supported
  Beyond TV:
    Resolution: 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition)
    UI Looks like: 1920 x 1080 @ 60.00Hz
    Mirror: Off
    Online: Yes
    Rotation: Supported
    Television: Yes
```

On a side note, I skimmed all the way from top to bottom. I can only speak for M2. I'm exclusively a Linux user, which has been treating me well so far. M2 is my new place to experience. I still haven't had any luck with Windows 8, as it's my primary computer.

I'd suggest people use arm64 as a VM if they want to install Gazebo on a VM inside a Mac. Macs don't work well with amd64. The issue above sounds so familiar, which is linked to https://github.com/gazebosim/gz-sim/issues/1362#issuecomment-1147729996.

M1 and M2 should be exactly the same, as I'm a developer for Docker. (I haven't tried parallel). Anyone is welcome to tag me on M2/M1 issues.

ARLunan, 2023-11-06:
I Use Mac Mini M1 and Parallels 19.1 VM Ubuntu 22.04/ROS 2 Humble. Successfully run rviz and ign gazebo. I note that there is no CPU & Memory "Enable Acceleration" setting, instead is "Advanced Hypervisor" checkbox. ign gazebo (ver 6) and rviz run ok enabled or not. Enable I get this message" libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.' Doesn't seem to affect performance.

Then installed gz harmonic from Binary and gz sim shapes.sdf loads ok. ros-gz is already installed.

staycoolish, 2023-11-07:
> I Use Mac Mini M1 and Parallels 19.1 VM Ubuntu 22.04/ROS 2 Humble. Successfully run rviz and ign gazebo. I note that there is no CPU & Memory "Enable Acceleration" setting, instead is "Advanced [...]

Thank you @ARLunan . In Parallels there is an option called "Enable 3D Accelerations" , and you disable it, right? Where is the "Advanced Hypervisor" checkbox by the way? I allow 3D Acceleration in Parallels, and in Ubuntu .bashrc, I write **export LIBGL_ALWAYS_SOFTWARE=1**, if I don't write that Gazebo automatically quits; but since I wrote that it doesn't quit. Nevertheless, in the _quadcopter world_ for instance, I get a super slow response, there is immense lag. Similarly, in _Prius world_, the response is slow. Only using _Nao robot_ (very simple world with just one robot), for instance, the performance is okay. Is that similar in your case? Can you run complex worlds fluently? Note that, I choose those worlds from the GUI of Gazebo, I write "ign gazebo" without specifying any worlds, and then choose the ones I mentioned. If you can run fluently those in Parallels, I will try to follow your steps.

staycoolish, 2023-11-07:
> Hi @azeey,
> 
> No, unfortunately, it's not just one step.
> 
> What I did was perform these steps:
> 
> ```shell
> ulimit -n unlimited
> /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
> brew tap osrf/simulation
> brew install gz-harmonic
> ```
> 
> In my case (using @scpeters's theory, since it is most likely the case), I had to reboot my Mac after encountering an error like this:
> 
> ```
> ==> Fetching osrf/simulation/gz-harmonic
> ==> Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f950
> Already downloaded: /Users/bwuk/Library/Caches/Homebrew/downloads/5e8fd956f9e881b791ed69ea43da80fd61a81bfb2cfefd2f777fbddcc2ec53a0--PyYAML-5.3.1.tar.gz
> ==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-harmonic/releases
> Already downloaded: /Users/bwuk/Library/Caches/Homebrew/downloads/f8eabb3491767929352f21e1ad1bbf1adf5d757e1ba72daf052a8e63913362d1--gz-harmonic-1.0.0.tar.bz2
> ==> Installing gz-harmonic from osrf/simulation
> Error: The maximum number of open files on this system has been reached. Use `ulimit -n` to increase this limit.
> bwuk@KOs-Mac-mini ~ % ulimit -n               
> unlimited
> bwuk@KOs-Mac-mini ~ % 
> ```
> 
> My Mac is an M2 mini Mac Pro with 16 CPU cores and Ventura 13.4 for those who wonder what kind of Mac I'm using.
> 
> ```
> Chipset Model: Apple M2 Pro
> Type: GPU
> Bus: Built-In
> Total Number of Cores: 16
> Vendor: Apple (0x106b)
> Metal Support: Metal 3
> Displays:
>   VN279:
>     Resolution: 1920 x 1080 (1080p FHD - Full High Definition)
>     UI Looks like: 1920 x 1080 @ 50.00Hz
>     Main Display: Yes
>     Mirror: Off
>     Online: Yes
>     Rotation: Supported
>   Beyond TV:
>     Resolution: 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition)
>     UI Looks like: 1920 x 1080 @ 60.00Hz
>     Mirror: Off
>     Online: Yes
>     Rotation: Supported
>     Television: Yes
> ```
> 
> On a side note, I skimmed all the way from top to bottom. I can only speak for M2. I'm exclusively a Linux user, which has been treating me well so far. M2 is my new place to experience. I still haven't had any luck with Windows 8, as it's my primary computer.
> 
> I'd suggest people use arm64 as a VM if they want to install Gazebo on a VM inside a Mac. Macs don't work well with amd64. The issue above sounds so familiar, which is linked to [#1362 (comment)](https://github.com/gazebosim/gz-sim/issues/1362#issuecomment-1147729996).
> 
> M1 and M2 should be exactly the same, as I'm a developer for Docker. (I haven't tried parallel). Anyone is welcome to tag me on M2/M1 issues.
> 
> On personal side, Since you, gazebo developers, are reading this...is there any difference between Harmonic and Fortress/Citadel/Garden in SDF programming? O:-)

Hello @Kakcalu13 , thank you for the information! I see that you put a lot of effort for making Gazebo work natively in M1, congratulations. I also saw that you run (or spin maybe) your robot ( https://github.com/gazebosim/gz-sim/issues/1362#issuecomment-1147729996 ). Did you try the complex worlds in Gazebo? Like, _quadcopter teleop_ or _Prius on a highway_ . Right now, I use Parallels VM with arm64 Ubuntu 22.04 and ROS Humble, with Gazebo Fortress (also tried Garden, performance is similar). when I run these complex worlds, the performance fluctuates between 20% and 100% (as written at the bottom right of the Gazebo GUI), but I can tell that it is unacceptable, a lat of lag is present. Can you run those environments smoothly? I've never used Docker but if it works smoothly I will also try that way. Of course, we have to keep in mind that you have an M2 Pro whereas I have a regular M1, still I remember many people mention RoboStack which is related with Docker. Some links:

https://www.linkedin.com/feed/update/urn:li:activity:6865851557758533632/
https://robostack.github.io

Kakcalu13, 2023-11-08:
Hi @staycoolish!

I've tried out multiple complicated worlds. You can see it in action in Gazebo's tools from 0:04 to 0:23. (Big shoutout to this OP! https://github.com/gazebosim/gz-sim/pull/855 for the camera!!)

https://www.youtube.com/watch?v=h-yra2-bHZE.

I've used a Blender environment, multiple sensors, and various types of robots. Oh boy!! I had to dust off my old Dockerfile and run it on my M2. Everything felt so old, and it's only been a year! The current setup is ten times better than before!

Anyway...

Take a look at this link: 

https://github.com/gazebosim/gz-sim/assets/65916520/014e4b94-a242-4e99-b80d-678a09fde022.

I was able to use quadcopter, Prius, and some other features on Docker. The thing is that my team and I are currently focused on other projects and haven't touched Gazebo in a while. We will resume with Gazebo later. I use Gazebo in my free time now.

Anyway....

I must emphasize that this is Citadel on Docker. Regarding your comment about performance, that's exactly why I chose Citadel. If you use Fortress, Garden, or Harmonic, it will be much more resource-intensive than Citadel. Fortress, Garden, or Harmonic are much larger and more complex than Citadel. Citadel was kind of new at the time, so it was able to work with Docker due to lower rendering requirements (I'm not entirely sure if I'm using the terminology correctly).

Honestly, Gazebo performs better with a local/native setup rather than in a virtual machine. If you still want to use a VM, go for 20.04, as I believe it's currently only supported on that Citadel. Citadel is now EOL, I believe.

I personally need smooth graphics too, and so far, native/local setups are the only ones I've used. You can use Docker, but you'll need to allocate more CPU cores, maybe up to 10 or the maximum available which is what you see in the second video.

Harmonic on the M2 Mac was my very first time on a Mac. If it worked on M2, it should work with M1 as well. I have a regular M1 at my office.

I'm not familiar with robotstack, sorry :( but thanks for the information!

Could you give this a try on your M1?

Open a terminal and run:

```
ulimit -n unlimited
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
brew tap osrf/simulation
brew install gz-harmonic
```

Alternatively, you can try running `ign gazebo --render-engine ogre`. It worked for me (the last time I tried this was last year, so it might have changed by now).

ARLunan, 2023-11-08:
[staycoolish]:On my Parallels Ver 19.1.0, Ubuntu 22.04 Desktop/ROS 2 Humble Full.  I Enable 3D Acceleration, which is under Configuration> Graphics>Advanced. I enable Adaptive Hypervisor under Configuration > Graphics > Advanced.

staycoolish, 2023-11-09:
Thank you @ARLunan , on MacBook Pro M1 the Adaptive Hypervisor option is not visible, I suppose due to the hardware difference. In that case, my setup is as good as it can be. Most probably I'll not keep using Parallels or any other VM.

staycoolish, 2023-11-09:
Thanks a lot @Kakcalu13 for the detailed information!

The project you and your team work on seem very very interesting, actually I will check it further. Also, the visuals you created look great! 

I already tried most of the VMs (Multipass, UTM, Parallels), Parallels was by far the most stable and the fastest one, UTM is said to have GPU Acceleration but after the installation all I had was a black screen, and I tried a couple of times. I think they are not good enough yet for simulating robots.

I will definitely try Citadel very soon. On M1 chip, there are 8 cpu cores; and for VMs it is recommended to use 4 of them. I believe performance would be at least on par with the VM Gazebo. I think the code you provided is for installing Gazebo only, I will also check how or whether I can install ROS2 on M1 as well. 

Thanks a lot, again, for making time for all these explanations! Perhaps this can be helpful for you too: We consider different simulation alternatives these days and found Coppeliasim also worth trying. I am not sure about their different capabilities, but it seems well established as well, and they even have a native Apple silicon version.

azeey, 2023-11-09:
@staycoolish Are you not able to use gz-harmonic natively on the M1? I guess I'm not sure why you're using a VM.

Kakcalu13, 2023-11-10:
Hi @staycoolish,

I'd strongly encourage you to install natively. I was able to install it on M2, so you should be able to as well.

Also, if I may humbly suggest, let's try to keep the conversation focused on Gazebo-related topics since this whole GitHub repository is dedicated to Gazebo. I don't think the Gazebo team can provide much assistance with ROS2 here, even if they are part of the same company as ROS.

Regarding your mention of Coppeliasim, thanks! I'll definitely check it out. I'm currently using Gazebo for gaming-related purposes in my free time, haha.

I hope Citadel worked out for you! @azeey's question has piqued my curiosity as well. I'm curious as to why you have to use a VM.

For me and my team, we wanted to use a VM to deploy and allow all platforms to use Gazebo on VM. Is this the case for you as well?

This is why I wanted to see if it's possible to export Gazebo projects like this: https://github.com/gazebosim/gz-sim/issues/1336.

You know how Unity, Godot, or some other programs allow you to export projects and deploy them on various platforms? I was hoping that Gazebo could do the same thing which is why I requested the feature!

staycoolish, 2023-11-10:
Hello @azeey , @Kakcalu13 , Regarding using VM; I actually never tried because so far I've used ROS 1 in Ubuntu with Gazebo, and I thought it would be safer to stay in the familiar Ubuntu environment, plus I could've used both ROS 1 and ROS 2. That was my main motivation, but recently I noticed that not all companies have published/updated their packages for ROS 2. Most probably I'll mainly keep using my old Windows/Ubuntu PC with ROS installed in that case :D

My case is totally academic work actually, with minimal development requirements like yours. I was very happy with portability & quietness of my Macbook, I didn't want to carry a second computer (and test M1's capabilities in a way). 

I understand the functionality you are looking for (similar to Unity, etc.), but since I've used Gazebo only for visualization, I couldn't get what can be a project to be exported in Gazebo sense (or how to use it for gaming :D). I guess it is somehow like a plugin for the Gazebo? Your team seems very ambitious and I hope you can get the needed functionality.

Thanks for all the help and insight!

AndromedaHelix, 2023-12-26:
> Hi @staycoolish!
>
> I've tried out multiple complicated worlds. You can see it in action in Gazebo's tools from 0:04 to 0:23. (Big shoutout to this OP! #855 for the camera!!) [...]

I ran `ign gazebo --render-engine ogre` and this worked for me on Gazebo Fortress. Thanks, really performant

pep248, 2024-11-07:
what worked for me was:
export LIBGL_ALWAYS_SOFTWARE=1

yangjames, 2024-11-08:
> what worked for me was: export LIBGL_ALWAYS_SOFTWARE=1

This worked for me as well. I have a 2019 M1 Macbook Pro with Parallels/Ubuntu 22.04. The approach was to force software rendering through llvmpipe which emulates OpenGL software rasterizing.

```
sudo apt install mesa-utils
export LIBGL_ALWAYS_SOFTWARE=1
ign -v 4 -r <my.sdf>
```

kitare17, 2025-08-29:
Work with win 11 and wsl ubuntu 22.04
```
sudo apt install mesa-utils
export LIBGL_ALWAYS_SOFTWARE=1
```

Test command :))
` ros2 launch ros_gz_sim gz_sim.launch.py gz_args:=empty.sdf`

alepistella, 2026-11-05:
For anyone still running into this on Parallels: I have Gazebo Harmonic running with hardware acceleration on an Apple Silicon Mac, including GPU LiDAR and an RGB camera.

Tested setup: **MacBook Pro M5 Max, Ubuntu 24.04 ARM64, ROS 2 Jazzy and Gazebo Harmonic (gz-sim 8.15.0).**

The working configuration uses:

- **OGRE 1**, with a patched `gz-rendering` build.
- **Private Mesa 24.0.5**, loaded only by the Gazebo processes. System Mesa 25.2.8 remains installed.
- A runtime profile that applies these settings to normal `gz sim` and `ros2 launch` commands.

The renderer is **VirGL**, rather than `llvmpipe`; `LIBGL_ALWAYS_SOFTWARE` is not needed. The original tests reached around **60 GUI FPS**, with working GPU LiDAR ranges and RGB camera output. That figure refers to the GUI, not the camera sensor’s publication rate.

| Parallels Desktop | Stock OGRE 2 | With the OGRE 1 profile |
|---|---|---|
| 27.0.3 (58680) | Fails | Working; original performance and sensor tests |
| 27.1.0 (58756) | Still fails | Rendering retested successfully after the update |

This **avoids OGRE 2 rather than fixing it**. OGRE 2-only features remain outside the scope of this workaround, and the failure on this newer setup is not necessarily the same exception reported originally.

Installation, rollback, patches and test evidence are documented here: **[Gazebo on Parallels: installation guide](https://github.com/alepistella/dum-e/blob/main/gazebo-parallels/INSTALL.md)**.

**Before: unresponsive Gazebo GUI**

[image]

**After: hardware rendering with LiDAR visualization, approximately 57 FPS in this frame**

[image]

Validated on **one machine so far**. Independent results, including from UTM or VMware Fusion, would be useful to establish how portable this configuration is.