gazebosim/gz-sim #1844: JointPositionController not working for nested models

srmainwaring, 2022-12-21:
## Environment
* OS Version: macOS Monterey
* Source or binary build? Source: gz-sim7, 31f7d5b384408eb88db3fcc85cc5529eb926f069

## Description
* Expected behavior: The JointPositionController should control joints of nested models.
* Actual behavior: The nested joints do not respond to commands.

## Steps to reproduce

[test_nested_model.sdf.zip](https://github.com/gazebosim/gz-sim/files/10279011/test_nested_model.sdf.zip)

1. Run the Gazebo using the example world:

  ```bash
  gz sim -v4 -r test_nested_model.sdf
  ``` 

2. In a separate terminal issue commands to move the rotors:

  ```bash
  # Move rotor in model1
  gz topic -t "/rotor1_cmd" -m gz.msgs.Double -p "data: 1"

  # Move rotor in model2 (nested)
  gz topic -t "/rotor2_cmd" -m gz.msgs.Double -p "data: 1"

  # Move rotor in model3
  gz topic -t "/rotor3_cmd" -m gz.msgs.Double -p "data: 1"
  ``` 

## Output

The single yellow block is model1, the teal block is model3 and the yellow block attached to the teal block is model2 which is nested in model3.  The rotors in the top level models (model1 and model2) both move. The rotor in the nested model (model2) does not respond. 

[image]

When stepping through the joint controller plugin the `JointPositionController::PreUpdate` returns at L268 because `jointPosComp->Data()` is always empty for the nested joints.

https://github.com/gazebosim/gz-sim/blob/31f7d5b384408eb88db3fcc85cc5529eb926f069/src/systems/joint_position_controller/JointPositionController.cc#L267-L268

srmainwaring, 2022-12-22:
## Investigations

Work in progress trying to tack down the origin of the issue. Notes below.

- The issue, as far as the `gz-sim` physics plugin is concerned, appears to be that the nested joints are not in the Physics plugin's `entityJointMap`.
- The cause of this occurs at a lower level in the dart physics engine, somewhere in `gz::physics::dartsim::EntityManagementFeatures` where the nested joint is not found in the call to `GetJoint`. Possibly because this does not recurse down to nested models or for some other reason?

#### Notes

1. `gz::sim::systems:: PhysicsPrivate:: CreateModelEntities`

There are two places where nested models are constructed. The first for models that are direct children of the world: 

https://github.com/gazebosim/gz-sim/blob/31f7d5b384408eb88db3fcc85cc5529eb926f069/src/systems/physics/Physics.cc#L1123-L1124

and the second for models that are not (and so must be nested):

https://github.com/gazebosim/gz-sim/blob/31f7d5b384408eb88db3fcc85cc5529eb926f069/src/systems/physics/Physics.cc#L1175

The first call will construct models recursively, so that when the second call is made the nested model is already constructed.

2. `gz::sim::systems::PhysicsPrivate::CreateJointEntities`

For the non-nested joints, the function `CreateJointEntities` hits the test if they already exist, and proceeds to insert them into the `entityJointMap`:

https://github.com/gazebosim/gz-sim/blob/31f7d5b384408eb88db3fcc85cc5529eb926f069/src/systems/physics/Physics.cc#L1680-L1689

For the nested joints this test fails (which is odd because we see below there is an error message reporting the joint already exists). In any case the function continues until it hits:

https://github.com/gazebosim/gz-sim/blob/31f7d5b384408eb88db3fcc85cc5529eb926f069/src/systems/physics/Physics.cc#L1723-L1734

where the test for a valid physics pointer to a joint fails and the the inserting of the joint into the `entityJointMap` is skipped. To see why the attempt to create the joint failed we need to look into `ConstructSdfJoint`.

3. `gz::physics::dartsim::SDFFeatures::ConstructSdfJoint`

The attempt to create the nested joint fails here:

[gz-physics6/dartsim/src/SDFFeatures.cc#L1074-L1083](https://github.com/gazebosim/gz-physics/blob/2a93515f7995a8a060d62acd93c88294e4cae65f/dartsim/src/SDFFeatures.cc#L1074-L1083)
```c++
    auto childsParentJoint = _child->getParentJoint();
    if (childsParentJoint->getType() != "FreeJoint")
    {
      gzerr << "Asked to create a joint between links "
             << "[" << _parent->getName() << "] as parent and ["
             << _child->getName() << "] as child, but the child link already "
             << "has a parent joint of type [" << childsParentJoint->getType()
             << "].\n";
      return this->GenerateInvalidId();
    }
```

because the joint has already been associated with a link, but for some reason is not recorded in the dart physics plugin as an existing link when queried using the entity management feature `GetJointFromModel`. So that's where to look next.

azeey, 2022-12-22:
Thanks for sleuthing @srmainwaring. I'm wondering if this is a regression from #1560 or something that existed all along.

srmainwaring, 2022-12-22:
> I'm wondering if this is a regression from #1560 or something that existed all along

I think it was around before then. I rolled back gz-sim to the commit before #1560 and the issue remained.

I'm slowly zeroing in on possible causes - I suspect it may either be a recursion thing - where the dart model only reports its immediate child joints, or there is a double entry of the nested joint into the `Base::AddJoint` and one of them is not connected up to anything. It's a little daunting navigating through the type lists but I'm getting there...

srmainwaring, 2022-12-26:
@azeey - the issue arises in the function `ConstructSingleAxisJoint` in `SDFFeatures ` where the joints from the nested model are reassigned to the container model. In the notes above, I've including a link to a fork containing some debug code for `Base.hh` to print the contents of the EntityStore as these are difficult to inspect in a debugger directly.

It's not clear to me what the correct fix is - any change is going to have knock on effects elsewhere. Options seem to be:

1. Don't reassign joints to the parent entity for nested models.
2. Continue to reassign them and update the search functions to resolve joints for nested models as though they were not reassigned.

I don't think Option 2 is a good choice - it would be very confusing and almost impossible to maintain (tracing the model state is quite difficult as it is). That leaves option 1, but I'm not familiar with DART to know how the object model will cope with this type of nesting.

srmainwaring, 2022-12-28:
I've started a fix for this. The branches will probably change as I tidy things up and remove debug code but currently the changes are here:

- https://github.com/srmainwaring/gz-physics/tree/srmainwaring/6/1844-joint-pos-nested-models
- https://github.com/srmainwaring/gz-sim/tree/srmainwaring/7/joint-pos-controller

The main changes are broken down as follows:

#### `gz-physics`

- [ ] Update `EntityManagementFeatures::GetJoint` to search recursively up to the top-level containing model for a named joint (or joint by index), as the joints are moved to the container body nodes as the skeleton is assembled. [gz-physics: b9b3ffd](https://github.com/srmainwaring/gz-physics/commit/b9b3ffd552f3c91df519c37f88a3fc7ebb9a7acc)

#### `gz-sim`

- [ ]  Update the `JointPositionController` to search using both scoped and unscoped joint names.  [gz-sim: 61bca3d](https://github.com/srmainwaring/gz-sim/commit/61bca3d5ffaf4079caee5f70158ed8f6ef89f766)
### Outstanding issues

- [ ] ECM Entity Id is not correctly mapped to Physics Id when the container and nested model contain joints with the same unscoped name.
- [ ] Joints between two nested models are not resolved correctly.

srmainwaring, 2023-02-01:
@azeey this should still be open as it requires https://github.com/gazebosim/gz-sim/pull/1851 to be merged as well.

PR #1851 (2023-02-08): Support nested joints in JointPositionController
# 🦟 Bug fix

Partially Fixes #1844

## Summary

The details of the bug and how to reproduce it are documented in [gz-sim: #1844](https://github.com/gazebosim/gz-sim/issues/1844).

This PR enables the `JointPositionController` plugin to search for scoped joint names as well as unscoped ones. The approach is similar to that used in the Buoyancy plugin. 

For the change to take effect the upstream change in https://github.com/gazebosim/gz-physics/pull/464 is also required.

## Context

The use case for this change is to be able to include standalon [... cut]
closed 2023-02-08 by azeey