gazebosim/gz-sim #1845: Write a tutorial about using transport parameters with system

scpeters, 2022-12-21:
## Desired behavior

A new feature was added in https://github.com/gazebosim/gz-transport/pull/305 to declare and use parameters. An interface to this functionality was added for gz-sim systems in https://github.com/gazebosim/gz-sim/pull/1431, with a demo in the [parameters-component-wheel-slip-demo6](https://github.com/gazebosim/gz-sim/compare/ign-gazebo6...ivanpauno/parameters-component-wheel-slip-demo6) branch and https://github.com/gazebosim/gz-msgs/commit/e19f8e5dddd27106ef874cccd38590b65acd81f0. This demo should be merged to a release branch along with a tutorial.

## Alternatives considered

## Implementation suggestion

Demo instructions from #305:

Steps to run the demo:

- Build from source ignition `fortress` checking out the two branches specified above (I can provide PRs for another version if needed).
- Launch the trisphere cycle world:
  ```
  ign gazebo trisphere_cycle_wheel_slip.sdf
  ```
- In another terminal with the workspace sourced:
  - List available parameters:
    ```
    ign param -r /world/wheel_slip -l
    ```
    <details>
      <summary>Expected output</summary>

      > systems.wheel_slip.trisphere_cycle1.wheel_front            [ignition.msgs.WheelSlipParameters]
      > systems.wheel_slip.trisphere_cycle1.wheel_rear_right            [ignition.msgs.WheelSlipParameters]
      > systems.wheel_slip.trisphere_cycle0.wheel_rear_right            [ignition.msgs.WheelSlipParameters]
      > systems.wheel_slip.trisphere_cycle1.wheel_rear_left            [ignition.msgs.WheelSlipParameters]
      > systems.wheel_slip.trisphere_cycle0.wheel_rear_left            [ignition.msgs.WheelSlipParameters]
      > systems.wheel_slip.trisphere_cycle0.wheel_front            [ignition.msgs.WheelSlipParameters]

    </details>
  - Get the `systems.wheel_slip.trisphere_cycle1.wheel_front` parameter:
    ```
    ign param -r /world/wheel_slip -g -n systems.wheel_slip.trisphere_cycle1.wheel_front
    ```
    <details>
      <summary>Expected output</summary>

      >
      > Getting parameter [systems.wheel_slip.trisphere_cycle1.wheel_front] for registry namespace [/world/wheel_slip]...
      > Parameter type [ign_msgs.WheelSlipParameters]
      >
      > ------------------------------------------------
      > slip_compliance_lateral: 1
      > slip_compliance_longitudinal: 1
      > ------------------------------------------------

    </details>
  - Set the `systems.wheel_slip.trisphere_cycle1.wheel_front` parameter to a different value:
    ```
    ign param -r /world/wheel_slip -s -n systems.wheel_slip.trisphere_cycle1.wheel_front -t ign_msgs.WheelSlipParameters -m "
      slip_compliance_lateral: 1
      slip_compliance_longitudinal: 2"
    ```
    <details>
      <summary>Expected output</summary>

      >
      > Setting parameter [systems.wheel_slip.trisphere_cycle1.wheel_front] for world [wheel_slip]...
      >
      > Parameter successfully set!

    </details>
  - Get the `systems.wheel_slip.trisphere_cycle1.wheel_front` parameter again, its value should have changed:
    ```
    ign param -r /world/wheel_slip -g -n systems.wheel_slip.trisphere_cycle1.wheel_front
    ```
    <details>
      <summary>Expected output</summary>

      >
      > Getting parameter [systems.wheel_slip.trisphere_cycle1.wheel_front] for registry namespace [/world/wheel_slip]...
      > Parameter type [ign_msgs.WheelSlipParameters]
      >
      > ------------------------------------------------
      > slip_compliance_lateral: 1
      > slip_compliance_longitudinal: 2
      > ------------------------------------------------

    </details>

## Additional context

Gaurang-1402, 2024-04-01:
Hi @scpeters and @azeey I have made a PR to the Gazebo tutorials repo that resolves this issue: https://github.com/osrf/gazebo_tutorials/pull/175

I'm curious to hear your feedback and improvise!

knmcguire, 2024-08-05:
So just to add on that (@Gaurang-1402 noticed it first on his PR), is that `ign param -r /world/wheel_slip -l` lists no parameters in this example. I confirm that this is also the case with gazebo harmonic with `gz param -r /world/wheel_slip -l`

I'm currently trying to build harmonic from source to see if this is still the case on the latest development branch but if this is the case than I assume we need to make an issue on gz-transport github repo right?

azeey, 2024-08-06:
The description mentions that the demo is in the [parameters-component-wheel-slip-demo6](https://github.com/gazebosim/gz-sim/compare/ign-gazebo6...ivanpauno/parameters-component-wheel-slip-demo6) branch along with a specific commit in gz-msgs. Can you give that a try?

knmcguire, 2024-08-06:
yes... I should really learn to read the whole description in not a hurry. Thanks for notifying me.

 Absolutely I'll try that out as soon as I manage to build source without my pc freezing. I'll report back once I do.

knmcguire, 2024-08-06:
wait, it seems that that branch was made from the gz-sim-6 era so that's probably not going to work unless I rebase the changes (if that is still possible).

Perhaps the easiest now is I'll use it as a reference to work on the example for the multicopter velocity control tuning that I've been working on: https://github.com/knmcguire/gz-plugin-multicopter-velocity-tuning. Bit of 2 birds one stone scenario since I'll need to work on this anyway for my work's project

knmcguire, 2024-08-06:
or option 3.... I look in the code of gz_sim8 and find out that most of the branch has already been implemented :smile: 

 Except for the wheel-slip plugin, of which I made a draft pull request for: https://github.com/gazebosim/gz-sim/pull/2518. There are still some iffy things happening so I'd need some help with that, but once that is merged than we are one step closer of making this tutorial a reality at least.