gazebosim/gz-sim #1864: Ubuntu 22.04 running on Parallels on a M1 Mac

ericrohloff, 2023-01-22:
## Environment
* OS Version: 
* Source or binary build?
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
# paste log here
```

</details>

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1.
2.
3.

## Output

Bezalelohim, 2023-02-01:
What are the commands you used to run ?

Bezalelohim, 2023-02-01:
Which version of Gazebo ?

Bezalelohim, 2023-02-01:
Try running `gz sim [NAME OF SDF].sdf  -v 4` for a verbose output you can see what is wrong in the terminal

BiggestRic, 2023-02-14:
> Try running `gz sim [NAME OF SDF].sdf -v 4` for a verbose output you can see what is wrong in the terminal

The errors I encounter are quite extensive as there are many
The first error though is: [Wrn] [gz.cc:100] Fuel world download failed because Fetch failed. Other errors
What does this signify?