gazebosim/gz-sim #1865: brew install gz-garden -> SHA256 mismatch

danielkroeni, 2023-01-23:
## Environment
* OS Version: macOS Ventura 13.1 M1 Max

I tried to install gz-garden using homebrew:

```
> brew info gz-garden
==> osrf/simulation/gz-garden: stable 1.0.0, HEAD
> brew install gz-garden
```

<details>

```
==> Fetching osrf/simulation/gz-rendering7
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-rendering/releases/gz-rendering-7.2.0.tar.bz2
Already downloaded: /Users/dk/Library/Caches/Homebrew/downloads/63d2ae5993e499b8c6180a34ca0441e78cae3b970efbd382287768f46e45bdc2--gz-rendering-7.2.0.tar.bz2
==> Fetching osrf/simulation/gz-transport12
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-transport/releases/gz-transport-12.1.0~pre1.tar.bz2
######################################################################## 100.0%
Error: gz-garden: SHA256 mismatch
Expected: 52aca2b3eb85d0eaf35871866254bfbf0a4d69857a90f96f710b90e4306334e3
  Actual: 0cd5adcefa2e892177af8d435d813384d1cd6983ccf3d8546d9b87ea6142725b
    File: /Users/dk/Library/Caches/Homebrew/downloads/04fbb6d1454f508ab72637aca716cb176a7dc12291ad7776c488ccf07cfd7c41--gz-transport-12.1.0~pre1.tar.bz2
To retry an incomplete download, remove the file above.
```

</details>

## Description
* Expected behavior: Successful installation
* Actual behavior: An error regarding "SHA256 mismatch" 

I deleted the file, rerun `brew install gz-garden` and got again the same error.

Thank you for your time!

azeey, 2023-02-06:
We've rebuilt bottles since then. Would you mind trying it again?

danielkroeni, 2023-02-07:
Thank you for your reply. 

Now I get an error when linking `sdformat13`. 
It is the same error as "reported" here: https://community.gazebosim.org/t/installing-gazebo-on-m1-ventura-13-1/1748/1

Command:
```
brew install -vd gz-garden
```

Error:
```
[ 17%] Linking CXX shared library ../lib/libsdformat13.dylib
cd "/tmp/sdformat13-20230207-50912-1ucnlpp/sdformat-13.3.0~pre1/build/src" && /opt/homebrew/Cellar/cmake/3.25.2/bin/cmake -E cmake_link_script CMakeFiles/sdformat13.dir/link.txt --verbose=1
/opt/homebrew/Library/Homebrew/shims/mac/super/clang++  -stdlib=libc++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -dynamiclib -Wl,-headerpad_max_install_names  -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 13.0.0 -current_version 13.3.0 -o "../lib/libsdformat13.13.3.0~pre1.dylib" -install_name @rpath/libsdformat13.13.dylib CMakeFiles/sdformat13.dir/Actor.cc.o CMakeFiles/sdformat13.dir/AirPressure.cc.o CMakeFiles/sdformat13.dir/Altimeter.cc.o CMakeFiles/sdformat13.dir/Atmosphere.cc.o CMakeFiles/sdformat13.dir/Box.cc.o CMakeFiles/sdformat13.dir/Camera.cc.o CMakeFiles/sdformat13.dir/Capsule.cc.o CMakeFiles/sdformat13.dir/Collision.cc.o CMakeFiles/sdformat13.dir/Console.cc.o CMakeFiles/sdformat13.dir/Converter.cc.o CMakeFiles/sdformat13.dir/Cylinder.cc.o CMakeFiles/sdformat13.dir/Element.cc.o CMakeFiles/sdformat13.dir/Ellipsoid.cc.o CMakeFiles/sdformat13.dir/Error.cc.o CMakeFiles/sdformat13.dir/Exception.cc.o CMakeFiles/sdformat13.dir/Filesystem.cc.o CMakeFiles/sdformat13.dir/ForceTorque.cc.o CMakeFiles/sdformat13.dir/Frame.cc.o CMakeFiles/sdformat13.dir/FrameSemantics.cc.o CMakeFiles/sdformat13.dir/Geometry.cc.o CMakeFiles/sdformat13.dir/Gui.cc.o CMakeFiles/sdformat13.dir/Heightmap.cc.o CMakeFiles/sdformat13.dir/Imu.cc.o CMakeFiles/sdformat13.dir/InterfaceElements.cc.o CMakeFiles/sdformat13.dir/InterfaceFrame.cc.o CMakeFiles/sdformat13.dir/InterfaceJoint.cc.o CMakeFiles/sdformat13.dir/InterfaceLink.cc.o CMakeFiles/sdformat13.dir/InterfaceModel.cc.o CMakeFiles/sdformat13.dir/InterfaceModelPoseGraph.cc.o CMakeFiles/sdformat13.dir/Joint.cc.o CMakeFiles/sdformat13.dir/JointAxis.cc.o CMakeFiles/sdformat13.dir/Lidar.cc.o CMakeFiles/sdformat13.dir/Light.cc.o CMakeFiles/sdformat13.dir/Link.cc.o CMakeFiles/sdformat13.dir/Magnetometer.cc.o CMakeFiles/sdformat13.dir/Material.cc.o CMakeFiles/sdformat13.dir/Mesh.cc.o CMakeFiles/sdformat13.dir/Model.cc.o CMakeFiles/sdformat13.dir/NavSat.cc.o CMakeFiles/sdformat13.dir/Noise.cc.o CMakeFiles/sdformat13.dir/OutputConfig.cc.o CMakeFiles/sdformat13.dir/Param.cc.o CMakeFiles/sdformat13.dir/ParamPassing.cc.o CMakeFiles/sdformat13.dir/ParserConfig.cc.o CMakeFiles/sdformat13.dir/ParticleEmitter.cc.o CMakeFiles/sdformat13.dir/Pbr.cc.o CMakeFiles/sdformat13.dir/Physics.cc.o CMakeFiles/sdformat13.dir/Plane.cc.o CMakeFiles/sdformat13.dir/Plugin.cc.o CMakeFiles/sdformat13.dir/Polyline.cc.o CMakeFiles/sdformat13.dir/PrintConfig.cc.o CMakeFiles/sdformat13.dir/Root.cc.o CMakeFiles/sdformat13.dir/SDF.cc.o CMakeFiles/sdformat13.dir/SDFExtension.cc.o CMakeFiles/sdformat13.dir/Scene.cc.o CMakeFiles/sdformat13.dir/SemanticPose.cc.o CMakeFiles/sdformat13.dir/Sensor.cc.o CMakeFiles/sdformat13.dir/Sky.cc.o CMakeFiles/sdformat13.dir/Sphere.cc.o CMakeFiles/sdformat13.dir/Surface.cc.o CMakeFiles/sdformat13.dir/Types.cc.o CMakeFiles/sdformat13.dir/Utils.cc.o CMakeFiles/sdformat13.dir/Visual.cc.o CMakeFiles/sdformat13.dir/World.cc.o CMakeFiles/sdformat13.dir/XmlUtils.cc.o CMakeFiles/sdformat13.dir/gz.cc.o CMakeFiles/sdformat13.dir/parser.cc.o CMakeFiles/sdformat13.dir/parser_urdf.cc.o CMakeFiles/sdformat13.dir/EmbeddedSdf.cc.o  /opt/homebrew/lib/libgz-math7.7.1.0.dylib /opt/homebrew/lib/libgz-utils2.2.0.0.dylib /opt/homebrew/Cellar/tinyxml2/9.0.0/lib/libtinyxml2.dylib /opt/homebrew/Cellar/tinyxml2/9.0.0/lib/libtinyxml2.dylib /opt/homebrew/opt/urdfdom/lib/liburdfdom_sensor.dylib /opt/homebrew/opt/urdfdom/lib/liburdfdom_sensor.dylib /opt/homebrew/opt/urdfdom/lib/liburdfdom_model_state.dylib /opt/homebrew/opt/urdfdom/lib/liburdfdom_model.dylib /opt/homebrew/opt/urdfdom/lib/liburdfdom_world.dylib /opt/homebrew/Cellar/console_bridge/1.0.2/lib/libconsole_bridge.dylib 
cd "/tmp/sdformat13-20230207-50912-1ucnlpp/sdformat-13.3.0~pre1/build/src" && /opt/homebrew/Cellar/cmake/3.25.2/bin/cmake -E cmake_symlink_library "../lib/libsdformat13.13.3.0~pre1.dylib" ../lib/libsdformat13.13.dylib ../lib/libsdformat13.dylib
[ 17%] Built target sdformat13
make: *** [all] Error 2
/opt/homebrew/Library/Homebrew/shims/shared/git --version
/opt/homebrew/Library/Homebrew/shims/shared/curl --version
/opt/homebrew/Library/Homebrew/ignorable.rb:29:in `block in raise'
BuildError: Failed executing: make install
1. raise
2. ignore
3. backtrace
4. irb
5. shell
Choose an action: 1
==> Temporary files retained at:
/private/tmp/sdformat13-20230207-50912-1ucnlpp

==> Formula
Tap: osrf/simulation
Path: /opt/homebrew/Library/Taps/osrf/homebrew-simulation/Formula/sdformat13.rb
==> Configuration
HOMEBREW_VERSION: 3.6.21
ORIGIN: https://github.com/Homebrew/brew
HEAD: 932d2cf3b77c9439a57b6a43577fc8d3b6399a62
Last commit: 28 minutes ago
Core tap origin: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 5a89d6b8e83286c858197f731310163e4108fcc3
Core tap last commit: 33 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.9X0ptCJvHp/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
[... cut ...]
/opt/homebrew/Library/Homebrew/shims/shared/curl --disable --cookie /dev/null --globoff --show-error --user-agent Homebrew/3.6.21\ \(Macintosh\;\ arm64\ Mac\ OS\ X\ 13.2\)\ curl/7.86.0 --header Accept-Language:\ en --retry 3 --location https://api.github.com/search/issues\?q=sdformat13\+repo\%3Aosrf\%2Fhomebrew-simulation\+state\%3Aopen\+in\%3Atitle\&per_page=100 --header Accept:\ application/vnd.github\+json --write-out '
'\%\{http_code\} --header Authorization:\ token\ ****** --header X-GitHub-Api-Version:2022-11-28 --dump-header /private/tmp/github_api_headers20230207-49996-etijo
➜  ~ 
```

azeey, 2023-02-07:
@scpeters do you know if sdformat works on M1s?

scpeters, 2023-02-07:
I'm guessing we need https://github.com/gazebosim/sdformat/pull/1216 to be released

time to make an sdformat 13.3.0

iche033, 2023-02-07:
I was about to make a sdformat release for gz-sensors. 

Just created PR: https://github.com/gazebosim/sdformat/pull/1229

danielkroeni, 2023-02-08:
Hi all

I just tried my luck again - and the installation worked!
Thank you all very much!

But when launching a simulation with the following command:
```
gz sim -s shapes.sdf
```
I got the following error:
```
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <EC5BB0F6-E93D-3AC0-8176-B1B1EA59B438> /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
```

That's the same issue as reported here:
https://answers.gazebosim.org/question/28548/does-gazebo-work-on-mac-m1-monterey-126/

I am going to search the web regarding this macOS security policy thing.

Thank you again.

azeey, 2023-02-08:
The issue with the  OSX System Integrity Protection (SIP) is addressed in https://gazebosim.org/docs/garden/troubleshooting#unable-to-load-dylib-file. Can you give that a try?

danielkroeni, 2023-02-08:
Thank you for the link. I followed the instructions and installed ruby:
```
> brew install ruby
Warning: ruby 3.2.0 is already installed and up-to-date.
To reinstall 3.2.0, run:
  brew reinstall ruby
```

So I tried it with this 3.2.0 version by adding it to the `PATH` in my `.zshrc` and sourced it.
(in the troubleshooting guide, `.bashrc` and older brew locations are used.)

This led to the following error:
```
> gz sim -s shapes.sdf           
Library error: Problem running [sim]() from /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/libgz-sim7-gz.7.3.0.dylib.
```

Then I tried again with ruby version 2.7: https://formulae.brew.sh/formula/ruby
```
> brew install ruby@2.7          
...
```
I extended the `PATH` variable, sourced again and launched the simulation:

```
> gz sim -s shapes.sdf                                                               
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <EC5BB0F6-E93D-3AC0-8176-B1B1EA59B438> /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <FA92DCCC-A54B-3516-B774-ACAB058AB860> /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim7-user-commands-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::UserCommands]
- Requested library name: [gz-sim-user-commands-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <3AD916DD-378E-31BF-A13D-2D4DA7A7FFAB> /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim7-scene-broadcaster-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::SceneBroadcaster]
- Requested library name: [gz-sim-scene-broadcaster-system]
[... cut ...] sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::SceneBroadcaster]
- Requested library name: [gz-sim-scene-broadcaster-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.3.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]
```

Same problem with ruby@3.1

Do you have any other ideas?

srmainwaring, 2023-02-08:
Hi @danielkroeni - this looks like a dylib path issue. `gz-sim7` definitely runs on M1 Monterey 12.6.2 - I always build from source because I want to debug through the libs, so if you are open to trying a source build I may be able to help.

They key is to prevent `@rpath` being written into the libraries when linking (if you don't want to disable SIP). The binaries are not relocatable, but that does not matter with a source build. Follow the 'from source' instructions and then use the build command:

```bash
colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=ON -DCMAKE_CXX_STANDARD=17 -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-up-to gz-sim7
```

There are a couple of patches I apply for Metal rendering (start with the standard build first to check it's running first though):

1. https://github.com/srmainwaring/gz-rendering/tree/srmainwaring/7/gz-rendering7-metal

This reverses a z-order change which was causing objects to hide behind the ground plane. May not be needed now.

2. https://github.com/srmainwaring/gz-gui/tree/gz-gui7-metal

Contains a custom QML gamma adjustment that works with Metal - stops the scene looking very dark. The OpenGL version does not work with Metal.

3. https://github.com/srmainwaring/gz-sim/tree/srmainwaring/7/gz-sim7-metal

Reverts a change to the visual plugin broadcasting. Specialised use if developing custom visuals - this stops them being loaded twice (not good for a sea state simulation).

You won't need these to start, and perhaps only (2) if you find the default lighting too dark.

scpeters, 2023-02-08:
> ```
> > gz sim -s shapes.sdf           
> Library error: Problem running [sim]() from /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/libgz-sim7-gz.7.3.0.dylib.
> ```

I believe this is due to a bug recently fixed in https://github.com/gazebosim/gz-sim/pull/1884. Can you try `brew install gz-sim --HEAD`? You may need to remove it first

iche033, 2023-02-09:
I recently updated the macOS install from source doc page for Garden:
https://github.com/gazebosim/docs/blob/master/garden/install_osx_src.md

If anything does not work, I would be interested in fixing and updating the doc.

@srmainwaring, we welcome those metal patches if you have the time :)

danielkroeni, 2023-02-09:
@scpeters running the proposed command did not work:
```
brew install gz-sim --HEAD
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
kuttl                                 libsolv                               micromamba                            tl-expected
==> New Casks
nanosaur2

You have 10 outdated formulae installed.
You can upgrade them with brew upgrade
or list them with brew outdated.

Warning: No available formula with the name "gz-sim". Did you mean gz-sim8, gz-sim7 or gpsim?
==> Searching for similarly named formulae and casks...
==> Formulae
osrf/simulation/gz-sim3       osrf/simulation/gz-sim6       osrf/simulation/gz-sim7 ✔     osrf/simulation/gz-sim8       gpsim

To install osrf/simulation/gz-sim3, run:
  brew install osrf/simulation/gz-sim3
```

But the installation went through without error using the following command:
```
> brew install gz-garden --HEAD
```

```
> brew install gz-garden --HEAD
==> Fetching dependencies for osrf/simulation/gz-garden: gz-sim7 and gz-launch6
==> Fetching osrf/simulation/gz-sim7
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-sim/releases/gz-s
######################################################################## 100.0%
==> Fetching osrf/simulation/gz-launch6
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-launch/releases/g
Already downloaded: /Users/dk/Library/Caches/Homebrew/downloads/aad1cc9c31279b181803f75bc82f6be9bc492672f84a3c06b327db3d558c8578--gz-launch-6.0.0.tar.bz2
==> Fetching osrf/simulation/gz-garden
==> Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f950
Already downloaded: /Users/dk/Library/Caches/Homebrew/downloads/5e8fd956f9e881b791ed69ea43da80fd61a81bfb2cfefd2f777fbddcc2ec53a0--PyYAML-5.3.1.tar.gz
==> Cloning https://github.com/gazebosim/gz-garden.git
Cloning into '/Users/dk/Library/Caches/Homebrew/gz-garden--git'...
==> Checking out branch main
Already on 'main'
Your branch is up to date with 'origin/main'.
==> Installing gz-garden from osrf/simulation
==> Installing dependencies for osrf/simulation/gz-garden: gz-sim7 and gz-launch6
==> Installing osrf/simulation/gz-garden dependency: gz-sim7
==> cmake .. -DCMAKE_INSTALL_RPATH=@loader_path/../lib
==> make install
🍺  /opt/homebrew/Cellar/gz-sim7/7.3.0: 1,456 files, 74.9MB, built in 1 minute 22 seconds
==> Installing osrf/simulation/gz-garden dependency: gz-launch6
==> cmake .. -DCMAKE_INSTALL_RPATH=@loader_path/../lib
==> make install
🍺  /opt/homebrew/Cellar/gz-launch6/6.0.0_1: 43 files, 1.2MB, built in 21 seconds
==> Installing osrf/simulation/gz-garden --HEAD
==> cmake ..
==> make install
==> /opt/homebrew/opt/python@3.9/libexec/bin/python3 -m venv --system-site-packa
==> /opt/homebrew/Cellar/gz-garden/HEAD-a4c7fba/libexec/bin/pip install -v --no-
==> /opt/homebrew/Cellar/gz-garden/HEAD-a4c7fba/libexec/bin/pip install -v --no-
🍺  /opt/homebrew/Cellar/gz-garden/HEAD-a4c7fba: 1,575 files, 17.7MB, built in 1 minute
==> Running `brew cleanup gz-garden`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Removing: /Users/dk/Library/Caches/Homebrew/gz-garden--1.0.0.tar.bz2... (2.5KB)
```

But running the application failed again.
```
➜  src gz sim -s shapes.sdf
Library error: Problem running [sim]() from /opt/homebrew/Cellar/gz-sim7/7.3.0/lib/libgz-sim7-gz.7.3.0.dylib.
➜  src ruby --version
ruby 3.2.1 (2023-02-08 revision 31819e82c8) [arm64-darwin22]
```

danielkroeni, 2023-02-09:
@iche033 Thank you!

I have the `brew 3.6.21` installed:
```
> brew --version
Homebrew 3.6.21
Homebrew/homebrew-core (git revision e364112ad3a; last commit 2023-02-09)
Homebrew/homebrew-cask (git revision 4a6eeca1f7; last commit 2023-02-09)
```

```
> brew cask install xquartz
Error: `brew cask` is no longer a `brew` command. Use `brew <command> --cask` instead.
```
The correct command for my brew version is:
```
brew install --cask xquartz
```

And the build went through:
```
Summary: 16 packages finished [17min 8s]
  16 packages had stderr output: gz-cmake3 gz-common5 gz-fuel_tools8 gz-gui7 gz-launch6 gz-math7 gz-msgs9 gz-physics6 gz-plugin2 gz-rendering7 gz-sensors7 gz-sim7 gz-tools2 gz-transport12 gz-utils2 sdformat13
[1027.735s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/opt/homebrew'
```

I rechecked that I ran the following command:
```
> python3 -m pip install -U colcon-common-extensions
```
which confirmed, that `Requirement already satisfied`.
So I ignored the error.

Then I launched the server and the client:
A lot of green messages in the terminal appeared and the UI showed up.
But then the rotating rainbow colored wait icon appeared and the following errors are shown:
```
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
Error while loading the library [/Users/dk/workspace/install/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.dylib]: dlopen(/Users/dk/workspace/install/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.dylib, 0x0005): Library not loaded: @rpath/libOgrePlanarReflections.2.3.1.dylib
  Referenced from: <B7ACBD12-7543-3820-AC3E-A928F8ED5F79> /opt/homebrew/Cellar/ogre2.3/2.3.1_1/lib/libOgreHlmsPbs.2.3.1.dylib
  Reason: tried: '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgrePlanarReflections.2.3.1.dylib' (no such file), '/usr/lib/libOgrePlanarReflections.2.3.1.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)Library not loaded: @rpath/libOgreMain.2.3.1.dylib
  Referenced from: <FE636C61-05A2-304B-B522-570D40EAF5BB> /opt/homebrew/Cellar/ogre2.3/2.3.1_1/lib/libOgreHlmsUnlit.2.3.1.dylib
  Reason: tried: '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libOgreMain.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreMain.2.3.1.dylib' (no such file), '/usr/lib/libOgreMain.2.3.1.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)Library not loaded: @rpath/libOgreHlmsUnlit.2.3.1.dylib
  Referenced from: <A1D8CE26-F193-38B1-9AF8-30880FC40F2B> /opt/homebrew/Cellar/ogre2.3/2.3.1_1/lib/libOgreOverlay.2.3.1.dylib
  Reason: tried: '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/ogre2.3/lib/OGRE-2.3/libOgreHlmsUnlit.2.3.1.dylib' (no such file), '/usr/lib/libOgreHlmsUnlit.2.3.1.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)Library not loaded: @rpath/libOgreMain.2.3.1.dylib
[... cut ...]
#8    Object "libsystem_malloc.dylib", at 0x185d576a3, in nanov2_allocate_from_block + 363
#7    Object "libsystem_malloc.dylib", at 0x185d730df, in nanov2_guard_corruption_detected + 43
#6    Object "libsystem_malloc.dylib", at 0x185d7b403, in malloc_zone_error + 99
#5    Object "libsystem_malloc.dylib", at 0x185d649f7, in malloc_vreport + 907
#4    Object "libsystem_c.dylib", at 0x185e462c7, in abort + 179
#3    Object "libsystem_pthread.dylib", at 0x185f0dceb, in pthread_kill + 287
#2    Object "libsystem_platform.dylib", at 0x185f3c2a3, in _sigtramp + 55
```

iche033, 2023-02-09:
hmm I recently submitted a patch that's supposed to fix the rpath issue: https://github.com/osrf/homebrew-simulation/pull/2183

when did you install ogre 2.3?

danielkroeni, 2023-02-09:
On line 33 there is:
```
"-DCMAKE_INSTALL_RPATH=#{rpath}",
```
which seems to be expected by https://github.com/osrf/homebrew-simulation/pull/2183

iche033, 2023-02-09:
one thing to try is change that line to `"-DCMAKE_MACOSX_RPATH:BOOL=FALSE",` and reinstall ogre2.3:

```
brew reinstall ogre2.3
```

and maybe rebuild gz-rendering7

```
colcon build --cmake-args -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --merge-install --packages-select gz-rendering7 --cmake-clean-cache
```

scpeters, 2023-02-10:
> @scpeters running the proposed command did not work:
> 
> ```
> brew install gz-sim --HEAD

apologies! I meant to suggest `brew install gz-sim7 --HEAD`

danielkroeni, 2023-02-11:
@iche033 
I replaced line 33 with the one you suggested and executed the following steps:
```
> nvim /opt/homebrew/Library/Taps/osrf/homebrew-simulation/Formula/ogre2.3.rb
>  brew reinstall ogre2.3
==> Fetching osrf/simulation/ogre2.3
==> Downloading https://github.com/OGRECave/ogre-next/archive/refs/tags/v2.3.1.tar.gz
Already downloaded: /Users/dk/Library/Caches/Homebrew/downloads/934d985a51c822920a6ba97a12c2c0c3cb158c69da77b31188e86eb50f4075e4--ogre-next-2.3.1.tar.gz
==> Reinstalling osrf/simulation/ogre2.3 
==> cmake .. -DCMAKE_CXX_STANDARD=11 -DCMAKE_CXX_STANDARD_REQUIRED:BOOL=ON -DCMAKE_CXX_EXTENSIONS:BOOL=ON -DCMAKE_MACOSX_RPATH:B
==> make install
🍺  /opt/homebrew/Cellar/ogre2.3/2.3.1_1: 860 files, 16.4MB, built in 47 seconds
==> Running `brew cleanup ogre2.3`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
> cd workspace/src 
> colcon build --cmake-args -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --merge-install --packages-select gz-rendering7 --cmake-clean-cache
Starting >>> gz-rendering7
[0.603s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /Users/dk/workspace/src/install/share/gz-cmake3/package.sh
- /Users/dk/workspace/src/install/share/gz-utils2/package.sh
- /Users/dk/workspace/src/install/share/gz-math7/package.sh
- /Users/dk/workspace/src/install/share/gz-common5/package.sh
- /Users/dk/workspace/src/install/share/gz-plugin2/package.sh
Check that the following packages have been built:
- gz-cmake3
- gz-utils2
- gz-math7
- gz-common5
- gz-plugin2
Failed   <<< gz-rendering7 [0.01s, exited with code 1]

Summary: 0 packages finished [0.36s]
  1 package failed: gz-rendering7
[0.606s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/opt/homebrew'
```

So I tried the command to build the whole application:
```
> colcon build --cmake-args -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --merge-install
```
The build went through!

The server launched:
```
gz sim -v 4 shapes.sdf -s
[Msg] Gazebo Sim Server v7.3.0
[Msg] Loading SDF world file[/Users/dk/workspace/install/share/gz/gz-sim7/worlds/shapes.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:1042] Loaded (3) plugins from file [/Users/dk/.gz/sim/7/server.config]
[Dbg] [Physics.cc:869] Loaded [gz::physics::dartsim::Plugin] from library [/Users/dk/workspace/install/lib/gz-physics-6/engine-plugins/libgz-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Msg] Create service on [/world/shapes/create]
[Msg] Remove service on [/world/shapes/remove]
[Msg] Pose service on [/world/shapes/set_pose]
[Msg] Pose service on [/world/shapes/set_pose_vector]
[Msg] Light configuration service on [/world/shapes/light_config]
[Msg] Physics service on [/world/shapes/set_physics]
[Msg] SphericalCoordinates service on [/world/shapes/set_spherical_coordinates]
[Msg] Enable collision service on [/world/shapes/enable_collision]
[Msg] Disable collision service on [/world/shapes/disable_collision]
[Msg] Material service on [/world/shapes/visual_config]
[Msg] Material service on [/world/shapes/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Msg] Serving world controls on [/world/shapes/control], [/world/shapes/control/state] and [/world/shapes/playback/control]
[Msg] Serving GUI information on [/world/shapes/gui/info]
[Msg] World [shapes] initialized with [default_physics] physics profile.
[... 155 lines cut ...]
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:845] [QT] file::/WorldStats/WorldStats.qml:53:3: QML RowLayout: Binding loop detected for property "x"
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:845] [QT] qUncompress: Z_DATA_ERROR: Input data is corrupted
[GUI] [Wrn] [Application.cc:845] [QT] Attempted to deserialize QShader with unknown version 0.
[GUI] [Wrn] [Application.cc:845] [QT] ShaderEffect: Failed to deserialize QShader from :/qt-project.org/imports/QtGraphicalEffects/shaders/gammaadjust.frag
[GUI] [Wrn] [Application.cc:845] [QT] ShaderEffect: shader preparation failed for qrc:/qt-project.org/imports/QtGraphicalEffects/shaders/gammaadjust.frag

[GUI] [Dbg] [MinimalScene.cc:664] Creating gz-renderering interface for Metal
[GUI] [Dbg] [MinimalScene.cc:819] Creating render thread interface for Metal
[... cut ...]
#4    Object "AGXMetalG13X", at 0x1d0beacc3, in AGXMetalG13X + 2305219
#3    Object "AGXMetalG13X", at 0x1d0be8f57, in AGXMetalG13X + 2297687
#2    Object "libsystem_platform.dylib", at 0x185f3c2a3, in _sigtramp + 55
#1    Object "libgz-tools2-backward.dylib", at 0x102b13587, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#0    Object "libgz-tools2-backward.dylib", at 0x102b135df, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 59
[1]    8815 segmentation fault  gz sim -v 4 -g
```

danielkroeni, 2023-02-11:
@scpeters 

Thank you. I just installed it using `brew install gz-sim7 --HEAD` and it seems to work. The installation went through but when launching it reports the following error:

```
> gz sim -s shapes.sdf
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <345AA966-2106-3628-B664-9C7CFF608D4A> /opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <DF95208C-7702-38B1-A687-D8822F7B584C> /opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim7-user-commands-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::UserCommands]
- Requested library name: [gz-sim-user-commands-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <A4432A14-D86B-3B2F-91D1-D5E6A55F3A08> /opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim7-scene-broadcaster-system.7.3.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::SceneBroadcaster]
- Requested library name: [gz-sim-scene-broadcaster-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/HEAD-5881559/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]
```

iche033, 2023-02-13:
I also ran into a gui crash recently: https://github.com/gazebosim/gz-gui/issues/513. For me, one of these steps resolved the issue for me:
* remove ~/.gz/sim/7/gui.config
* comment out Component Inspector plugin in ~/.gz/sim/7/gui.config or in the world file if it's specified there, e.g. https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/minimal_scene.sdf#L254
* clean and rebuild entire gz workspace

I noticed that there are still some crashes when using certain gui features though, e.g. closing a gui plugin window causes a crash for me.

danielkroeni, 2023-02-21:
@iche033 sorry for the late answer

After commenting out the ComponentInspector, the application launched:

```
> gz sim -s shapes.sdf -v 4 
[Msg] Gazebo Sim Server v7.3.0
[Msg] Loading SDF world file[/Users/dk/workspace/install/share/gz/gz-sim7/worlds/shapes.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:1042] Loaded (3) plugins from file [/Users/dk/.gz/sim/7/server.config]
[Dbg] [Physics.cc:869] Loaded [gz::physics::dartsim::Plugin] from library [/Users/dk/workspace/install/lib/gz-physics-6/engine-plugins/libgz-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Msg] Create service on [/world/shapes/create]
[Msg] Remove service on [/world/shapes/remove]
[Msg] Pose service on [/world/shapes/set_pose]
[Msg] Pose service on [/world/shapes/set_pose_vector]
[Msg] Light configuration service on [/world/shapes/light_config]
[Msg] Physics service on [/world/shapes/set_physics]
[Msg] SphericalCoordinates service on [/world/shapes/set_spherical_coordinates]
[Msg] Enable collision service on [/world/shapes/enable_collision]
[Msg] Disable collision service on [/world/shapes/disable_collision]
[Msg] Material service on [/world/shapes/visual_config]
[Msg] Material service on [/world/shapes/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Msg] Serving world controls on [/world/shapes/control], [/world/shapes/control/state] and [/world/shapes/playback/control]
[Msg] Serving GUI information on [/world/shapes/gui/info]
[Msg] World [shapes] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/shapes/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:513] Creating PostUpdate worker threads: 2
[Dbg] [SimulationRunner.cc:524] Creating postupdate worker thread (0)
[Msg] Serving scene information on [/world/shapes/scene/info]
[Msg] Serving graph information on [/world/shapes/scene/graph]
[Msg] Serving full state on [/world/shapes/state]
[Msg] Serving full state (async) on [/world/shapes/state_async]
[Msg] Publishing scene information on [/world/shapes/scene/info]
[Msg] Publishing entity deletions on [/world/shapes/scene/deletion]
[Msg] Publishing state changes on [/world/shapes/state]
[Msg] Publishing pose messages on [/world/shapes/pose/info]
[Msg] Publishing dynamic pose messages on [/world/shapes/dynamic_pose/info]
[Dbg] [EntityComponentManager.cc:1619] Updated state thread iterators: 10 threads processing around 3 entities each.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] [Wrn] [Model.hh:69] [Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldESkipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models `operator<<`. Component will not be serialized.

[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.with //pose/@relative_to attribute.
with //pose/@relative_to attribute.
```

```
> gz sim -v 4 -g              
[Msg] Gazebo Sim GUI    v7.3.0
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:104] Qt using Metal graphics interface
[GUI] [Dbg] [Application.cc:596] Create main window
[GUI] [Wrn] [Application.cc:845] [QT] Populating font family aliases took 49 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[... 138 lines cut ...]
[GUI] [Msg] Move to service on [/gui/move_to]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[GUI] [Msg] Follow offset service on [/gui/follow/offset]
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
[GUI] [Dbg] [InteractiveViewControl.cc:178] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
[... cut ...]
#4    Object "AGXMetalG13X", at 0x1f44f2cc3, in AGXMetalG13X + 2305219
#3    Object "AGXMetalG13X", at 0x1f44f0f57, in AGXMetalG13X + 2297687
#2    Object "libsystem_platform.dylib", at 0x1a98442a3, in _sigtramp + 55
#1    Object "libgz-tools2-backward.dylib", at 0x102ab7587, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#0    Object "libgz-tools2-backward.dylib", at 0x102ab75df, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 59
[1]    46724 segmentation fault  gz sim -v 4 -g
```

danielkroeni, 2023-02-21:
Thank you all for your time and effort.
I will use gazebo on my ubuntu installation for the time being.

If there is something to be tested on macOS m1, feel free to send me a message.

kev278, 2023-04-06:
Is there a stepwise documented fix for OSX? I see a lot of answers, any suggestions on which one to follow?
@danielkroeni

danielkroeni, 2023-04-19:
@kev278 unfortunately i couldn't get gazebo to run stable on m1 osx.

mkarklins, 2023-04-21:
I'm basically getting all of the same issues on macOS ventura running intel.

GLtangzero, 2024-04-16:
Please using gazebo harmonic, which is LST and suit of m1, I guess is the same problem with the m1 computer.