gazebosim/gz-sim #1867: CmdLine tests failing in gz-sim7 and ign-gazebo3

Crola1702, 2023-01-24:
## Environment
* OS Version: MacOS Bigsur
* Binary build - Garden and Citadel

## Description
* Expected behavior: Test should pass
* Actual behavior: CmdLine Tests are failing

## Steps to reproduce

1. Run a build in [gz-sim7-homebrew](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/k)
2. See 5 test regressions

## Output
Reference builds: 
* [gz-sim7](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/#showFailuresLink)
* [ign-gazebo3](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/273/)

Failing tests:
* [CmdLine.Server](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/testReport/junit/(root)/CmdLine/Server/)
* [CmdLine.CachedFuelWorld](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/testReport/junit/(root)/CmdLine/CachedFuelWorld/)
* [CmdLine.GazeboServer](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/testReport/junit/(root)/CmdLine/GazeboServer/)
* [CmdLine.Gazebo](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/testReport/junit/(root)/CmdLine/Gazebo/)
* [CmdLine.ResourcePath](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-homebrew-amd64/51/testReport/junit/(root)/CmdLine/ResourcePath/)

#### Gz-sim7 log output:
```
The following tests FAILED:
	 53 - UNIT_gz_TEST (Failed)
	263 - testFixture_TEST (Failed)
```

Test 53 - UNIT_gz_TEST:
<details>

```
test 53
        Start  53: UNIT_gz_TEST

53: Test command: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/bin/UNIT_gz_TEST "--gtest_output=xml:/Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/test_results/UNIT_gz_TEST.xml"
53: Working Directory: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/src
53: Environment variables: 
53:  GZ_CONFIG_PATH=/Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/test/conf
53:  GZ_SIM_SYSTEM_PLUGIN_PATH=/Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib
53: Test timeout computed to be: 240
53: Running main() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/test/gtest_vendor/src/gtest_main.cc
53: [==========] Running 7 tests from 1 test suite.
53: [----------] Global test environment set-up.
53: [----------] 7 tests from CmdLine
53: [ RUN      ] CmdLine.Server
53: Running command [/usr/local/opt/ruby/bin/ruby /usr/local/bin/gz sim -s  -r -v 4 --iterations 5 /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/test/worlds/plugins.sdf]
53: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/src/gz_TEST.cc:77: Failure
53: Expected: (output.find("iteration " + std::to_string(i))) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
53: Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib.
53: 
53: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/src/gz_TEST.cc:77: Failure
53: Expected: (output.find("iteration " + std::to_string(i))) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
53: Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib.
53: 
53: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/src/gz_TEST.cc:77: Failure
53: Expected: (output.find("iteration " + std::to_string(i))) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
53: Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib.
53: 
53: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/src/gz_TEST.cc:77: Failure
53: Expected: (output.find("iteration " + std::to_string(i))) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
53: Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib.
53: 
53: /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/ign-gazebo/src/gz_TEST.cc:77: Failure
53: Expected: (output.find("iteration " + std::to_string(i))) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
53: Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib.
[... 249 lines cut ...]
13: Expected: (output.find("Unable to find or download file")) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
13: Library error: Problem running [gazebo]() from /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/build/lib/libignition-gazebo3-ign.3.14.0.dylib.
13: 
13: [  FAILED  ] CmdLine.ResourcePath (1120 ms)
13: [ RUN      ] CmdLine.GazeboHelpVsCompletionFlags
13: [       OK ] CmdLine.GazeboHelpVsCompletionFlags (151 ms)
13: [ RUN      ] CmdLine.ModelHelpVsCompletionFlags
13: [       OK ] CmdLine.ModelHelpVsCompletionFlags (144 ms)
13: [----------] 7 tests from CmdLine (3953 ms total)
13: 
13: [----------] Global test environment tear-down
13: [==========] 7 tests from 1 test suite ran. (3953 ms total)
13: [  PASSED  ] 2 tests.
13: [  FAILED  ] 5 tests, listed below:
13: [  FAILED  ] CmdLine.Server
13: [  FAILED  ] CmdLine.CachedFuelWorld
13: [  FAILED  ] CmdLine.GazeboServer
13: [  FAILED  ] CmdLine.Gazebo
13: [  FAILED  ] CmdLine.ResourcePath
13: 
13:  5 FAILED TESTS
 13/174 Test  #13: UNIT_ign_TEST ........................................***Failed    4.10 sec
```

</details>

scpeters, 2023-02-06:
I wonder if it's a `DYLD_LIBRARY_PATH` issue? I'll see if I can reproduce on my machine

~~~
Library error: Problem running [sim]() from /Users/jenkins/workspace/ignition_gazebo-ci-gz-sim7-homebrew-amd64/build/lib/libgz-sim7-gz.7.1.0.dylib
~~~

scpeters, 2023-02-06:
I have a fix for some of the tests in https://github.com/gazebosim/gz-sim/pull/1884 targeting `gz-sim7`. We can backport to `ign-gazebo3` if it works

Crola1702, 2023-02-07:
> We can backport to ign-gazebo3 if it works

Actually, it works fine. We don't have any new test regressions. I see backport to gazebo3. Do we have one for gazebo6?

scpeters, 2023-02-08:
> > We can backport to ign-gazebo3 if it works
> 
> Actually, it works fine. We don't have any new test regressions. I see backport to gazebo3. Do we have one for gazebo6?

we'll forward port from `ign-gazebo3` to `ign-gazebo6` after https://github.com/gazebosim/gz-sim/pull/1885 is merged

Crola1702, 2023-03-06:
Closing as this problem was solved with the PRs mentioned on previous comments